dbuezas/lgt8fx #1: SoftwareSerial - 32Mhz support

HI-SEBA, 2019-08-28:
In 32Mhz Mode SoftwareSerial does not work.

#error This version of SoftwareSerial supports only 20, 16 and 8MHz processors

Maybe it only needs new DELAY_TABLE for F_CPU == 32000000 in SoftwareSerial.cpp?

dbuezas, 2019-08-28:
Haven't used SoftwareSerial yet in this board, but you're probably right! let me know if you get it to work and we can add it :)

dbuezas, 2019-09-02:
Do you know anything about how this table is computed? I'm thinking about connecting the Hardware Serial to the Software Serial and make a sketch that searches for values that succeed in transferring data through the Software Serial Pins.

HI-SEBA, 2019-09-02:
I had tried the following Perl script but unfortunately without success.

[soft-serial-calc](http://www.wiblocks.com/docs/app-notes/software-serial.html)

dbuezas, 2019-09-02:
Ok thanks. I'll try the brute force approach soon.

I just experimented connecting tx and rx to pins 10 and 11 and uploading  Examples/SoftwareSerial/SoftwareSerialExample to see if I can send and receive to/from the same board and I get feedback.

I'll soon try storing the table in memory and looping through all cells in the table until I get feedback for each baud rate. That should work.

dbuezas, 2019-09-03:
This is the first combination that works  at 32Mhz (found through brute force) 
- both ways communication with it's own hardware serial
bauds: 57600 	 rxcenter: 20	 rxintra: 85	 rxstop: 72	 tx: 84

Would you test if it works communicating with whatever you are trying to communicate so I know i'm on the right track?
@HI-SEBA

dbuezas, 2019-09-03:
at 32Mhz and 57600 bauds, these ranges seem to work
rxcenter 0-62
rxintra 85-96
rxstop 44-200 and more
tx 84-92

at 115200 bauds it doesn't seem to work with the "factory" table at 16Mhz

dbuezas, 2019-09-03:
SoftwareSerial seems to work at 230400 bauds at 32 Mhz!
{ 230400,	1,	7,	7,	6	},

dbuezas, 2019-09-03:
@HI-SEBA 
Would you try if this table works for you?

There are some baud rates for 32Mhz and corrections on the 16Mhz one. Plus there is now 230400 for both clocks.

I can currently only test this within the same board so I'm not sure if my method is robust enough.
Just add this to SoftwareSerial.cpp where the tables are defined:
```
//Edit by dbuezas

#if F_CPU == 32000000
static const DELAY_TABLE PROGMEM table[] = 
{
  //  baud    rxcenter   rxintra    rxstop    tx
  	{ 230400,	1,	18,	18,	14	},
  { 115200,	3,	43,	43,	32	},
  { 57600,	24,	88,	88,	66	},
  { 38400,	58,	134,	134,	100	},
  { 19200,	124,	275,	275,	210	},
};
const int XMIT_START_ADJUSTMENT = 5;

#elif F_CPU == 16000000
//Edit by Joinj
//LGT MCU is faster than mega

static const DELAY_TABLE PROGMEM table[] = 
{
  //  baud    rxcenter   rxintra    rxstop    tx
  { 230400,	1,	7,	7,	6	},
  // { 115200,   2,         21,        21,       16,    },
 	 { 115200,	2,	18,	18,	17	},

  // { 57600,    12,        43,        43,       38,    },
  	{ 57600,	12,	42,	42,	39	},

  // { 38400,    29,        66,        66,       62,    },
  	{ 38400,	29,	66,	66,	64	},

  { 31250,    36,        81,        81,       79,    },
  { 28800,    39,        89,        89,       86,    },
  { 19200,    62,        135,       135,      132,   },
  { 14400,    85,        180,       180,      177,   },
  { 9600,     132,       273,       273,      270,   },
  { 4800,     269,       548,       548,      545,   },
  { 2400,     544,       1098,      1098,     1096,   },
  { 1200,     1093,      2199,      2199,     2197,  },
  { 600,      2196,      4398,      4398,     4397,  },
  { 300,      4392,      8806,      8806,     8810,  },
};
//end Edit by Joinj
//end Edit by dbuezas
```

dbuezas, 2019-09-03:
This is my branch for tests
https://github.com/dbuezas/lgt8fx/tree/brute-force-software-serial-tables

HI-SEBA, 2019-09-04:
Well done! Sure i will test it. 
Pls add values for 9600 baud. 
My LGT8F328P is hardwired to a device that requires 9600 for initial setup.

dbuezas, 2019-09-04:
Darn... just where I stopped. I'll try to fill that one tonight

dbuezas, 2019-09-05:
Couldn't get it to work at 9600 yet

dbuezas, 2019-09-07:
@HI-SEBA 
This one has all the mid points for all timings@32Mhz:
	{ 9600,	284,	578,	578,	564	},

Let me know if it works and I'll generate the rest and update the lib.

If you send and receive at the same time through software serials it get's buggy

closed 2019-09-08 by dbuezas

HI-SEBA, 2019-09-08:
Great! 9600 works! I keep testing next days.

dbuezas, 2019-09-09:
Awesome! I uploaded an update. It has all the delays for 32Mhz at all baud rates, and also fixed one for 16Mht (the coded ones didn'T work for me).

I think the final values were slightly different than the ones I send you in this issue, they worked better for me. Would you mind double checking if the new ones also work for you?
Such a pitty that my oscilloscope only has 1 channel....

One can actually go to faster speeds even at lower clocks by changing the delay function to what the newer Arduino cores use (it consumes 2 ticks instead of 4). But I didn't want to potentially break it.