dbuezas/lgt8fx #2: fastio_digital.h not included

dcfusor, 2019-09-06:
At least in linux.  It does exist in the original git, and copying it in to:
/home/doug/.arduino15/packages/LGT8fx Boards/hardware/avr/1.0.2/cores/lgt8f
(I'm doug on my own box) makes it available, but you still have to explicitly #include <fastio_digital.h> to use it.
Using arduino 1.8.8 at the moment.

FWIW, these functions seem to require direct use of pins like D10 - defining "SOME_IO_PIN D10" for example, does not work when you use "SOME_IO_PIN" and gives an error.

Neither is a big deal, the missing file is.  It looks like there are several missing, but I haven't worked through the entire thing yet.

dbuezas, 2019-09-06:
Good catch!

I see what happened. This repo is based on Larduino_HSP_v3.6c.zip, not in the last core in https://github.com/LGTMCU/Larduino_HSP. 

The last version has the timers configured differently, that exposes more powerful features but breaks "standard" arduino compatibility (see: https://github.com/LGTMCU/Larduino_HSP/issues/4#issuecomment-506194745)

Not sure what to do about this the trade off is a bit harsh.

Thoughts?

dcfusor, 2019-09-06:
> powerful features but breaks "standard" arduino compatibility (see: LGTMCU/Larduino_HSP#4
> (comment)

I've only copied in the one file so far, and only have about 1 day looking
at this (and most of that was semi-porting something I'd written long ago),
so I'll defer saying much till I know something...

I see what that thread says, and I might vote for keeping timer1 as it was
- I'm doing nuclear instrumentation etc with these and
that counter is/was an extremely valuable resource - it can count an
external signal very fast and use interrupts on overflow to get large
counts without danger of losing any.  I'll get back to you once I've looked
it all over more thoroughly.  I also used to use the MsTimer2 interrupt
callback to control precisely how long some gas valves were open so that
matters too (to me anyway).  All that works (and has for years) on an
uno...hmmmm.  I guess we expect some difference here.

But as long as I have a way to do things like that, I don't much care *how*
I get them done - but I do have some really hard real time requirements,
which probably aren't that common in the community.  I will look into it
and get back to you.  You'd hope there was some way to set things up either
way, but from long (since the '70's) experience, I know what a hassle that
can be.

dcfusor, 2019-09-07:
Sorry, I've not had time to get on this more...I'm an off-grid homesteader,
and while the weather is good, I'm cutting wood for heat...
(Using solar power to run my chainsaws!)  It's a narrow time window I have
to work with.

I'm not even sure that guy's post on the other issue is *right *at this
point.  IIRC, time2 was already the millis timer.  Which is why I used that
MsTimer2 shim originally for valve timing rather than directly try to hook
its interrupt...I get the idea they do a fair amount with that one
themselves, so I left it alone.

There are (at least) two things to think about here.
1. Do you want to be come the "benevolent leader for life" of your
port/fork? While right now, you could probably solve it all with a couple
added #defines and #ifdefs, that way lies madness when people change things
upstream - especially without testing the few functions most arduino users
actually *do* use - that's not a good omen.  I guess it depends on how much
time you want to spend here.

2.  One possibility (?) could be to teach the upstream guys how to do what
you did and get arduino ide support done the correct way, then quietly back
away, except as you want to spend time on the conditional defines, related
projects, etc.

3.  I haven't though of a meaningful way to do a logical diff of what the
upstream guys did - not what lines of code changed so much, but the changes
in what hardware they take away because they figure they know what to do
with it better than the average user (they are probably right - how many
arduino people even have interrupts working? ).  The "we know better than
you" syndrome can go either way.
When I set up timer1 for my counter use, I had to talk to the registers
myself (not a big deal for me) - no one had written any code for it.
Maybe that's why the upstream thought it was free to grab?

The documentation I have doesn't seem to say if timer1 hardware is required
for more than 2 channels of the pwm, the rest are on other timers.  I'd
guess timer one would have more resolution, but be a lot slower if you used
it.  I know that in the things I used it as a counter for - I wasn't also
using pwm...

When I need any sort of good pwm, I use either a PIC chip or a cheap i2c
chip originally meant for leds that has a lot of channels anyway.
https://www.adafruit.com/product/815  for example works quite nicely here
and I'd bet is more versatile...(16 channels, 12 bit rez)

Hopefully I'll get a break from woodcutting soon and really get a dive into
this set of issues going on.

dbuezas, 2019-09-08:
wow, you're doing really interesting stuff! :)

keep me up to date with this!
I'm hoping the original dev of the core will pick this up so the LGT is easier to use to everybody, but I think they are not very active anymore on this.

For now I'll stick to the "compatible" version in this repo until I have more information.

closed 2019-09-08 by dbuezas

dbuezas, 2019-09-09:
It's in in V1.0.3 :)
I added you to a "thanks" section in the main readme for the help

commit a5b9f9e0 (2019-09-08): Added fast io. fixes #2