dbuezas/lgt8fx #4: LGT8F328P-SSOP20 not programming

dwillmore, 2019-09-10:
Hello, David, it's me from youtube.  I have some of the little SSOP20 boards like you have showin on the main page of this repo.  I have one hooked to a PL2303 USB<>serial adapter hooked to the boad on the end with the GND, VCC, RXI, TX0, DTR. The board is currently wired for 5V, FWIW.

I'm using the Arduino 1.8.5 environment and 1.0.3 of your support software.  The app I'm trying to flash in the Blink example from Arduino.

The errors I get are:
`avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.`

I do not know for certain that my board has a bootloader, but after I release the reset button, there are three very fast blinks and then about a 1 second delay before the built in blink sketch starts.  That seems likely to indicate there is abootloader in there.

Thank you for all of your efforts and for any help you can offer!

dbuezas, 2019-09-11:
Hi Willmore!
do you have this board:
https://github.com/dbuezas/lgt8fx/raw/master/docs/boards/ssop20.png

or this one?

dwillmore, 2019-09-11:
The first one, the SSOP20 version.  Thank you.

dbuezas, 2019-09-11:
Ok. And you swapped RX with TX, right?
adapter RX -> board TX
adapter TX -> board RX

I tested last night and I only get this kind of error if rx, tx and dtr are not solidly wired (or I configure the wrong port).

If all of that is right I am out of ideas except resetting everything by deleting 

```
(windows) C:\Users\<user>\AppData\Local\Arduino15
(mac) ~/Library/Arduino15
(linux) /home/<user>/.arduino15/
```

, restarting the IDE and starting from zero just in case the toolchain is broken.

If you end up uploading it via ISP I'd be interesting to know how it went.

Hope you get it working!

dwillmore, 2019-09-11:
Are you using the TX and RX on the end of the board next to GND/VCC and DTR?

dwillmore, 2019-09-11:
I have poked at the board a bit.  The DTR pin goes to RST vi a capacitor and resistor network--so a toggle will briefly assert reset.  I'm not seeing that happen though.

dwillmore, 2019-09-11:
Okay, I've changed programming cables and now it works.  This is a genuine original FTDI TTL-232R-5.0V  It doesn't have a DTR output, but instead has an RTS.  That may be one of the differences--but I do see a change on the DTR line of the other adapter.  

I'll try a few more with different wiring and get back to you.  Thank you for your support and assistance and I'm sorry for the mistake being my fault.

dwillmore, 2019-09-11:
Interesting.  Okay, I have modified another USB<>serial adapter (a ch341 based one) to have an RTS output.  I hooked it to the board and programming goes fine--but verification fails.  How odd is that?

Changing the command from the default write with verify to a simple verify *still fails*. So, there's something going on with the programmer/serial adapter/firmware that doesn't like to verify.

I'll look into it more.

closed 2019-09-11 by dwillmore

dwillmore, 2019-09-11:
I tried another FTDI breakout board and it works with DTR--writing and reading (verifying).  Works at 3.3V as well as 5V.

dbuezas, 2019-09-11:
That's great, happy to hear that!
It looks like I got really lucky lucky with my FTDI boards :)
BTW I this chip should work fine at full 32Mhz at voltages as low as 1.8v

dbuezas, 2019-09-11:
Thank you also for sharing the troubleshooting you did. I'm sure somebody else will face similar issues :)

dwillmore, 2019-09-11:
I'm glad to share my mistakes--especially if they'll help someone else avoid them.

Do you need to do something to enable the wiki for this repo?  I clicked on it and got nothing.  I'd be glad to start adding things there.

I think I need to dig the scope out and see what's different on that DTR line.

dbuezas, 2019-09-11:
You'll definitely save somebody else the frustration!
Haven't used github wikis yet, I hope creating the initial page suffices. Your stuff is very welcomed!

Keep me posted on the DTR line. I actually thought they would work equivalently in this scenario.

dwillmore, 2019-09-11:
Okay, the blank main page comes up now at least, so that's progress.  Can you give me edit/write ability to it and I'll start adding info to it.

dwillmore, 2019-09-11:
Okay, I reproduced the problem with a different machine--one close to the scope.  I haven't reporduced the working case, though, yet.

dbuezas, 2019-09-11:
Ok, now I understand how the wiki works.
Collaborators can edit wikis or one can open them up for editing to anybody.
Now they are open!

dbuezas, 2019-09-11:
> Okay, I reproduced the problem with a different machine--one close to the scope. I haven't reporduced the working case, though, yet.

So the same serial to usb adapter, right? curious to hear when (if at all) a pulse is sent through RTS

dwillmore, 2019-09-11:
Nothing shows up on the scope looking wrong.  I'll keep poking at it.  I guess I'll break out the LA and get some longer protocol captures.  Wonder if sigrok/pulseview have st500k?

dwillmore, 2019-09-12:
How confident are you in the current consumption numbers you posted?  Even at 32MHz, the chip feels cold--colder than the USB<>serial conver chip.  I'll make some measurements once I settle this issue.

dbuezas, 2019-09-12:
Somewhat confident... I tested a nano-styled one though, and through the linear regulator, so not a pure measurement.

dbuezas, 2019-09-13:
Wow! This is awesome!

dbuezas, 2019-09-13:
Can I add you in telegram, WhatsApp, or messenger?
Write to my name dot lastname @gmail com.

dwillmore, 2019-09-13:
email sent.

dwillmore, 2019-09-14:
Here are some screen shots of the write transaction:
[image]
[image]

I will try to attach the source files, but github is being stupid.  Sigrok files are zip compressed, but they use a different suffix and github is freaking out.  I'll rename them.  Hopefully anyone using sigrok/pulseview will figure out what to do.
[lgt_ch341_verify.sr.zip](https://github.com/dbuezas/lgt8fx/files/3612797/lgt_ch341_verify.sr.zip)
[lgt_ch341_write.sr.zip](https://github.com/dbuezas/lgt8fx/files/3612798/lgt_ch341_write.sr.zip)
[lgt_ftdi_verify.sr.zip](https://github.com/dbuezas/lgt8fx/files/3612799/lgt_ftdi_verify.sr.zip)
[lgt_ftdi_write.sr.zip](https://github.com/dbuezas/lgt8fx/files/3612800/lgt_ftdi_write.sr.zip)

dbuezas, 2019-09-15:
I'm not sure I know what this means. Ftdi and ch314 aren't equivalent
(after installing the drivers)?
I'd be really interested to learn what the screenshots mean. I have zero
experience with logic analysers

dwillmore, 2019-09-15:
FTDI and ch314 are both USB to serial converters.  I use Linux, so they both are supported and appear to the user as standard serial ports.  The program using them will have no idea that they're USB<>serial adapters or a actual hardware serial port of some kind.

What the charts show is the actual signals between the adapter and the SSOP20 board.  D0 is TX to the board, D1 is DTS, and D2 is RX.  Below those three traces are the decoded serial data on those lines.  It's zoomed way out so you can't see the actual data, but it's there if you want to load those files into pulseview. 

The thing you can see from the charts is how the delays between the various interactions are so much different between the two types of adapters.  I don't know if this is something in the kernel driver for the adapters or what, but I'm surprised to see how much different they are.

Kaitoukid93, 2020-04-01:
I have the same problem with my SSOP20 custom board, I use SWDICE MKII to upload the optiboot bootloader (that I found on the board directory) and the D13 LED flashing in sequence after power on
However, I cant upload the code because when Arduino IDE finished writing, it stuck at reading,
but this is not happen when I use CH330N adapter with the breadboard (Which only have DTR instead of RTS). I use the SWDICE MKII to upload the program code and it works just fine.
Then I look back my custom board Schematic and found that I use RTS line instead of DTR ( the USB serial chip CH340E on my custom board only has RTS pin)
is there any way I can make this work with RTS line instead of DTR (I meand upload using serial)
thanks

maxint-rd, 2020-06-09:
FYI: today I succesfully programmed my LGT8F328P SSOP20 EVB (BTE17-13) using a Baite BTE13-009 serial programmer (with DTR, voltage set to 5V). That USB-Serial converter uses the CH340G chip. No problems found (yet).

Kaitoukid93, 2020-06-09:
> FYI: today I succesfully programmed my LGT8F328P SSOP20 EVB (BTE17-13) using a Baite BTE13-009 serial programmer (with DTR, voltage set to 5V). That USB-Serial converter uses the CH340G chip. No problems found (yet).

your LGT board already has bootloader in it, so any serial converter will do the job, without DTR or RTS, you just simply press reset button, hold it, and release it right at the moment arduino IDE verbose windows show "Uploading.." 

In my case, if I burn the bootloader( from library directory) to a blank chip, but after that I cant upload the code using USB serial, both software (dtr,rts) and hardware (press reset manually) reset didn't work
I think there's something wrong with the bootloader because with same setup, the breakout board( BTE version) upload ok

maxint-rd, 2020-06-09:
@Kaitoukid93 - You're right. I've used that method on pro-mini's with other USB-serial boards, but using the Arduino IDE portable on a USB can be quite slow, so I prefer the one with DTR.
Unfortunately I have no SWDICE and have not yet burned a bootloader to this chip. Good luck in solving this.

freaklabs, 2020-08-15:
> Interesting. Okay, I have modified another USB<>serial adapter (a ch341 based one) to have an RTS output. I hooked it to the board and programming goes fine--but verification fails. How odd is [...]

Just adding my $0.02. In my experience with many CH340s, they have a very low tolerance for frequency deviation from their target frequency. So for ATMega328Ps using the internal RC oscillator, the frequency deviation often falls outside the tolerance which creates a lot of errors. So I have a lot of hit and miss attempts programming ATMega328Ps using CH340s when the MCU is running on internal oscillator. I suspect you are running into similar problems where you will get intermittent fails in programming.

dbuezas, 2020-08-15:
> In my case, if I burn the bootloader( from library directory) to a blank chip, but after that I cant upload the code using USB serial, both software (dtr,rts) and hardware (press reset manually) [...]

That's odd. I burned around 5 bootloaders with the lib provided in the package and I had no issues. 
* Did you use the option in the menu? (Burn bootloader
* Did you use another lgt8f328p as ISP?
* Was the target board ssop20 or lqfp32?

dbuezas, 2020-10-13:
you could test it with the ardu-scope 👓  :)
https://dbuezas.github.io/arduino-web-oscilloscope/

SuperUserNameMan, 2020-10-13:
> you could test it with the ardu-scope eyeglasses :)
> https://dbuezas.github.io/arduino-web-oscilloscope/

I'd be glad to, but unfortunately : "Requires an updated version of Chrome (≥ 86.x.x)"
(i'm on firefox) :-P

dbuezas, 2020-10-13:
"Open source project doesn't support Firefox" ups....
There's a good reason though: the WebSerial API is only available in Chrome :/

dwillmore, 2020-10-13:
I was confused when I went there because I was expecting a normal github
page for a project.  I didn't expect to be running the project.

SuperUserNameMan, 2020-11-01:
Apparently all my problems came from my CH340E adapter. 

According to this [datasheet](https://www.mpja.com/download/35227cpdata.pdf), CH340B, CH340C and CH340E use internal built-in "clock generator".

dbuezas, 2020-11-01:
> Apparently all my problems came from my CH340E adapter.
> 
> According to this [datasheet](https://www.mpja.com/download/35227cpdata.pdf), CH340B, CH340C and CH340E use internal built-in "clock generator".

Do you have an oscilloscope? It would be interesting to see the actual timings it ends up using.

dbuezas, 2020-11-01:
You could use the web oscilloscope for that! 🤑

SuperUserNameMan, 2020-11-01:
> > Apparently all my problems came from my CH340E adapter.
> > According to this [datasheet](https://www.mpja.com/download/35227cpdata.pdf), CH340B, CH340C and CH340E use internal built-in "clock generator".
> 
> Do you have an oscilloscope? It would be interesting to see the actual timings it ends up using.

I have a [MiniDSO DS213](https://www.seeedstudio.com/MiniDSO-DS213-Nano-4-Channel-100MSa-s-p-2891.html). 
Though I'm not sure the full upload signal could fit into it channel RAM buffer.

> You could use the web oscilloscope for that! :money_mouth_face:

Yes, but it does not work on Firefox ....... :'-(

dbuezas, 2020-11-01:
Oh. Firefox doesn't seem interested in the web serial api :(

In any case, if you want to debug that darn board, you can measure the width of a pulse. Just hook the oscilloscope to it's tx line and measure the width of the shortest pulse you can find.
It should be (1,000,000 us / bauds) wide

dwillmore, 2020-11-01:
I would strongly recommend that everyone doing this kind of debugging pick
up an inexpensive fx2lafw compatible logic analyzer.  Like the generic
Saleae Logic 8 channel boards.  They are very inexpensive and can stream 8
channels at 24Ms/s.  You can use Sigrok/Pulseview as a UI.  It supports
lots of protocols, etc.  I would recommend getting one before getting a
'scope.  Really, they're only a few dollars.