dbuezas/lgt8fx #5: WEMOS TTGO XI 8F328P uses LED_BUILTIN 12 not 13

cyzen, 2019-09-16:
The WEMOS TTGO XI 8F328P board uses LED_BUILTIN Pin 12 not 13
could you fix this in pins_arduino.h ?

thanks for you great work btw. !

dbuezas, 2019-09-16:
Hello there!
thanks for the kind words :)

I could add it but it'll take a while and as I don't have that board it would be untested.

If you want to make a PR, this is what I would try first:
1. duplicate `lgt8f/variants/lgt8fx8p/pins_arduino.h`to `lgt8fx8p-wemos-TTGO-XI/pins_arduino.h` and this at the end 
```c++
#undef LED_BUILTIN 
#define LED_BUILTIN 12
```
2. `lgt8f/boards.txt` -> add an entry by copying the one from 328P-LQFP32 and reference the new variant from (1) in `build.variant` (the rest should be ok as-is)
3. test it by simlinking the repo to where the arduino IDE looks for the package definitions (check this file, you'll need to change the folders to the ones you used https://github.com/dbuezas/lgt8fx/blob/master/simlink.sh)

dwillmore, 2019-09-16:
I've got these boards, so I can take this on.  I may have to go figure out how do do the github side of the work, though.

dwillmore, 2019-09-16:
I made those changes as you describe them and I have a flashing XI board.  It works as expected.  Now to figure out pull  requests...

cyzen, 2019-09-16:
excellent - thank yiou so much!

dwillmore, 2019-09-16:
You're welcome.

dwillmore, 2019-09-16:
Okay, pull request created.  It's #6   Please review and merge if it's okay.

@cyzen if you want to try it, it's branch TTGI-XI.

dbuezas, 2019-09-16:
Oh, It overrode the original 328p, I'll fix & deploy

dwillmore, 2019-09-16:
Okay, I see what I did.

@cyzen Do you have boards other than the Wemos TTGO-XI?  The wiki is in need of better photos of more boards.  I only have TTGO-XI and the SSOP20 variant of the Pro mini.

dbuezas, 2019-09-16:
fixed and deployed.
v1.0.4 it is now :)
Thanks guys!

If one updates via the boards manager and can check if the pin is now set correctly, please close the issue

dwillmore, 2019-09-16:
Works here!

I'll close when @cyzen verifies.  Thanks, David.

dbuezas, 2019-09-16:
Btw, I have the nano and wavgat pro mini versions. Front and back?

dwillmore, 2019-09-17:
Yes, please.  I used 4800dpi or something way overkill.  We can always scale it down later.  Don't worry about rotation, I can fix that easily in GIMP.

So it's the pseudo clone and the dev board style that we're missing?

dwillmore, 2019-09-19:
@cyzen Have you had a chance to test?  Shall we close?

cyzen, 2019-09-21:
So sorry for my late answer 
it works after updating like a charm  
thank you so much for you fast help. 

i'm happy - it can be closed now ... and other noops like me are not condemned to fail any more with the blink sketch :-)

dwillmore, 2019-09-22:
Glad it worked out.  If you run into any other issues, feel free to open another issue.

closed 2019-09-22 by dwillmore