dbuezas/lgt8fx #8: thanks

Upcycle-Electronics, 2019-09-24:
Thanks for all the effort you've put into this. 
   I don't know how far I'll get but I have already redone the first 8 pages of the PDF. These are somewhat reformatted as a test to see if others have issues. I'm not sure if I'm making it better or worse in the end. I am not anything more than a hobbyist hack here, but I'm trying to make this datasheet more useful for me and perhaps others as well. 
-Jake
https://github.com/Upcycle-Electronics/LGT8Fx-Datasheet/blob/master/LGT8FxDS_p1-8_scaled.pdf

dbuezas, 2019-09-25:
WOW! this is awesome!
there appears to be v1.0.5, but I couldn't find the original one.

here's a translation of it: https://www.avrfreaks.net/comment/2760926#comment-2760926

By all means keep me up to date, more than happy to add your translation here once you're ready for it

Upcycle-Electronics, 2019-09-25:
That is interesting. The quality of the doc is better, but even at first glance I see some issues. I don't understand a bit of Chinese myself, but I noticed, if I take each paragraph of Chinese, copy/paste it manually into Google translate, and then remove all of the spacing so that it appears exactly like it does in the PDF the translation makes perfect sense. There are a few minor terminology issues specific to electronics, but the flow and wording is really good. It's a lot of brute force work to do it like this. The only reason I see for continuing what I'm doing is the metadata bookmarks. If I can create a version that is a page for page match I can import the Chinese index. I have a tutorial guide to follow about making an index. Without the index and OCR text this thing will not get used as much. Ultimately I want to understand this thing well enough to use AVRdude and get a Forth system working on it. We'll see how far I get.

SuperUserNameMan, 2019-09-26:
duno if it will be useful to any of you, but : https://github.com/watterott/LGT8F328P-Testing

Upcycle-Electronics, 2019-09-27:
Have you seen where anyone has done a full register by register comparison between the LGT8XM and ATmega328? 
 After looking into the Atmel original datasheet, the LGT 1.0.4 'translates into English so well once the Chinese is formatted properly,' because it is actually the Atmel datasheet plagiarized in Chinese. By translating the bulk of the text it is creating all kinds of errors and issues. I think I need to start by looking at exactly what registers are different and using this to sort out what parts of the Logic Green datasheet are original. I have already revised (but not yet updated on git) the first dozen pages to remove any Logic Green names. I'm simply using the Atmel text directly section by section, where nothing has been added by LG. The only way I can do this effectively, in an organized fashion is to compare all the registers directly.
-Jake

dbuezas, 2019-09-28:
That's very Interesting! How sneaky of them :p

BTW, the board has a couple of functionalities that it shares with other atmel boards, maybe those sections are taken from their respective datasheets. For example the atmega32u4 and mega2560 have a Differential ADC with programmable gain, which may be similar to the diff amplifier of the lgt.

dwillmore, 2019-09-28:
I opened an issue over at the Watterott repo suggesting they link to this one and that, if they're designing a board for one of these chips, that they open an issue here to add support. We'll see if anything comes from that.  But it makes sense to try to coordinate our actions.

dbuezas, 2019-10-21:
Let me know when you have more news in the documentation front :)

closed 2019-10-21 by dbuezas