dbuezas/lgt8fx #9: Bootloader upload and freeRTOS

CanHobby, 2019-11-15:
Hi - thanks for all of your work... I have several LGT8fs and love them..... except...

2 issues when using your package....

#1 - I cant load the bootloader...
Arduino: 1.8.10 (Linux), Board: "LGT8F328, 32 MHz internal, 328P-LQFP32 wemos-TTGO-XI"

/home/art/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/tools/avr/bin/avrdude -C/home/art/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0x07:m -Uhfuse:w:0xff:m -Ulfuse:w:0xff:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/art/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/art/.avrduderc"
avrdude: error at /home/art/.avrduderc:3: syntax error
avrdude: error reading user configuration file "/home/art/.avrduderc"
Error while burning bootloader.
I can upload my sketches with no problem..

#2 problem is running freeRTOS... so far vTaskDelay does not work... does not delay..

???? any ideas   ???
Thanks in advance....

Art
The download process is not picking up the USB Port selected in the Arduino IDE... and I can't figure out the syntax for .avrdudrc

HI-SEBA, 2019-11-15:
As far as i know Bootloader programming for LGT8F328**P** Version is currently only possible with SWDICE mkII Pro.

https://github.com/LGTMCU/LarduinoISP/issues/1

Upcycle-Electronics, 2019-11-15:
> LGTMCU/LarduinoISP#1 <https://github.com/LGTMCU/LarduinoISP/issues/1>

There is no way around getting their programmer. I went as far as trying to
scrape info from the main Chinese forum, but the logic green support forum
is invite only and is password protected. The forum's official logic green
section was public up until 2017. It can be found on archive.org. I found a
few AliEx sellers that appear to be on forum and selling logic green (or
massduino) stuff. I asked about getting a SWICE but didn't get a reply. I
also sent a few emails to the contact info on logic green. I never got a
reply from them. My interest is in an AVR because of the open toolchain
policy. If anyone finds a way to get the LG SWICE I'd like to hear about
it...and hook up a logic analyzer. Without an open protocol for the SWD
port the controller is not as interesting. They wrote a ton of info about
how the system works but they left out the most important part.
-Jake

HI-SEBA, 2019-11-15:
Hi Jake, here you will find everything to produce one.

Firmware, driver, .sch, .brd, BOM, aso.

https://maker.lceda.cn/p/costar/lgt_mini_box

CanHobby, 2019-11-15:
Thanks so much for the quick reply..  I spent most of yesterday banging my head against this problem... My conclusion is the fuse bits set in the supplied bootloader... SPI programming disabled as well as the WOTON watchdog is not set which is necessary for freeRTOS for it's tick timer...

Upon inspection I see there is no 6-pin ISP connector on any of the boards... Only a 5 pin UART connection...

What I don't understand ( which is a lot ) is why a SWD/JTAG can program via the UART and we can't..
Must just be a software protocol or something...  or something to do with the lock bit...

I found this on eBay but it's technical disclaimers looses a lot in translation.... Any ideas?? I'm handicapped and unable to solder so I would need a ready built board.  The site you gave me is all in Chinese so it does me no good...

Too bad - these board work great on most Arduino projects...   But with the price/performance of the ESP32 this inflexibility is a big drawback..

But many thanks for your lengthy reply....

https://www.ebay.ca/itm/OB-ARM-Debugger-Programmer-Downloader-replace-v8-SWD-M74-For-J-Link/400943523132?hash=item5d5a18a53c&_trkparms=ispr%3D1&enc=AQAEAAADEKvsXIZtBqdkfsZsMtzFbFsbX3WcW5fmB%2Fx7ZbaZTyexGjfr%2FNqIWnmAGHQu5GuGGHF1VqgONRE8Si2W1HhriOUnLVsn3CzgKjqXbvZSlaWz%2FTznwzQGfr%2Bj%2BrfiSszqQtCAmwjABgL%2F4Oej%2FMxfFKUPNIVJ5ew4mPJOjn2Q9AU2gWSNOCIFlyXydQRpv%2BYBYGb7WuYOj8aaD%2B68nF5EkdmoJZdu02oOCrAFY82VfrKDnTn3yccl7vyuB6KJShk5rQl6iyGObXmi33cB6c8S9JxTeQnkNs8V2wxFc0sQyt9UKBNFqEkTqy7980TiR6XXKXkZ8lLqc5V6QV3HxEa8S8Z01%2FZc41jCcGdrUzgRZ4XM8EcmqvjLZ0Di3MNLSnU3jRHRq4JS1Hn8HeC3lFXs%2FyZc%2F6LAGCAvAFXaKHjPcvcrjFHykCbzJtBjI6xSZ3%2FuAlaVfitWie5n%2FjilABbREbNgS6OB0bf7uc4sN5lUMbULkIZ%2BMHfZMi4T3GVJVSZteRhOkMf%2Fc8dzdwZJUVM%2BuVL2ujUUQWFx6HBZ2%2F7HqtVipFi63XOA1EaxMJThtJzArph3FJ9hqBLwx9WvxrvLwR%2B03B7XV0f4a3QyTRJ5hhNMpILcTgzxF9%2FUPaXPcpCr%2BXIipVXV59uzOgpF%2FjxP1GrKdr5A99NMRE2kTLGE0%2BvdYCYCEWKLAeClJ%2FpEKWdBZWFQPKYmz7Ls9MnQKSIJJTV5sKC%2BPAnpcqYbwULmZ7YkAeRUipLNEp6Je9cSDD7llHSRDg5wAH6FRtcGddmCw8GKH9uwJYuwtTZQ%2F8uzKXQdw2qI%2BZpppQYwx2hpngr55PnbbSHzTk2aRcVXJGZ%2BrEi3FD2Ne17dFVCSGGqFmP%2BFW2PUoymYFOeZXcYQqbNQl6%2FT7gu%2FfOm5L%2BVQPNUHDR1Rwp9uShPuN9CkIfAuCDp7Mv1b%2Fd6cDTe95yuDIRwqp3Br%2FLmxPQ%2FwjRrsJkYS3q5BaDg8uvI%2FcTf56Aebxl0L6O1952jGMbXeLHMfgnWKX7%2F02dhIm6l%2F0j%2F9d8HxL88%3D&checksum=400943523132d0afdfe69c7d4f439d95999dc83b2db0

dbuezas, 2020-01-13:
Regarding the fuse you are talking about, I think this chips doesn't hold configurations in fuses. It looks like everything is stored in control registers accessible at runtime.

Check out page 59 Datasheet (English) [LGT8FX8P_databook_V1.05](https://github.com/watterott/LGT8F328P-Testing/raw/master/LGT8FX8P_databook_v1.0.5-English.pdf) By Watterrott: 

> WDIE and WDE in combine to determine the working mode of watch dog

So if you can configure the watchdog timeout at setup by doing:
```cpp
bitClear(WDTCSR, WDTCSR);
bitSet(WDTCSR, WDIE);
```

Or rather set the whole register at once including the prescaler you wish. e.g.:
```cpp
WDTCSR =
    0 << WDE | 1 << WDIE | // set to interrupt mode without resetting
    1 << WDE | // enable Watchdog
    1 << WDP3 | 0b001 << WDP; // set prescaler to 512ms
```
If you want longer times you'll have to set the WCLKS bit to 1 in the Clock management register-PMCR (page 42 & 43 of the same doc).

You'll probably have to dig a bit in the datasheet but what I wrote may do something already.
BTW it is totally worth exploring the datasheet, I got to use the ADC at 384kHz with a 32x gain to do some ultrasound audio processing and spit the result as audio through the integrated DAC by playing with a bunch of registers.

dbuezas, 2020-01-13:
> As far as i know Bootloader programming for LGT8F328**P** Version is currently only possible with SWDICE mkII Pro.
> 
> [LGTMCU/LarduinoISP#1](https://github.com/LGTMCU/LarduinoISP/issues/1)
> 
> LGTmix_ISP Software
> http://www.lgtic.com/lgt8fx8p/
> 
> DIY Programmer
> https://maker.lceda.cn/p/costar/lgt_mini_box

You did some serious digging here @HI-SEBA!

dbuezas, 2020-07-10:
It is now possible to burn bootloaders w/othe SWICE, more in [issue 13](https://github.com/dbuezas/lgt8fx/issues/13#issuecomment-656392739)

closed 2020-07-21 by dbuezas