dbuezas/lgt8fx #13: Use picoboot instead of optiboot

dwillmore, 2020-01-12:
I'm not sure how small optiboot compiles down to, but picoboot gets under 256 bytes.  Would it be useful to move to it?  Since we have our own boards.txt anyway, we avoid the problems of the Atmel chips needing their boards.txt file manually edited.

The only concern would be that boards probably come with optiboot and this would require all boards to be reflashed to picoboot.  That's probably a non-starter.

But I though I'd bring it up.

dbuezas, 2020-01-13:
Unfortunately there is currently no easy way of flashing the bootloader in the 328p boards, there is more info in this issue: https://github.com/dbuezas/lgt8fx/issues/9#issuecomment-554247316

Maybe the engineers from Larduino or LGT decide to make the the ISP programmer for 328d  compatible with 328p some day 🙏

dwillmore, 2020-01-13:
Seems correct.  I just thought I'd bring it up. :)  Maybe LGT reads this and decides to preprogram a different bootloader. ;)

dbuezas, 2020-07-09:
There seems to be code now to make an ISP programmer for the 328p
https://github.com/brother-yan/LGTISP

And @seisfeld also noticed me that the more "official" Larduino ISP also does now:
https://github.com/LGTMCU/LarduinoISP

I haven't tested it yet.

dbuezas, 2020-07-09:
I can confirm that the LarduinoISP does now work on 328p chips 🎉 
It required a bit of fiddling but it works:
- Increase the SERIAL_RX_BUFFER_SIZE to 250 in HardwareSerial.h in wherever core you use as programmer.
- Add the cap to reset or just bridge that pin of the programmer to vcc.
- Add this to programmers.txt of the lgt328p core:
```
avrisp.name=AVR ISP
avrisp.communication=serial
avrisp.protocol=stk500v1
avrisp.speed=19200
avrisp.program.tool=avrdude
avrisp.program.protocol=stk500v1
avrisp.program.tool=avrdude
avrisp.program.extra_params=-P{serial.port}
```
And presto!

I can make a small tutorial and update this core to make it easier if there is demand. I can also add an extra switch in the menu to increase the SERIAL_RX_BUFFER_SIZE for this to be super easy when one uses another lgt328p as the programmer.

I do not know yet if one can just use picobot as is (probably not), but it is possible to just not use any bootloader for some extra flash space and faster startup.

CanHobby, 2020-07-15:
Fantastic job all of you!!!  Just managed to load a bootloader onto my LGT8F328p based 32MHz Nano...

Worked better when I reversed the order of 2 lines of code ( pinMode of the RESET line )

I did some fudging and kludging and got the extra interrupts working ( timer 3 etc. )... 

Tomorrow I try the WatchDog interrupt which was my original concern....

I was hoping to be able to use miniCORE but that didn't work...

so I munged David's core and got printf working...

Happy Camper Here....

dbuezas, 2020-07-21:
Hey @CanHobby I curious to know what you did with the WatchDog interrupt. If you got some example code, please post it as an issue so others can use it as an example :)

closed 2020-07-21 by dbuezas