dbuezas/lgt8fx #15: PlatformIO support?

ameeuw, 2020-01-29:
Hi dbuezas - great work on integrating all the parts of this chip for the Arduino IDE - it works perfectly!
As I am doing most of my projects in PlatformIO my question is whether it is possible to port the board definitions over?

Maybe someone else has experience and could chime in how much work this requires.

Thanks again. Best, Arne.

dbuezas, 2020-01-30:
Hey ameeuw!
I've used vscode with the arduino extension until now. I'll try platformio after vacations and see if I can add this core to the supported boards. Thanks for the issue :)

kostyamat, 2020-02-28:
Hi. Any progress?

featherfly, 2020-04-10:
two way worked with platformio
### defined boards and pcakge in project
1. edit platform.json to add lgt package support
add framework-arduino-avr-lgt8f to the file %platformio_home%.platforms\atmelavr\platform.json
`
    "framework-arduino-avr-lgt8f": {
      "type": "framework",
      "optional": true,
      "version": "~3.6.3"
    }
`
2.  add borads support file
  add borads dir to project dir and add boards_support.json in it, such as lgt8fx8p.json
  file example here [](https://github.com/maxgerhardt/larduino-pio/tree/master/boards)
3.  edit project platformio.ini
`
[env:lgt8f]
platform = atmelavr
board = lgt8fx8ps20  ;LGT8F328P-LQFP48 MiniEVB
framework = arduino
platform_packages =
	framework-arduino-avr-lgt8f @ https://github.com/maxgerhardt/Larduino_HSP.git
`
### defined boards and package global
1. same
2. add custom url to download framework-arduino-avr-lgt8f arch file
`
"packageRepositories": [
    "https://dl.bintray.com/platformio/dl-packages/manifest.json",
    "http://dl.platformio.org/packages/manifest.json",
    "https://raw.githubusercontent.com/eerimoq/simba/master/make/platformio/manifest.json",
    {
      "framework-arduino-avr-lgt8f": [
        {
          "url": "https://yoursite.com/path/framework-arduino-avr-lgt8f-3.6.3.zip",
          "sha1": "your file sha1 code",
          "version": "3.6.3",
          "system": "*"
        }
      ]
    }
  ]
`

> note this arch file must has a package.json file such like this:

`
{
    "name": "framework-arduino-avr-lgt8f",
    "description": "Arduino Wiring-based Framework (AVR Core, 1.6)",
    "url": "https://github.com/LGTMCU/Larduino_HSP/",
    "version": "3.6.3"
}
`
directory structure:
- bootloaders
- cores
- libraries
- sketches
- variants
- boards.txt
- package.json
- platform.txt
- programeers.txt

2. add borads support file to %platformio_home%.platforms\atmelavr\boards

> after added global boards, when create platformio project can choose it

more info see [Custom Development Platforms](https://docs.platformio.org/en/latest/platforms/creating_platform.html)  and [Custom Embedded Boards](https://docs.platformio.org/en/latest/platforms/creating_board.html)

darkautism, 2020-04-17:
Hi all,
I create a platform which can running program in platformIO.
The code is here.
https://github.com/darkautism/pio-lgt8fx

Just clone this repository into your `%USER%\.platformio\platforms\`

seisfeld, 2020-04-17:
@darkautism Does your code support the change of clock source and clock speed?

darkautism, 2020-04-17:
@seisfeld It does not support now. Maybe next version.

seisfeld, 2020-04-17:
That would be nice, because running the chip at 32Mhz breaks certain Arduino libraries. The change to external clock source is also beneficial in certain cases.

darkautism, 2020-04-17:
@seisfeld

An example setting
``` ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:LGT8F328P]
platform = lgt8f
board = LGT8F328P
framework = arduino

board_build.f_cpu=16000000L
;internal clock
board_build.clock_source=1
```
Example code:
``` cpp
#include <Arduino.h>
void setup() {
  Serial.begin(9600);
  #if ((F_CPU) == 1000000)
  #warning F_CPU = 1MHz
  Serial.println("1MHz");
  #elif ((F_CPU) == 2000000)
  #warning F_CPU = 2MHz
  Serial.println("2MHz");
  #elif ((F_CPU) == 4000000)
  #warning F_CPU = 4MHz
  Serial.println("4MHz");
  #elif ((F_CPU) == 8000000)
  #warning F_CPU = 8MHz
  Serial.println("8MHz");
  #elif ((F_CPU) == 16000000)
  #warning F_CPU = 16MHz
  Serial.println("16MHz");
  #elif ((F_CPU) == 32000000)
  #warning F_CPU = 32MHz
  Serial.println("32MHz");
  #endif

  #if ((CLOCK_SOURCE) == 1)
  Serial.println("internal clock");
  #elif ((CLOCK_SOURCE) == 2)
  Serial.println("external clock");
  #endif

  while(1);
}

// the loop function runs over and over again forever
void loop() {
}
```

Example output:

```
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM8  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
32MHz
internal clock
```

maxgerhardt, 2020-05-08:
Parallel to that, some also integrated this core in PIO another way: https://github.com/rick3rt/framework-arduino-avr-lgt8f 

And also I integrated the original chinese Larduino core (v3.6c version): https://github.com/maxgerhardt/larduino-pio and https://github.com/maxgerhardt/Larduino_HSP

Seems like we should combine our efforts to get one best version officially integrated into PlatformIO 😅

dbuezas, 2020-07-21:
I'd welcome a PR on this :)
I use VSCode with the Arduino extension. It is probably better that I don't diverge too much from the Arduino IDE since most users go that route.

maxgerhardt, 2020-07-21:
The "Arduino Core" repository are separate from the "PlatformIO Platform" repository; The core files never contain any of the needed Python logic and JSON packaging descriptions. For example https://github.com/platformio/platform-ststm32 is the PlatformIO platform logic (available board descriptions, compilation logic, uploading logic, packages like compilers and uploads, ..) while https://github.com/stm32duino/Arduino_Core_STM32 is only 1 of like 4 underlying repos where the core is sourced from. 

So really this repo is fine already, but there shouldn't be different divergent PlatformIO integrations for this core with differing board or configuration settings.

xfjx, 2020-12-27:
I also tried to use your core in Platform.io today.

There is just one file missing in your distribution - the package.json

So I just added this file to the zip!
```
{
    "name": "framework-arduino-avr-lgt8f",
    "description": "dbuezas's LGT Core",
    "url": "https://github.com/rick3rt/framework-arduino-avr-lgt8f",
    "version": "1.0.6"
}
```

After that I added this section to the platform.ini in order to use your core as the framework-lgt8x. If you add this file to the release zip in the future it will be possible to just use a http reference to your file directly.

```
[env:LGT8F328P]
platform = lgt8f
board = LGT8F328P
board_build.f_cpu = 16000000L
board_build.clock_source = 1
framework = arduino
platform_packages = framework-lgt8fx@file:///Users/xfjx/Documents/PlatformIO/Projects/TonUINO/lgt8f-1.0.6.zip

```

dwillmore, 2023-01-12:
Has this been corrected in later releases?  @xfjx can you verify?