dbuezas/lgt8fx #17: There's no option for LGT8F328"P" ssop 20 package

Kaitoukid93, 2020-02-15:
I have a new board with the P version of ssop20 package, but I can't find the board in the board menu
because I also want to burn the bootloader with the SWDICE_MKII bought from china
Can I just add it to the board menu (all the same with 328P except for the bootloader's path, because I found the bootloader for ssop20 P version in the bootloader directory )
I'm new to coding
thanks

dbuezas, 2020-02-19:
For now try the option for lqf32, it should work. Keep us posted about the programmer :)

dbuezas, 2020-02-19:
Check the pull requests section! Somebody did the correction! I'll get back to this soon

dwillmore, 2020-06-09:
If any of you with variant boards would be so kind as to take some nice pictures of them and include a link to where you got them, I'll add it to the wiki so that others can benefit.

Kaitoukid93, 2020-06-09:
It's all OK now because I decided to disable the firmware update function on the product, bootloader no more needed
the programmer from LGT (SWDICE) work quite well, I program almost a thousand unit of ssop20 version without any problem, not a single one "dead"
this chip is "cheap" , only 0.2 USD  for a fully arduino compatible 
If anyone try to put it on a custom board, be careful to select "ssop20w", not "ssop20" footprint, this drive me crazy, the first batch using wrong footprint
If you need any picture of custom board of this chip, I can send you, I've made a lot 
[image]

dbuezas, 2020-07-10:
Nice board! Congrats on making a thousand of them :tada: 

A bait late, now it is possible burn bootloaders with another Arduino. More in #9 

Feel free to send more pictures or talk about your product, I'm quite curious about it!

dbuezas, 2020-07-21:
This was solved in v1.0.6 thanks to @LaZsolt

closed 2020-07-21 by dbuezas