dbuezas/lgt8fx #18: delayMicroseconds()

LaZsolt, 2020-03-10:
In wiring.c delayMicroseconds() function using SBIW and BNRE instructions for timing.
In AVR MCUs it takes usually 4 clock cycles while in LGT MCUs takes usually 3 clock cycles.
Therefore this function delay less than expected.
I found a better version in a chinese site. The author corrected only 16 and 32 MHz cases.
https://www.geek-workshop.com/thread-38486-1-1.html

LaZsolt, 2020-03-10:
I think the best solution for LGT MCUs is if we use a NOP between SBIW and BRNE.

LaZsolt, 2020-03-14:
`// In case of using LGT MCU need one more NOP instuction to compensate shorter executiom time.
#if defined(__LGT8F__)
  #define _ONENOP_ " nop \n\t"
#else
  #define _ONENOP_ ""	
#endif

#define _MORENOP_ ""

/* Delay for the given number of microseconds.  Assumes a 1, 8, 12, 16, 20 or 24 MHz clock. */
void delayMicroseconds(unsigned int us)
{
	// call = 4 cycles + 2 to 4 cycles to init us(2 for constant delay, 4 for variable)

	// calling avrlib's delay_us() function with low values (e.g. 1 or
	// 2 microseconds) gives delays longer than desired.
	//delay_us(us);
#if F_CPU >= 32000000L
	// for the 32 MHz clock on most Arduino boards

	// The overhead of the function call takes 14 (16) cycles,
    //	which is half microsecond

	// zero delay fix
	if (!us) return; //  = 3 cycles, (4 when true)

	// the following loop takes 1/4 of a microsecond (8 cycles)
	// per iteration, so execute it four times for each microsecond of
	// delay requested.
	us <<= 2; // x4 us, = 4 cycles
	
	// 32 MHz is two times faster than 16 MHz so need more nop in the
	// wait cycle for to keep the possible longest delay capability
#undef  _MORENOP_
#define _MORENOP_ " nop \n\t  nop \n\t  nop \n\t  nop \n\t"

	// account for the time taken in the preceeding commands.
	// we just burned 19 (21) cycles above, remove 3, (3*8=24)
	// us is at least 4 so we can substract 3
	us -= 3; // = 2 cycles,

#elif F_CPU >= 24000000L
	// for the 24 MHz clock for the aventurous ones, trying to overclock

	// zero delay fix
	if (!us) return; //  = 3 cycles, (4 when true)

	// the following loop takes a 1/6 of a microsecond (4 cycles)
	// per iteration, so execute it six times for each microsecond of
	// delay requested.
	us *= 6; // x6 us, = 7 cycles

	// account for the time taken in the preceeding commands.
	// we just burned 22 (24) cycles above, remove 5, (5*4=20)
	// us is at least 6 so we can substract 5
	us -= 5; //=2 cycles

#elif F_CPU >= 20000000L
	// for the 20 MHz clock on rare Arduino boards

	// for a one-microsecond delay, simply return.  the overhead
	// of the function call takes 18 (20) cycles, which is 1us
	__asm__ __volatile__ (
		"nop" "\n\t"
		"nop" "\n\t"
		"nop" "\n\t"
		"nop"); //just waiting 4 cycles
	if (us <= 1) return; //  = 3 cycles, (4 when true)

	// the following loop takes a 1/5 of a microsecond (4 cycles)
	// per iteration, so execute it five times for each microsecond of
	// delay requested.
	us = (us << 2) + us; // x5 us, = 7 cycles

	// account for the time taken in the preceeding commands.
	// we just burned 26 (28) cycles above, remove 7, (7*4=28)
	// us is at least 10 so we can substract 7
	us -= 7; // 2 cycles

#elif F_CPU >= 16000000L
	// for the 16 MHz clock on most Arduino boards

	// for a one-microsecond delay, simply return.  the overhead
	// of the function call takes 14 (16) cycles, which is 1us
	if (us <= 1) return; //  = 3 cycles, (4 when true)

	// the following loop takes 1/4 of a microsecond (4 cycles)
	// per iteration, so execute it four times for each microsecond of
	// delay requested.
	us <<= 2; // x4 us, = 4 cycles

	// account for the time taken in the preceeding commands.
	// we just burned 19 (21) cycles above, remove 5, (5*4=20)
	// us is at least 8 so we can substract 5
	us -= 5; // = 2 cycles,

#elif F_CPU >= 12000000L
	// for the 12 MHz clock if somebody is working with USB

	// for a 1 microsecond delay, simply return.  the overhead
	// of the function call takes 14 (16) cycles, which is 1.5us
	if (us <= 1) return; //  = 3 cycles, (4 when true)

	// the following loop takes 1/3 of a microsecond (4 cycles)
	// per iteration, so execute it three times for each microsecond of
	// delay requested.
	us = (us << 1) + us; // x3 us, = 5 cycles

	// account for the time taken in the preceeding commands.
	// we just burned 20 (22) cycles above, remove 5, (5*4=20)
	// us is at least 6 so we can substract 5
	us -= 5; //2 cycles

#elif F_CPU >= 8000000L
	// for the 8 MHz internal clock

	// for a 1 and 2 microsecond delay, simply return.  the overhead
	// of the function call takes 14 (16) cycles, which is 2us
	if (us <= 2) return; //  = 3 cycles, (4 when true)

	// the following loop takes 1/2 of a microsecond (4 cycles)
	// per iteration, so execute it twice for each microsecond of
	// delay requested.
	us <<= 1; //x2 us, = 2 cycles

	// account for the time taken in the preceeding commands.
	// we just burned 17 (19) cycles above, remove 4, (4*4=16)
	// us is at least 6 so we can substract 4
	us -= 4; // = 2 cycles

#else
	// for the 1 MHz internal clock (default settings for common Atmega microcontrollers)

	// the overhead of the function calls is 14 (16) cycles
	if (us <= 16) return; //= 3 cycles, (4 when true)
	if (us <= 25) return; //= 3 cycles, (4 when true), (must be at least 25 if we want to substract 22)

	// compensate for the time taken by the preceeding and next commands (about 22 cycles)
	us -= 22; // = 2 cycles
	// the following loop takes 4 microseconds (4 cycles)
	// per iteration, so execute it us/4 times
	// us is at least 4, divided by 4 gives us 1 (no zero delay bug)
	us >>= 2; // us div 4, = 4 cycles
	

#endif

	// busy wait
	__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles in Atmel AVR or 1 cycle in LGT MCU
		_ONENOP_
		_MORENOP_
		"brne 1b" : "=w" (us) : "0" (us) // 2 cycles
	);
	// return = 4 cycles
}
`

LaZsolt, 2020-03-15:
[mybetter_delayMicroseconds.c.txt](https://github.com/dbuezas/lgt8fx/files/4334849/mybetter_delayMicroseconds.c.txt)

LaZsolt, 2020-05-05:
If you don't want to delay zero time and every timing are constans (not variable), this most compact and precise LGT8Fx specific code is for you. (Must be in Arduino.h) Link time optimization (LTO) does not matter.
```
#define delayMicroseconds(us)            \
(__extension__({                         \
  __asm__ __volatile__ (                 \
    "usL_%=:" "sbiw %0,1" "\n\t"         \
              "brne usL_%="              \
	          :  /* no outputs */    \
              :"w" ((uint16_t) ((F_CPU/3000000L) * (uint16_t)us))  \
  );                                     \
}))
```

seisfeld, 2020-05-05:
Would this work for any selected speed? I'm asking because of the `F_CPU/3000000L`.

LaZsolt, 2020-05-05:
Yes I think. I did some compilation with different selected speeds and 
code disassembling. Timing values seems correct, but I can't measure.
Tested on 16 MHz with DS16B20 temperature sensor.

2020.05.05. 15:05 keltezéssel, Stephan Eisfeld írta:

LaZsolt, 2020-05-05:
When I write timing are constans, I mean compiler calculate ( F_CPU/3000000L * microseconds) like this example at 16 MHz:
```
    delayMicroseconds(3);

Compiled to:
 2b8:   0f e0           ldi     r24, 0x0F
 2ba:   10 e0           ldi     r25, 0x00
000002bc <usL_217>:
 2bc:   01 97           sbiw    r24, 0x01
 2be:   f1 f7           brne    .-4        ; 0x2bc <usL_217>

```
If you call delayMicroseconds() macro with variables, MCU will do necessary calculations. It cause more delay than load two registers, so delay would be not precise. Don't use my macro like this:
```
  uint16_t nnn=12;
  for( int i = 0; i < 118; i++) {
    delayMicroseconds(nnn);        // nnn is a variable
    nnn = 3.14*nnn+1;
  }  
```

seisfeld, 2020-05-05:
Allright, thank you for the explanation! Really appreciate it. :)

LaZsolt, 2020-05-06:
Now this code really precise.
If you don't want to delay zero time and every timing are constans (not variable), this most compact and precise LGT8Fx specific code is for you. (Must be in Arduino.h) Link time optimization (LTO) does not matter. (3000000L is the LGT8Fx specific value. The Atmel specific value is 4000000L)
```
#define delayMicroseconds(us)     \
(__extension__({                  \
  __asm__ __volatile__ (          \
    "usL_%=:" "sbiw %0,1" "\n\t"  \
              "brne usL_%="       \
	          :  /* no outputs */ \
              :"w" ( (uint16_t) ((F_CPU * (uint32_t) us) / 3000000L) )  \
  );                              \
}))
```

dbuezas, 2020-07-10:
Ill add the new delayMicroseconds version you posted first on the next version. Thanks for that!

dbuezas, 2020-07-21:
I forgot about this one in the last release.
I'll put it in in the next one

jayzakk, 2020-08-12:
Uhm... Is it just me, or....

Test.ino:

```
#include <Ethernet.h>
#include <EthernetUdp.h>

byte localMacAddress[] = { 0xDE, 0xAD, 0xC0, 0xA8, 0xB2, 0x0A }; // 192.168.178.10
byte* ip = localMacAddress+2;
EthernetUDP MySocket;

void setup() {
  Ethernet.begin(localMacAddress, IPAddress(ip[0], ip[1], ip[2], ip[3]));
  Ethernet.setSubnetMask(24);
  MySocket.begin(1234);

}

void loop() {
  
}
```

Compile error:
```
C:\Users\Oliver\AppData\Local\Temp\ccViAldi.ltrans0.ltrans.o: In function `EthernetClass::socketBeginMulticast(unsigned char, IPAddress, unsigned int) [clone .constprop.7]':

C:\devel\arduino\arduino-1.8.10\libraries\Ethernet\src/socket.cpp:170: undefined reference to `delayMicroseconds'

C:\Users\Oliver\AppData\Local\Temp\ccViAldi.ltrans0.ltrans.o: In function `socketBegin':

C:\devel\arduino\arduino-1.8.10\libraries\Ethernet\src/socket.cpp:106: undefined reference to `delayMicroseconds'

collect2.exe: error: ld returned 1 exit status

```

jayzakk, 2020-08-12:
( if anyone read my comment I just deleted - it's just too hot and I did a mistake ^^ )

LaZsolt, 2020-08-12:
If you want to try on 32 MHz, you must use my delayMicroseconds code.
I wrote two different type of delayMicroseconds. Which one you choose?
 - a macro type: https://github.com/dbuezas/lgt8fx/issues/18#issuecomment-624450770
 - a fuction type: https://github.com/dbuezas/lgt8fx/issues/18#issuecomment-599248628

jayzakk, 2020-08-12:
@LaZsolt , i used the macro version, but accidently put it into wiring.c instead Arduino.h.

LaZsolt, 2020-08-31:
@dbuezas, 
I found a library (Adafruit DHT temperature sensors) which using delayMicroseconds() with a variable parameter. This mean the macro version of delayMicroseconds() not fully compatibile. So don't put the macro version in next release.
 But I have a new idea with the old style function: void **__attribute__ ((noinline))** delayMicroseconds() { ... }
I am still testing it.

dbuezas, 2020-09-20:
with #51 merged, this can be closed, right?

LaZsolt, 2020-09-20:
Not yet. I would like to comment here my ideas for a while.

XGIACOMO, 2020-10-20:
ciao, well done for your job!! 
i'm using LGT8F with your libreries but i faced  problems due to lack of precision with microseconds in bit banging. 
(this is the bus that i'd like to use with LGT8F https://www.pjon.org/ )
are you planning to fix the microsecond issues or are you no longer pursuing it?
thank you very much
Giacomo

LaZsolt, 2020-10-26:
@XGIACOMO 
What type of precision problems have you faced?

XGIACOMO, 2020-10-26:
hy LaZsolt, thank you for your answer!
this is what i've got with a simple delayMicroseconds(50) sketch
values are so far from what they should be!
i'll try your new libraries. thank you, i'll keep you in touch!

[image]
[image]

LaZsolt, 2020-10-26:
Be aware digitalWrite() takes 2 to 5 microseconds.
Any port bitSet(), bitClear() much more faster than digitalWrite() but need to calculate its execution time when calculating pulse time.

dbuezas, 2020-10-26:
If you want to get as efficient, precise, and fast as possible at having something happening at regular intervals, there is a trick you can do using counters. It is the same as using interrupts but you just busy-wait for the counter to reach its target instead of consuming the 50+ cycles of the whole interrupt prelude, return & stuff. 

```c++
// setting up timer 1
// secPerSample means "seconds per sample"
// this will work up to a max wait of 2ms (i.e secPerSample=0.002)
void startCPUCounter(float secPerSample) {
  TCCR1A = 0;
  TCCR1B = (1 << WGM13) | (1 << WGM12)   // CTC mode, counts to ICR1
        | (1 << CS10); // prescaler set to 1
  ICR1 = secPerSample * F_CPU - 1;
 
  TCNT1 = 0;
  setBit(TIFR1, OCF1A);  // clear overflow bit
}

__attribute__((always_inline)) inline void delay(){
   loop_until_bit_is_set(TIFR1, OCF1A); // this can be off by at most 3 clock cycles, but error won't accumulate because the timer will keep counting
  TIFR1 = 255;  // setBit(TIFR1, OCF1A); is actually enough, but clearing all flags at once is quicker and I'm not using the other timer flags anyway.
}
```

And then you busy-wait for a very precise timing that doesn't accumulate error:

```c++
void setup(){
    startCPUCounter(1.0/1000000); // 1us cycle
    
}
void loop(){
  noInterrupts(); // if you don't do this, it will be a bit off some times, but error won't accumulate.
  for (int i = 0; i< 10000;i++){
    doTheThingThatNeedsToHappenAtVeryPreciseAndShortIntervals();
  }
  interrupts();
}
```

It is the trick I ended up using in the oscilloscope project to get the oscilloscope [here](https://github.com/dbuezas/arduino-web-oscilloscope/blob/master/arduino/src/fillBuffer.h#L5-L32) to go very fast even while handling multiple channels and checking for triggering conditions. Only there I'm using Timer3 and fiddle with prescalers to get to higher waiting intervals when necessary. Obvious in hindsight I did feel really clever about this.

XGIACOMO, 2020-10-27:
hy dbuezas, thank you for your answer!!
i'm building in my caravan a battery charger with a 15v powersupply were i work on the feedback regulation to create the target charge stages. i find lgt8f very powerfull due to integrated dac and Differential Amplifier with 32x gain: with this chip i don't need other external components, and it works very well!!
unfortunatly i need to comunicate with the central unit using a bitbang protocol, 
https://www.pjon.org/
the same that i'm using to connect all devices in my caravan, but i'm not able to make lgt8f compatible with the protocol becouse of different bits lenght.
in coming days i'll try new LaZsolt libraries. i hope to succede!
than you again

XGIACOMO, 2020-11-02:
great job @LaZsolt!!!!!!  much better than before ;-)

[image]

dbuezas, 2020-11-02:
It's about time we make a new release including all these improvements from @Lazsolt et al.

LaZsolt, 2020-11-03:
@XGIACOMO 
I am also happy with the better result.

LaZsolt, 2020-11-03:
@dbuezas 
I am still working on a clock tick precise verison of `delayMicroseconds()`, but most of the case not needed clock tick tight precisity. But on lower freq, like 1 MHz, could useful the clock tick precise timing.
This task is more complex than I thought at first and code is became complex too. (I found posts about to correcting the `delayMicroseconds()`, older than 15 years.)
I have ideas, half made codes, and a testing code, but none of them are good enough to publish yet.

dbuezas, 2020-11-03:
@LaZsolt that's awesome! 
Should I wait then?

LaZsolt, 2020-11-03:
@dbuezas only few days.
I think need to discuss about `delayMicroseconds()` code, before we will reject it. ;)

SuperUserNameMan, 2020-11-03:
@LaZsolt :
Since your still working on this, just throwing an idea that is bugging me since I discovered LGT8Fx boards : do you think it would be possible to make clock speed software defined so it would be possible to change the speed of the board at runtime and have delay(), delayMicroseconds(), millis(), micros(), Serial(), I2C, SPI timings etc etc to adapt themselves at runtime too ?

LaZsolt, 2020-11-04:
@SuperUserNameMan 
>do you think it would be possible to make clock speed software defined ... ?

I may can create a source for the case of runtime variable clock speeds, but the delay count calculations became more complex, so short timing will be more inaccurate. Better idea is, when caller will calculate before calling `delayMicroseconds()` or duplicate the bit-banging routine for two different clock speeds.
`delay()` is clock speed independent now, because it calling `micros()` which reads the timer. If timer set is correct, then `delay()` will measure the correct milliseconds.

LaZsolt, 2020-11-04:
@dbuezas 
```
#define delayMicroseconds(us)       \
    if (__builtin_constant_p(us)) { \
        delayMicroseconds_c(us);    \
    } else {                        \
        delayMicroseconds_v(us);    \
    }
#define delayMicroseconds_c(us) lgt8fx_delay_cycles((double)us*F_CPU/1000000)    // for constant case
 void   delayMicroseconds_v(unsigned int us) __attribute__ ((noinline));         // for variable case which is same as earlier
```
More sources in the coming days.

LaZsolt, 2020-11-06:
Just finished, but not tested yet: https://github.com/LaZsolt/delayMicroseconds/tree/master/for_LGT8F

LaZsolt, 2020-11-07:
Watch my new readme. https://github.com/LaZsolt/delayMicroseconds/tree/master/for_LGT8F

SuperUserNameMan, 2020-11-08:
@LaZsolt : 
I've seen your code includes a trick to prevent the linker from discarding a function you want to make available for debugging.

I don't know if it it could be useful to you, but if you want to protect some of your functions from compiler optimization, you can encapsulate them this way :

```` C++
#pragma GCC push_options
#pragma GCC optimize ("keep-static-functions") 
static void foo( int a )
{
  // code i want to protect from compiler optimization
}
#pragma GCC pop_options
````
You can also specify a level of optimization `O0`, `O1`, `O2`, `O3`, `Ofast`, or `Os` this way `#pragma GCC optimize ("O0", "keep-static-functions")`.

More info here : 
- https://gcc.gnu.org/onlinedocs/gcc/Function-Specific-Option-Pragmas.html 
- https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

LaZsolt, 2020-11-11:
Finished, tested.

LaZsolt, 2020-11-21:
An interesting trick:

This code will wait x*4 - 1 clocks cycle as we know.
```
__asm__ __volatile__ (
    "1: sbiw %0,1   \n\t"          // 1 cycle in LGT
    "   nop         \n\t"          // 1 cycle
    "   brne 1b"                   // 2 cycles ( 1 cycle when counter became 0 )
    : "=w" (x)                     // No outputs, but it is inform the compiler about a modified register
    : "0"  (x)
);
```
<br><br>But this code will wait x*4 clocks cycle exactly.
```
__asm__ __volatile__ (
    "1: sbiw %0,1   \n\t"          // 1 cycle in LGT
    "   breq .+0    \n\t"          // 1 cycle  ( 2 cycle when counter became 0 )
    "   brne 1b"                   // 2 cycles ( 1 cycle when counter became 0 )
    : "=w" (x)                     // No outputs, but it is inform the compiler about a modified register
    : "0"  (x)
);
```

This little trick, with same code size, can make a bit readabe delay calculations inside delayMicroseconds_v().

LaZsolt, 2021-01-26:
@SuperUserNameMan 

> do you think it would be possible to make clock speed software defined so it would be possible to change the speed of the board at runtime ... ?

Yes, I found a solution for `delayMicroseconds()`. The compiler may can select the best code for different clock speeds. The source not uploaded yet.

Other thing is, the delay calculations and the delaying cycle all written in assembly language, so it will avoid compiler or linker optimizations.

SuperUserNameMan, 2021-01-26:
@LaZsolt : sounds great !

LaZsolt, 2021-01-26:
When I tested, the compiler handled the different speeds.
```
namespace CPU32 {
  #undef  F_CPU
  #define F_CPU 32000000
  #include "delayMicroseconds.h"
}
namespace CPU16 {
  #undef  F_CPU
  #define F_CPU 16000000
  #include "delayMicroseconds.h"
}
namespace CPU4 {
  #undef  F_CPU
  #define F_CPU 4000000
  #include "delayMicroseconds.h"
}
```
<br>
But after I put the final source to the core (without namespace) I getting redefinition errors in every namespace.

SuperUserNameMan, 2021-01-26:
I haven't tried this code below, but i'm wondering if it would work ... and also, how much extra space it would take into Flash ... ?

````c++

#define LGT_CLK_PRESCALER_1_MHz 5
#define LGT_CLK_PRESCALER_2_MHz 4
#define LGT_CLK_PRESCALER_4_MHz 3
#define LGT_CLK_PRESCALER_8_MHz 2
#define LGT_CLK_PRESCALER_16_MHz 1
#define LGT_CLK_PRESCALER_32_MHz 0

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_1_MHz
#undef  F_CPU
#define F_CPU 1000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_2_MHz
#undef  F_CPU
#define F_CPU 2000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_4_MHz
#undef  F_CPU
#define F_CPU 4000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_8_MHz
#undef  F_CPU
#define F_CPU 8000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_16_MHz
#undef  F_CPU
#define F_CPU 16000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v
#define delayMicroseconds_v _delayMicroseconds_v_32_MHz
#undef  F_CPU
#define F_CPU 32000000
#include "delayMicroseconds.h"

#undef  delayMicroseconds_v

void (*delayMicroseconds_v)( unsigned int us ); // declare as function pointer

void lgt_change_clk_prescaler_during_runtime( uint8_t clk_prescaler )
{
	noInterrupts();

	CLKPR = 0x80; // unlock clock prescaler change
	CLKPR = clk_prescaler;

	switch( clk_prescaler )
	{
		case LGT_CLK_PRESCALER_1_MHz :
			delayMicroseconds_v = &_delayMicroseconds_v_1_MHz;
		break;

		case LGT_CLK_PRESCALER_2_MHz :
			delayMicroseconds_v = &_delayMicroseconds_v_2_MHz;
		break;

		case LGT_CLK_PRESCALER_4_MHz :
			delayMicroseconds_v = &_delayMicroseconds_v_4_MHz;
		break;

		case LGT_CLK_PRESCALER_8_MHz :
			delayMicroseconds_v = &_delayMicroseconds_v_8_MHz;
		break;

		case LGT_CLK_PRESCALER_32_MHz :
			delayMicroseconds_v = &_delayMicroseconds_v_32_MHz;
		break;

		default:
			delayMicroseconds_v = &_delayMicroseconds_v_16_MHz;
		break;

	}

	Interrupts();
}
````

----

Or, instead, maybe it's possible to read the clock prescaler register into `delayMicroseconds()` and do something like that :
```` C++
void delayMicroseconds( unsigned int us )
{
    // applies current clock speed prescaler to the given us delay :

    us >>= ( CLKPR & 0x0f );

    // then rewrite the rest of delayMicroseconds() as if it was for 32 MHz
    ...
    ...
}
````

But i guess it would be too difficult to write a generic and precise delayMicroseconds() this way ? :-/

SuperUserNameMan, 2021-01-31:
Oh ! I've just noticed I misread your message in my first answer that i've just deleted.

> When I tested, the compiler handled the different speeds.
> But after I put the final source into the core (without namespace) I getting redefinition errors in every namespaces.

So the namespace trick works if the delayMicroseconds() function is stored into a separate header. If that's jsut a matter of copy/pasting, I don't think it's a problem.

Your namespace trick is more elegant than what I proposed in the answer I deleted.

Thanks for the idea !

LaZsolt, 2021-01-31:
A very new delayMicroseconds() are finished and tested again.
This version has better accuracy than before. The lowest frequency for clock tick accurate delay is:
- 4 Mhz when the parameter is a variable ( or 3 Mhz when the resonator is 12 MHz on a Wavgat board)
- 1 MHz when the parameter is constant.

You may use it with different clock frequencies in one source. This example shows how to use it on the menu selected and the other frequencies together:
```
#define FCPUSAVE F_CPU
namespace CPU32 {
  #undef  F_CPU
  #define F_CPU 32000000
  #include "delayus.h"
}
namespace CPU1 {
  #undef  F_CPU
  #define F_CPU 1000000
  #include "delayus.h"
}
#undef  F_CPU
#define F_CPU FCPUSAVE

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  delayMicroseconds(3);            // Default speed from Arduino IDE
  // set speed to 32 MHz
  CPU32 :: delayMicroseconds(3);   // 32 MHz speed
  // set speed to 1 MHz
  CPU1  :: delayMicroseconds(3);   // 1 MHz speed
  // set speed back to the basic speed
}
```
The source is here: https://github.com/LaZsolt/delayMicroseconds/tree/master/for_LGT8F
How to install:
- Replace the old `delayMicroseconds()` definition in <b>Arduino.h</b> in your directory `C:\Users\__yourusername__\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.5\cores\lgt8f\` with all my <b>Arduino.h</b> source.
- Edit your <b>wiring.c</b> and delete `delayMicroseconds()` function.
- Copy my <b>delayus.h</b> into your skech directory.

<br><br>
The previous attept was failed because I wanted to define `delayMicroseconds()` by a macro, but the compiler found it too complex when I tried to compile in different namespaces. Perhaps I run into a compiler issue. The macro which was bad I mentioned earlier: https://github.com/dbuezas/lgt8fx/issues/18#issuecomment-721539223

closed 2021-02-08 by LaZsolt

DurandA, 2021-02-26:
@LaZsolt sorry to hijack this issue. I am trying to port waiting functions from the [AVR Transistortester](https://www.mikrocontroller.net/articles/AVR_Transistortester) (in [`wait.S`](https://github.com/DurandA/transistor-tester-lgt328p/blob/lgt328p/wait.S)) to the LGT8F328P. Do you know if these figures are still valid with the LGT8F328P?

 *  rcall needs 3 clock cycles
 *  ret needs 4 clock cycles

SuperUserNameMan, 2021-02-26:
@DurandA : according to chinese datasheet, rcall needs 1 cycle, and ret 2 cycles.

See page 252 : https://raw.githubusercontent.com/dbuezas/lgt8fx/master/docs/LGT8FX8P_databook_v1.0.4.ch.pdf

DurandA, 2021-02-26:
@SuperUserNameMan Thank you very much. I missed it in the translated datasheet.