dbuezas/lgt8fx #19: could you tell me how to manually add supporting of lgt8f328d support?

youxiaojie, 2020-04-19:
because of GFW, the raw.githubusercontent.com was banned.
 C:\Users\Allan>ping raw.githubusercontent.com
Ping 请求找不到主机 raw.githubusercontent.com。请检查该名称，然后重试。
so how to use downloaded v1.05 release manully? thanks.

youxiaojie, 2020-04-20:
got it! 
Manually install:
1.download package_lgt8fx_index.json
2.palce it into C:\Users\Allan\AppData\Local\Arduino15
3.doanload hw support release file from https://github.com/dbuezas/lgt8fx/releases
4.uppack it to C:\Users\Allan\AppData\Local\Arduino15\staging\packages
5.click install in board manager. or unpack hw support file to C:\Users\Allan\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\<version>\
here <version> is like 1.0.5
6 modify C:\Users\Allan\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\<version>\boards.txt

adding these lines:

328.menu.variant.modelD=328D-B19200 (BTE17-11 19200 Baud)
328.menu.variant.modelD.bootloader.path=lgt8fx8e
328.menu.variant.modelD.bootloader.file=lgt8fx8e/optiboot_lgt8f328d.hex
328.menu.variant.modelD.build.variant=lgt8fx8e
328.upload.speed=19200

alpharesearch, 2020-06-05:
I ordered some "Acxico 5Pcs 5V 16M LGT8F328P-LQFP32 MiniEVB for Arduino Nano V3.0 HT42B534-1 Replace ATMeag328P" here in the USA on Amazon with the same 19200 baud rate requirements. I guess I could update the boot loader but it would be nice if there were some baud rate drop-down settings like on other boards.

youxiaojie, 2020-06-06:
you can try different the baud rate, may be the boot loader is custom.my 19200 is a nonstandard bootloader viariant.  you can try 115200 78600 etc

alpharesearch, 2020-06-06:
Thanks youxiaojie for your solution. The download works with your settings. Looks like I have the same BTE17-11 board like you. I cannot get my board to work with 32 Mhz, do you have any problems like this?

youxiaojie, 2020-06-07:
it seem that the 16Mhz version is the outdated 328d, equally to mega 168 though marked with "328", only with 16Mhz, 32Mhz does not work.

dwillmore, 2020-06-09:
If any of you with variant boards would be so kind as to take some nice pictures of them and include a link to where you got them, I'll add it to the wiki so that others can benefit.

youxiaojie, 2020-06-10:
agree!

alpharesearch, 2020-06-10:
Here is the link I bought it from:
https://www.amazon.com/dp/B0837GNYY2/
And here is the photo:
https://imgur.com/gallery/a7dQ5R0
https://imgur.com/gallery/QcfuSw5

youxiaojie, 2020-06-18:
from here:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.1c932e8dKugRs3&id=575845863557&_u=igtqrcba22

dbuezas, 2020-06-25:
Interesting, I wasn't aware of the "nano style" board with the 328d variant. I guess I've been lucky and always got 328p's.
BTW, as @youxiaojie said, this variant cant handle 32MHz even though it has a 32MHz internal clock.
From the datasheet:
>LGT8FX8D maximum operating frequency of 20MHz, so when the choice of internal 32MHz RC as the main clock source, be sure to
Make sure CLKPR is set to the correct frequency division (minimum 2).

dbuezas, 2020-07-21:
I'll welcome a PR with the added support for this board

dwillmore, 2023-01-12:
Does this issue require further effort?

LaZsolt, 2023-01-12:
I my opinion no need futher effort.
With the new board selection menu in case of selecting generic board ( `LGT8F328 + 16 MHz crystal` ) the upload speed selection available.

dwillmore, 2023-01-12:
Okay, that's that I had concluded, but I hadn't been a participant to this converstaion, so I didn't feel right in closing it.  I'll do that now.

closed 2023-01-30 by LaZsolt