dbuezas/lgt8fx #21: Support for high current pin configuration

rokweom, 2020-05-04:
According to the datasheet (page 81 in v1.0.5) some pins can be set to a a high current push-pull configuration, allowing for 80mA, instead of the standard 12/30mA. This can be done by setting a register. Any plans to add support for that? It would be useful for stuff like driving mosfets.

dbuezas, 2020-07-09:
You are right, this is quite interesting. I would rather provide an example sketch though. If you have a tested one, I'll welcome a PR! :)

rokweom, 2020-07-10:
Actually, my coding skills are really basic and I have no idea how to do it. That's why I was hoping for a simple solution, like `pinMode(pin, OUTPUT_HIGHCURR)`.
There's also a matter of testing if this mode actually works. Question is, how it can be done? Measure pin impedance? Draw high current and see if the pin dies? I'm really not sure.

dbuezas, 2020-07-11:
I may give this a try today. There are only 3 usable pins available though:
Pins D2, D5 and D6. 

Also D1 but if you connect it to something else, it may interfere with serial communication (including uploading).

There are two more (E4 and E5), but these are typically left floating in the breakout boards. You could solder wires directly to the chip though.

dbuezas, 2020-07-11:
I just tried it on pin D2, it works fine:
```c++
void setup() {
  pinMode(D2, OUTPUT);
  digitalWrite(D2, HIGH);  
  HDR |= 1 << 3;  // Enable high current D2  
}
```

For the other pins:
```c++
  HDR |= 1 << 0;  // Enable high current D5
  HDR |= 1 << 1;  // Enable high current D6
  HDR |= 1 << 2;  // Enable high current D1
  HDR |= 1 << 3;  // Enable high current D2
  HDR |= 1 << 4;  // Enable high current E4
  HDR |= 1 << 5;  // Enable high current E5
```

And of course you can combine all bits into 1 line if you need more than one:
```
   HDR = 0b00000011; // enable high current for D5 and D6 and disable for the rest
```

Thanks for the issue, I wasn't aware of this and it may come handy :)

rokweom, 2020-07-16:
I just tested it and can confirm PD5 and PD6 work correctly on LQFP32. What about PD2 tho? It works when you enable `pinMode(F2, OUTPUT); ` and `digitalWrite(F2, HIGH);` and select LQFP48 as the board, but not when you select LQFP32.

dbuezas, 2020-07-17:
Yes, those pins are not exposed in the LQFP32 core.
I dug a bit deeper into the difference of both cores last night and it
should be safe to use the LQFP48 core if you need 80mA on D1 and D2. Just
make sure not to digitalWrite(F2, HIGH) and digitalWrite(D1, LOW) at the
same time or you'll be shorting the pin internally.
David Buezas

closed 2020-07-21 by dbuezas

dbuezas, 2021-05-18:
the 80mAh drive is spectacular to drive logic level Mosfets w/o a driver at high speed. Unfortunately the only timer that has "convenient" 80mAh pins (in the QFP32) as output is timer0 (D5 and D6), which is the 8 bit timer with less modes and one looses millis() and delay().

Timer 3 can also be used (by enabling the alternate outputs to F1 and F2), but F1 and F2 are exposed as D1 and D2 which are the standard serial pins.