dbuezas/lgt8fx #22: In some edge cases, upload gets stuck

thegoodhen, 2020-06-11:
Hi, great work on the project!
I have encountered the following problem: when programming the board, the upload sometimes fails. From then on, the board is unresponsive. Plugging it out and in again and/or restarting the IDE does not work, either. What worked for me was to unplug it, plug it in again and then upload a different sketch from a different ide (OCROBOT005) . My board was Wemos TTGO XI.

I do believe the problem to be related to errors in serial communication caused by using baud rates that are too high, since the problem seemed to be much more noticeable with certain sketches and almost non-existent with other ones. Seems like an error occurs when a certain byte sequence appears in the stream? This is just a guess at this point; the problem is hard to reproduce, since it doesn't occur often. I'll try to look into it some more.

dbuezas, 2020-07-09:
That's weird, it never happened to me. I only have one single TTGO XI board though, and I always use the smaller boards for my projects so it may be board specific.
I see the test LED is connected to pin D1, which is also the TX line of the Serial. That is probably interfering, I don't like 
these boards... All the other (Wavgat, ssop20 mini, pro mini style and nano style) always work perfectly for me.

thegoodhen, 2020-07-10:
Hello, thank you for your reply. I do not believe the thing you described to be the issue, though. While it does appear that the LED is connected to TX, I do not believe it is. The board silkscreen is (probably?) messed up and 2 pins are labeled as D1; however, they aren't internally connected. Instead, the D12 pin (which the LED is connected to) is connected to the "non-serial" pin D1 for some reason. the LED also doesn't blink during the upload. The problem might still be on the side of the TTGO, rather than in the software, though!

closed 2020-07-18 by dbuezas