dbuezas/lgt8fx #23: Incorrectly named voltage reference

thegoodhen, 2020-06-27:
Hello, the internal 2.048 voltage reference is incorrectly named as INTERNAL2V56 .

LaZsolt, 2020-06-29:
According to database LGT8Fx8D series MCU has internal 1.25V / 2.56V ± 1% calibrated reference voltage source.

dbuezas, 2020-07-09:
Yes, you are right @thegoodhen and also partially @LaZsolt.
According to the datasheet, the available internal references are:
- 1.024V
- 2.048V
- 4.096V

But the core has these constants available:
- #define INTERNAL 3
_- #define INTERNAL1V25 3_
- #define INTERNAL2V56 2
_- #define INTERNAL1V024 3_
- #define INTERNAL2V048 2
- #define INTERNAL4V096 4
[here](https://github.com/dbuezas/lgt8fx/blob/db47042686b2c9455a7690ca98712c6e8514871c/lgt8f/cores/lgt8f/Arduino.h#L82-L87)

The ones in _italic_ are just aliases of the nearest ones. They were there in the original core this one derives from so I assume they are there for compatibility (as they are kinda close).

Thanks for the issue anyway :)

closed 2020-07-21 by dbuezas