dbuezas/lgt8fx #24: SSOP20 unable to do anything with SD Card

Oxmaster, 2020-07-10:
I have the standard Pro Mini clone with SSOP20 chip, I removed the LDO and use cp2102 as a programmer. I tried INA3221 that communicates with I2C and it worked. Now I'm trying SPI with SD card and it's not working. I'm using Arduino IDE 1.8.13 and LGT8Fx boards 1.0.5. I tried the standard SD library and the SDFat, at 8MHz and 32MHz with chip select pins 8, 9 and 4. To be sure I used my Nano clone with old bootloader and the SD card reader and the library work properly.

Am I running a standard sd card info test without any modifications.

rokweom, 2020-07-11:
> I tried the standard SD library and the SDFat, at 8MHz and 32MHz 
Have you tried it at 16MHz? I found that some libraries don't work properly at 32 MHz.

Oxmaster, 2020-07-12:
> > I tried the standard SD library and the SDFat, at 8MHz and 32MHz
> 
> Have you tried it at 16MHz? I found that some libraries don't work properly at 32 MHz.

Just tried it and it still won't init the card. I'm will see if I have anything other SPI related in my toolbox

dbuezas, 2020-07-12:
Hi @Oxmaster,

It was reported by @LaZsolt that the new boards with ssop20 now use the 328P chip (not D). I think he was having the same problem as you do, given what he wrote [here](https://github.com/dbuezas/lgt8fx/pull/16/files#diff-09274b0105a7570ceaf63c65941dc9a5R76)

 I haven't had the opportunity yet to test and build his pr, but in the mean time, you could try the `328p-LQFP32 (e.g. MiniEVB nano-style)` variant in the menu and see if SPI works. 
If you try it, please report your results back so I take that into account for the next version.

Cheers!

Oxmaster, 2020-07-12:
Just tried SSD1306 SPI version and it works on 32MHz, 16Mhz, 8Mhz and 1Mhz (changed the CLK pin from 10 to 7 as there is conflict). As far as I know the SSD1306 library from Adafruit uses software SPI and the SD card libs use hardware SPI.

After that I tried the SDFat library at 32MHz on software SPI and it worked.

Also tried building and uploading 328p-LQFP32 (e.g. MiniEVB nano-style) but after the upload I got nothing on serial and when trying to program for the SSOP20 (or any other) I'm getting information from avrdude that ```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d
```
I can open the serial monitor, but can't upload any new sketches. Not sure if that bricked the LGT8F.

dbuezas, 2020-07-12:
Oh no! really? I'm so sorry for that!
I have used the "wrong" variant many times and I never had this kind of issues!
Manually resetting with the button while uploading doesn't help either?

If you have another board (also a normal arduino), you can try reuploading the bootloader with the method in [here](https://github.com/dbuezas/lgt8fx/issues/13#issuecomment-656392739)

Oxmaster, 2020-07-12:
After some trying with reseting during programing (a lot of tries), I manged to upload blink sketch. It seems like my DTR doesn't want to work, so I will try uploading a new bootloader. But before that I tried to upload blink sketch with some serial messages for the LQFP32 and they don't work but the LED still blinks.

LaZsolt, 2020-07-12:
LGT8F328P SPI has more features than ATmega328P SPI. One of these features is 4 byte input and 4 byte output buffer. (Translated databook v1.0.5 page 195: -Input/output both have 4 buffer register.) If you won't handle this buffer, SPI will not work as expected. Your SD library will not work if not using the **lgt8f package SPI library**.
Perhaps Arduino IDE not including SPI library from lgt8f package?

In my project I using SPI for ENC28J60 ethernet adapter with EtherCard library which not using SPI library. So I modified a bit EtherCard_enc28j60.cpp.

LaZsolt, 2020-07-12:
If you use 328p-LQFP32 (e.g. MiniEVB nano-style) variant in the menu with SSOP20, serial communication will not work because of wrong pin redirection.
You must select SSOP20 from the menu for right pin redirecion of RXD, TXD and SPI SS.

Oxmaster, 2020-07-12:
It seems like it uses the right SPI library. I'm using SDFat library https://github.com/greiman/SdFat

```F:\Programs\Arduino\arduino-builder -dump-prefs -logger=machine -hardware F:\Programs\Arduino\hardware -hardware C:\Users\me\AppData\Local\Arduino15\packages -tools F:\Programs\Arduino\tools-builder -tools F:\Programs\Arduino\hardware\tools\avr -tools C:\Users\me\AppData\Local\Arduino15\packages -built-in-libraries F:\Programs\Arduino\libraries -libraries D:\Google Drive\programers\Arduino\libraries -fqbn=LGT8fx Boards:avr:328:clock_source=internal,clock=32MHz,variant=modelD_SSOP20 -ide-version=10813 -build-path C:\Users\me\AppData\Local\Temp\arduino_build_736571 -warnings=all -build-cache C:\Users\me\AppData\Local\Temp\arduino_cache_527224 -prefs=build.warn_data_percentage=75 -verbose C:\Users\me\AppData\Local\Temp\arduino_modified_sketch_500283\SdInfo.ino
F:\Programs\Arduino\arduino-builder -compile -logger=machine -hardware F:\Programs\Arduino\hardware -hardware C:\Users\me\AppData\Local\Arduino15\packages -tools F:\Programs\Arduino\tools-builder -tools F:\Programs\Arduino\hardware\tools\avr -tools C:\Users\me\AppData\Local\Arduino15\packages -built-in-libraries F:\Programs\Arduino\libraries -libraries D:\Google Drive\programers\Arduino\libraries -fqbn=LGT8fx Boards:avr:328:clock_source=internal,clock=32MHz,variant=modelD_SSOP20 -ide-version=10813 -build-path C:\Users\me\AppData\Local\Temp\arduino_build_736571 -warnings=all -build-cache C:\Users\me\AppData\Local\Temp\arduino_cache_527224 -prefs=build.warn_data_percentage=75 -verbose C:\Users\me\AppData\Local\Temp\arduino_modified_sketch_500283\SdInfo.ino
Using board '328' from platform in folder: C:\Users\me\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.5
Using core 'lgt8f' from platform in folder: C:\Users\me\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.5
Detecting libraries used...
"F:\\Programs\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DCLOCK_SOURCE=1 -DF_CPU=32000000L -DARDUINO=10813 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\cores\\lgt8f" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20" "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_736571\\sketch\\SdInfo.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
"F:\\Programs\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DCLOCK_SOURCE=1 -DF_CPU=32000000L -DARDUINO=10813 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\cores\\lgt8f" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\libraries\\SPI\\src" "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_736571\\sketch\\SdInfo.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for SdFat.h: [SdFat@1.1.4]
ResolveLibrary(SdFat.h)
  -> candidates: [SdFat@1.1.4]
"F:\\Programs\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DCLOCK_SOURCE=1 -DF_CPU=32000000L -DARDUINO=10813 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\cores\\lgt8f" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\libraries\\SPI\\src" "-ID:\\Google Drive\\programers\\Arduino\\libraries\\SdFat\\src" "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_736571\\sketch\\SdInfo.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"F:\\Programs\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DCLOCK_SOURCE=1 -DF_CPU=32000000L -DARDUINO=10813 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\cores\\lgt8f" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\libraries\\SPI\\src" "-ID:\\Google Drive\\programers\\Arduino\\libraries\\SdFat\\src" "C:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\libraries\\SPI\\src\\SPI.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"F:\\Programs\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DCLOCK_SOURCE=1 -DF_CPU=32000000L -DARDUINO=10813 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR [... cut ...] are\\avr\\1.0.5\\cores\\lgt8f" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20" "-IC:\\Users\\me\\AppData\\Local\\Arduino15\\packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\libraries\\SPI\\src" "-ID:\\Google Drive\\programers\\Arduino\\libraries\\SdFat\\src" "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_736571\\sketch\\SdInfo.ino.cpp" -o "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_736571\\sketch\\SdInfo.ino.cpp.o"
```

LaZsolt, 2020-07-12:
Wrong variant: "...packages\\LGT8fx Boards\\hardware\\avr\\1.0.5\\variants\\lgt8fx8ds20"
The good variant: lgt8fx8ps20

Replace C:\Users\me\AppData\Local\Arduino15\packages\LGT8fx Boards\\hardware\avr\1.0.5\boards.txt with this file: https://github.com/LaZsolt/lgt8fx/blob/master/lgt8f/boards.txt

Then reload Arduino IDE, try a new upload.

Oxmaster, 2020-07-13:
Close but no cigar. The standard SD library seem to work at 32MHz/16MHz, although the CardInfo example sketch doesn't seem to be able to read the internal structure of the card and when using SD.begin SPI speed has to be explicitly set `SD.begin(SPI_FULL_SPEED, chipSelect)`

The SDFat library still won't work though.

Oxmaster, 2020-07-15:
I've soldered some header pins to my LGT8F-328P-LQFP32 (MiniEVB nano) and at 32MHz the standard SD library works, it can read the internal structure of the SD card, but the SDFat still isn't working; can't init the card.

dbuezas, 2020-07-15:
Thanks for the updates, keep them coming :)
I may try this myself soon if I find the time, I'll let you know if I find something out

LaZsolt, 2020-07-15:
I think Greiman's SdFat unable to recognize LGT boards. Check this file:
https://github.com/greiman/SdFat/blob/master/src/SpiDriver/boards/GpioPinMap.h

Oxmaster, 2020-07-15:
I'm testing the SDFat library because I've heard that the included SD library is not the best and in my next project I wanted to use the LGT8F to check it out as there is not much into about these boards.

TL;DR of what I tried so far that doesn't work (with SSOP20 version): on everything default (boards.ini) nothing worked; different SPI speeds and clock speeds. 
With modified boards file (change of 328D to 328P) the standard SD library is not working fully (CardInfo sketch): begin function has to have explicitly set the SPI speed and the sketch can't read the structure of the card itself, but can write and read data while the SDFat library is not working with any settings. 
Seems like the clock and SPI speed doesn't matter.

LaZsolt, 2020-07-16:
You may try SdFat-beta, because:

 greiman Major internal restructuring
Latest commit 172c685 on Dec 6 2019
History

**Added features** for SD chip select on port expanders and **external custom SPI drivers**.

dbuezas, 2020-07-19:
try 328p-ssop20 in the new core v1.0.6 (note the p at the end)

Oxmaster, 2020-07-20:
The standard library seems to work fully now, it can read the sectors. The SdFat main branch is not working at all and SDFatBeta branch can init the card but can't do anything with it. Now that the standard library works, I will try to make something usefull instead of using the CardInfo sketch to check if it's working for sure.

dbuezas, 2020-07-20:
Sounds like an improvement!
That's with the new v1.0.6 core, right? you can thank @LaZsolt for that

closed 2020-07-21 by dbuezas