dbuezas/lgt8fx #25: 'A9' was not declared in this scope

Hcreak, 2020-07-17:
I want to using Pin A9、Pin A10 and Pin A11 on LGT8F328P-LQFP32, Do you help me? Thanks.

dbuezas, 2020-07-17:
Pins A9 and A11 are not exposed in the LQFP32 chip. You should be able to use A10 (shares function with AREF) if you use the LQFP48 core without problems.

Hcreak, 2020-07-20:
[image]
In datasheet, I have found  Pin PE4、Pin PE5 and Pin PE6 are exposed in the LQFP32 chip, Do it have no mapping Arduino pins？Thanks for your answer.

dbuezas, 2020-07-20:
As you noticed, these are typically not exposed in the breakout boards. Also ADC10 is labelled as REF in the silk screen of the LQFP32 boards. Furthermore, you can refer to D1 and D2 as F1 and F2 if you want to use them as high current pins, but D1 is also TX, so if you use Serial communication while also setting F1 as output you'll be shorting those outputs and likely destroy the board.

More advanced users can upload using the LQFP48 core (in the menu), solder cables to PE4 and PE5, and do all the other things, but I'm afraid this whole business should stay "hidden" or it will end up confusing the typical user.

What's your opinion?

dbuezas, 2020-07-20:
This comment in issue 21 may interest you, it has more info on PE4 and PE5: https://github.com/dbuezas/lgt8fx/issues/21#issuecomment-657020605

Hcreak, 2020-07-21:
Thanks. I will compile with the LQFP48 core and upload it into LQFP32 core for testing. Looking forward to good results.

dbuezas, 2020-07-21:
Looking forward to your results. I'll close this issue now, but please post how that worked out for you here for others to use it :)

closed 2020-07-21 by dbuezas