dbuezas/lgt8fx #30: LGT8F Lock Bits

WGR7, 2020-07-25:
Hi
Can this ISP burn the lock bits? Is the lgt8f lock bits the same as ATMega? Do  read out protection work by the same manner as ATMega?

Regards

dbuezas, 2020-07-31:
I couldn't find anything on lock bits for this board in the datasheet

WGR7, 2020-08-02:
I guess the LGTMix can program the Readout protection. I've found on file of Migrating from Atmega to LGT8F88 that says that they use a different aproach.

ftp://imall.iteadstudio.com/Microcontroller/IM131218002/Migrating_from_ATmega88_to_LGT8F88A_v1.0.pdf

dwillmore, 2023-01-12:
Is anyone is looking for a simple project, this issue may be what you're looking for.

LaZsolt, 2023-01-16:
Since LGT8Fx has no lock bits, it is not an issue.
I think it could be mentioned in the ~~documentation~~ readme, as a negative feature.

dbuezas, 2023-02-06:
I mentioned it in the readme now

closed 2023-02-06 by dbuezas