dbuezas/lgt8fx #35: Better/matching pinout in docs for LGT nano

jayzakk, 2020-07-28:
Is there anyone out there who would "finalize" this attached pinout graphics I collected from layout+data sheet for wiki/docs?

We do have so much more I/O and possibilities (more DACs, differential analogs, the timer#3, more GPIO, even debugger pins can be resused as GPIO,,...):

[image]

After seeing this possibilities, there may be some small changes needed to support all of that ^^

So, some graphics guy here?

jayzakk, 2020-07-29:
Rant to myself: Never ever reuse the PC6 pin, if you want to upload some different sketch later on.
Resetting into the bootloader for avrdude does not work anymore without some reset line. Surprise!

Next one. 9 left ;)
Removing the PC6/D27 definitions again.

dbuezas, 2020-07-29:
This is really good!

Did you check out these projects?
https://github.com/sparkfun/Graphical_Datasheets
https://github.com/stevenj/GenPinoutSVG

These are to generate the diagrams programatically from a CSV file. The format looks reasonable and the output looks quite good:
[image]

dbuezas, 2020-07-29:
> Rant to myself: Never ever reuse the PC6 pin, if you want to upload some different sketch later on.
> Resetting into the bootloader for avrdude does not work anymore without some reset line. Surprise!

Hahaha, that's hilarious! Time to try the ISP in the examples! :) just follow this: https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md
I had to add this hack to the menu, but I tested it to from one LGT to another and it does flash the bootloader just fine

dbuezas, 2020-07-29:
BTW, there are some pins that can output 80mA if you want to add that to the diagram
more info here: https://github.com/dbuezas/lgt8fx/issues/21#issuecomment-657020605

jayzakk, 2020-07-29:
Oookay, that generator was exactly what I was looking for, perfect!
The bricked now-only-80kHz-generator lgt has a problem: not only disabled I the reset line, but also the debugger. Same test sketch. I am good in creating such pitfalls :)
Yes, the 80ma outputs should be marked, too. Good Idea. 

Thanks!

dbuezas, 2020-07-30:
Oh, good thing these boards cost less than a soda. I killed 3. 2 of them I killed twice (bridged the burnt diode and then burned for good a month later 😅 )

Looking forward to your findings with the generators!

jayzakk, 2020-07-30:
I think this here should be ok. 
How does it find it's way into the wiki? ;)

[LGT8F328P-nano.zip](https://github.com/dbuezas/lgt8fx/files/5004009/LGT8F328P-nano.zip)

Btw @dbuezas : 
ISP programmer did un-brick the bricked LGT! I just had to keep it in reset state while powering on: nothing from my brick-code executes and the ports keep their connection to SWC/SWD/RESET, as the ISP is not related to the CPU core itself...

dbuezas, 2020-07-31:
Oh that's great news! I'll add an unbricking section and put your successful unbricking tips.

The pinout looks amazing 🥇 ! Do you mind if I add your GitHub hook as author and the url of this core to it?
And would you upload the csv also? Just in case some addition can be done in the future or make the breakout of other boards :)

dbuezas, 2020-08-01:
I'm using your graphical pinout all the time

jayzakk, 2020-08-02:
Here are all the files, including the ones for generator.
Please also check #36 , as it is needed to use A10.

[LGT8F_for_GenPinoutSVG.zip](https://github.com/dbuezas/lgt8fx/files/5012708/LGT8F_for_GenPinoutSVG.zip)

dbuezas, 2020-08-06:
Great! I'll add this to the repo and docs soon. Maybe others also help making pinouts of more boards based in your work 🎉

dbuezas, 2020-10-11:
I freaking love the pinout, I use it each time!

closed 2020-10-11 by dbuezas