dbuezas/lgt8fx #37: List of working-with-LGT8F / tested / patched libraries?

jayzakk, 2020-08-02:
Can we have some list of such libraries.

As of FastLed, this did not work out-of-the-box for clockless stripes due to different cycle timings of the CPU.
There wasn't a lot to change, and I created a PR for FastLED, but I know that they don't update/release that often.

So it would be nice to have a list of libraries someone already checked, and - if possible - an working/patched one with link to the fork.

If someone needs a working FastLED right now and/or likes to test it: https://github.com/jayzakk/FastLED/tree/lgt8f-support
That one works with 8,16 and 32MHz (slower prohibited by FastLED itself), checked the timings with my oscilloscope.

dwillmore, 2020-08-03:
Sounds like a good addition to the wiki?

jayzakk, 2020-08-05:
Can anyone add to the Wiki? I would like to add a couple of libs I already checked.
Again, I am such a GitHub noob ^^

dwillmore, 2020-08-05:
You bet.  David's can. :)  We can use this issue to track them.  Just post
them here and we'll see about getting it added to the wiki.

dbuezas, 2020-08-06:
Hi guys!
I love the idea! :)
Thanks @dwillmore for bringing the wiki up

@jayzakk you have now write access to the wiki, thanks for your really valuable contributions!

jayzakk, 2020-08-07:
Wow, thats cool. Thanks a lot for access!
Still in the checking-and-fixing libraries cycle and sending out PRs.
At the moment I checked/fixed FastLED, TFT.h for SPI ST7735 LCDs, Adafruit_GFX/ for SPI ST7735 LCDs, SSD1306 for I2C-OLEDs. Next will be W5500 SPI Ethernet, which could massively improve from the 4 bytes SPI buffer - if implemented ^^

I will think about a concept how to keep the wiki pages clean but still informative for the libraries things.

dwillmore, 2020-08-07:
Maybe just a table with a per library icon saying as of what version that
library supported lgt8fx?  If you want to get fancy, the library name could
be a link to the repo and the version number could be a link to the release
where it was first supported?

jayzakk, 2020-08-09:
I started with https://github.com/dbuezas/lgt8fx/wiki/Libraries-support
So I close this issue.

closed 2020-08-09 by jayzakk