dbuezas/lgt8fx #39: Missing definitions for non-standard interrupt register bit masks

thegoodhen, 2020-08-08:
Hello! Recently, I was trying to get the PinChangeInterrupt to work with the board, but it turns out that some defines are missing. Notably PCINT24, PCINT25... and so on are undefined. Thank you!!

jayzakk, 2020-08-09:
Good finding!

I first guessed the problem is at hardware/avr/1.0.6/boards.txt, it sets 328.build.mcu=atmega328p, which does not have the relevant defines.

My first idea was to change to 328pb.h, which has all those bits, but the vector definitions >25 are not compatible to LGT8F.

So we could do 2 things:
- create a correct io header file and patch avr-gcc  ^^ 
- try some (much quicker) workaround and add this pcints and vectors "later" (in hardware/avr/1.0.6/variants/lgt8fx8p/pins_arduino.h) - and hope we don't have weird side effects

I could try to check changes as lgt8fx8p. Or anyone else? ;)
Additionally, we'd need documentation for the new 3 vectors (PCINT3, PCINT4, TIMER3), especially as timer 3 does not have different vectors as all others (CAPT,COMPA,COMPB,OVF).

jayzakk, 2020-08-09:
I don't see the 3 missing vectors working by now, as they would need support in avr-gcc as some dedicated model we can set as build.mcu.

Currently used is the 328p (packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/lib/avr5/crtatmega328p.o), selected by the build.mcu in boards.txt. It does not have those vectors defined. I'd define using the 328p that as a "bug", but:
From informations I can find, the whole avr-gcc toolchain comes from Atmel, so they won't implement the LGT8F core.
Someone would need to patch and package a completely new toolchain (as ESP32 and others did). YMMV.

I did some code fixes for the digitalPinToPCMSK(), digitalPinToPCICRbit(), digitalPinToPCMSKbit(), digitalPinToPCICR(), as current ones don't work correctly with Ports E and F - but as we don't have a chance to make an ISR, there is no real need to PR that. Those new functions just would take some more bytes (additional PRGMEM table) without no real use.

thegoodhen, 2020-08-10:
> I don't see the 3 missing vectors working by now, as they would need support in avr-gcc as some dedicated model we can set as build.mcu.
>
> Currently used is the 328p [...]

Hey, sorry, I forgot to inform you about it, but I came to the same conclusions. Adding the ISRs just doesn't work, even if we set the vectors up correctly. 

If we don't have support in C, do you think it's possible to use low level assembly to service the ISRs? I am not really well-versed in stuff THIS low-level to be able to tell. 

If it's possible, could we perhaps add support for the remaining interrupt vectors on library level by coding parts in assembly? As in, we could make a library for Timer3, with the ISRs made in assembly to "bypass" the compiler. Similarly, we could just edit the Pcint library and add the interrupt handlers in assy.

jayzakk, 2020-08-10:
This is not possible:

The mcu specific .o file contains a segment "_vectors", in this case, exactly 26 of them.
The vector table starts at adress 0 of progmem (flash).

The compiler checks for each "ISR(x)", which is a macro for "__vector_x", and - if existant - adds a ISR stub for the vector and alters the vector table (beginning at adress 0, defined in mcu .o) at compile time to point to it.
The default value for each vector in the table is __bad_interrupt -> you already have seen that. It's just a "vector-without-defined-ISR" dummy.

Adress 0 is __vector_0 (the reset vector), then 25 more vectors each 4 byte (jmp commands) for each interrupt. 
The next address 0x68 is the first byte of the compile (as the mcu .o file defines). There is no room to insert 3 more vectors - especially: how should we tell the compiler? ;)

There MAY be one solution working with the IVBASE register. This allows to remap the vector table to some other space in PROGMEM (IVBASE*0x200 would be the starting address). We could steal some more bytes, add a non-relocatable segment at 0x200, with default jumps to the first 26 vectors (handled by compiler), and 3 own ISR stubs (hard to say how to handle them, we won't be able to reuse the ISR() macro for it).

Here some example with onlinedisassembler.com, easy to use and no playing around with command line arguments never working as you want:

[image]

You may see, that you never ever want to get an __vector_27, 28, 29 - there is some code executing you definately do not want to run ;)

thegoodhen, 2020-08-11:
Okay, thank you very much for your explanation. The workaround for my specific case was to use Timer0 interrupt to periodically check the buttons, instead of handling them using pin change interrupts. 

That being said, I think it might be useful to add the support for the other vectors, especially to support Timer3. 
I think I have a generic understanding of the problem now. If I am not mistaken, it would cost us 4*29=116 bytes of Progmem. That is less than 0.5% and is probably acceptable. We just need to introduce our own vector table, tell the linker where to put it and then initialize IVBASE to something else, right?

jayzakk, 2020-08-11:
If playing around with the linker, you could also just add a own 29 vectors segment at #0, and add .text (starting with the "original" vector table) after it. Then you would even not need to change IVBASE ( `Wl,--section-start=.myvectors=0 Wl,--section-start=.text=0x74`). Is it even possible to add own LDFLAGS? 

And, how to implement the `ISR()` macro functionality? I tried around and don't get it how they did it.
ISR() globally exports its function as `extern "C" void __vector_## (void) __attribute__ ((signal,used, externally_visible))` and some "magic" puts it into the vector table.

I had to link manually, to get the 3 new vectors working with hardcoded targets. That can't be a solution ^^

thegoodhen, 2020-08-11:
Oof. Well, to implement the equivalent of ISR(), maybe we could use stub C functions, placed on a precise address, that would then just call a user-supplied function by the virtue of a function pointer if such a function exists and do nothing if it doesn't. 

Do you think it's even possible to force code on segment #0 when this is where the vector table should be?
I mean, it would get rid of the "double jumping" of your previously suggested approach, which seems a bit problematic, as that delays the isr execution significantly.

jayzakk, 2020-08-11:
First, this approach should only be taken, if someone really needs those ISRs (larger codebases - we already have quite a lot of missing flash space - and slightly slower ISRs).
Second, it's not THAT much slower. Just that single JMP doesn't add much to the ISR-push-all-registers-and-flags-and-restore-them-afterwards; feel free to count cycles:

```
.text:0000040c <__vector_19>:
                           .text:0000040c 1f 92                            push	r1
                           .text:0000040e 0f 92                            push	r0
                           .text:00000410 0f b6                            in	r0, 0x3f	; 63
                           .text:00000412 0f 92                            push	r0
                           .text:00000414 11 24                            eor	r1, r1
                           .text:00000416 2f 93                            push	r18
                           .text:00000418 3f 93                            push	r19
                           .text:0000041a 4f 93                            push	r20
                           .text:0000041c 5f 93                            push	r21
                           .text:0000041e 6f 93                            push	r22
                           .text:00000420 7f 93                            push	r23
                           .text:00000422 8f 93                            push	r24
                           .text:00000424 9f 93                            push	r25
                           .text:00000426 af 93                            push	r26
                           .text:00000428 bf 93                            push	r27
                           .text:0000042a ef 93                            push	r30
                           .text:0000042c ff 93                            push	r31
                           .text:0000042e 83 e3                            ldi	r24, 0x33	; 51
                           .text:00000430 91 e0                            ldi	r25, 0x01	; 1
                           .text:00000432 0e 94 1d 01                      call	0x23a	;  0x0000023a <_ZN14HardwareSerial17_tx_udr_empty_irqEv>
                           .text:00000436 ff 91                            pop	r31
                           .text:00000438 ef 91                            pop	r30
                           .text:0000043a bf 91                            pop	r27
                           .text:0000043c af 91                            pop	r26
                           .text:0000043e 9f 91                            pop	r25
                           .text:00000440 8f 91                            pop	r24
                           .text:00000442 7f 91                            pop	r23
                           .text:00000444 6f 91                            pop	r22
                           .text:00000446 5f 91                            pop	r21
                           .text:00000448 4f 91                            pop	r20
                           .text:0000044a 3f 91                            pop	r19
                           .text:0000044c 2f 91                            pop	r18
                           .text:0000044e 0f 90                            pop	r0
                           .text:00000450 0f be                            out	0x3f, r0	; 63
                           .text:00000452 0f 90                            pop	r0
                           .text:00000454 1f 90                            pop	r1
                           .text:00000456 18 95                            reti
```

The "ISR()" adds push&pop-code (except you attribute it "naked", where you have to take care yourself of used flag + registers).

thegoodhen, 2020-08-11:
My bad, I forgot about the context switch. In that case, it's irrelevant.  So... is the best approach to just ignore it for the Arduino core?

jayzakk, 2020-08-11:
I'm currently looking for some different option: linker with `-nostartfiles` to suppress of all those mcu specific stuff of `328p.o`, and add some own `lgt8f.S` file including the missing vectors+startup to link at `.vectors` and `.text:__ctors`

There must be a CLEAN solution to that.

Next try would be to ask the guys at mikrocontroller.net, there are quite a lot of ppl with deep understanding of avr-gcc.

jayzakk, 2020-08-11:
Damn, I got it :-)

It sooo simple, now that I learned that lot about this linker and the segments avr-gcc uses.
We do NOT need special (jump) tables.
We do NOT need special linker options.
We just need a patch here and there.

Only downside is, that we can not jump into BADISR by default, it will jump to 0 (RESET) if such interrupt occurs. There is some gcc limitation in weak linkage I run into. But as the current situation is no different, I won't care about that.

I did not try if it actually is WORKING, but it should (after initializing PCIFR/CR,...).

Test .ino:
```
void setup() {
  Serial.begin(9600);
  Serial.println("Start");
  }

ISR (PCINT3_vect) { 
  Serial.println("26"); 
}

void loop() {
}
```

Disassembles into:
```
.text:00000000 <__heap_end>:
                           .text:00000000 0c 94 3b 00                      jmp	0x76	;  0x00000076 <__ctors_end>
                           .text:00000004 0c 94 63 00                      jmp	0xc6	;  0x000000c6 <__vector_20>
                           .text:00000008 0c 94 63 00                      jmp	0xc6	;  0x000000c6 <__vector_20>
[... vectors, originally the last one (25) is at 0x0064, but we have 3 more now ...]
                           .text:00000060 0c 94 63 00                      jmp	0xc6	;  0x000000c6 <__vector_20>
                           .text:00000064 0c 94 63 00                      jmp	0xc6	;  0x000000c6 <__vector_20>
                           .text:00000068 0c 94 26 02                      jmp	0x44c	;  0x0000044c <__vector_26>
                           .text:0000006c 0c 94 00 00                      jmp	0	;  0x00000000 <__heap_end>
                           .text:00000070 0c 94 00 00                      jmp	0	;  0x00000000 <__heap_end>

.text:00000074 <__ctors_start>:
                           .text:00000074 f5 02                            muls	r31, r21

.text:00000076 <__ctors_end>:
                           .text:00000076 11 24                            eor	r1, r1
                           .text:00000078 1f be                            out	0x3f, r1	; 63
                           .text:0000007a cf ef                            ldi	r28, 0xFF	; 255
[... all of the initcode is there, as is main. and our isr...]

.text:0000044c <__vector_26>:
                           .text:0000044c 1f 92                            push	r1
                           .text:0000044e 0f 92                            push	r0
                           .text:00000450 0f b6                            in	r0, 0x3f	; 63
                           .text:00000452 0f 92                            push	r0
                           .text:00000454 11 24                            eor	r1, r1
                           .text:00000456 2f 93                            push	r18
                           .text:00000458 3f 93                            push	r19
                           .text:0000045a 4f 93                            push	r20
                           .text:0000045c 5f 93                            push	r21
                           .text:0000045e 6f 93                            push	r22
                           .text:00000460 7f 93                            push	r23
                           .text:00000462 8f 93                            push	r24
                           .text:00000464 9f 93                            push	r25
                           .text:00000466 af 93                            push	r26
                           .text:00000468 bf 93                            push	r27
                           .text:0000046a ef 93                            push	r30
                           .text:0000046c ff 93                            push	r31
                           .text:0000046e 8b e1                            ldi	r24, 0x1B	; 27
                           .text:00000470 91 e0                            ldi	r25, 0x01	; 1
                           .text:00000472 0e 94 6b 01                      call	0x2d6	;  0x000002d6 <_ZN5Print7printlnEPKc.constprop.3>
                           .text:00000476 ff 91                            pop	r31
                           .text:00000478 ef 91                            pop	r30
                           .text:0000047a bf 91                            pop	r27
                           .text:0000047c af 91                            pop	r26
                           .text:0000047e 9f 91                            pop	r25
                           .text:00000480 8f 91                            pop	r24
                           .text:00000482 7f 91                            pop	r23
                           .text:00000484 6f 91                            pop	r22
                           .text:00000486 5f 91                            pop	r21
                           .text:00000488 4f 91                            pop	r20
                           .text:0000048a 3f 91                            pop	r19
                           .text:0000048c 2f 91                            pop	r18
                           .text:0000048e 0f 90                            pop	r0
                           .text:00000490 0f be                            out	0x3f, r0	; 63
                           .text:00000492 0f 90                            pop	r0
                           .text:00000494 1f 90                            pop	r1
                           .text:00000496 18 95                            reti
[...]
```

I'll try to redo that in a clean checkout and PR it.
But not today, it's already midnite.

jayzakk, 2020-08-11:
(correction: I overlooked one additional new vector ANALOG_COMP_1, it's 4 new ones, now)

Checked the code, ISRs called as expected.

thegoodhen, 2020-08-12:
Wow, that's amazing! I still don't get how you did it, though. :D

jayzakk, 2020-08-12:
PR #40 is the answer ^^

for ANALOG_COMP_vect, I aliased another ANALOG_COMP_0_vect, as we now also have ANALOG_COMP_1_vect. Just to be able to correctly adress the one some needs, but also keep backwards compatible.

The other new ones are PCINT3_vect, PCINT4_vect and TIMER3_vect.

thegoodhen, 2020-08-12:
Amazing! That really seems like a very clean solution! Amazing work, thank you very much!

dbuezas, 2020-09-20:
Closing as it seems like #40 solved it.

closed 2020-09-20 by dbuezas