dbuezas/lgt8fx #41: Wrong analogReadResolution on boot

jayzakk, 2020-08-12:
As https://www.arduino.cc/en/Reference.AnalogReadResolution states

> It defaults to 10 bits (returns values between 0-1023) for backward compatibility with AVR based boards. 

This doesn't seem to be true - at LTG8F. 
I got values up to 4096, which made my "regular" code not working.

I guess we should change this behaviour.

seisfeld, 2020-08-21:
I can confirm, 12 bit seems to be default. I only see 4064 as max value though, not 4096.

dbuezas, 2020-09-20:
I'm willing to take a PR to start with 10 bits by default :)

jayzakk, 2020-09-20:
Ok, will PR it

LaZsolt, 2020-10-23:
If in wiring_analog.c at line 30 default variables will be set like below, the analogRead() default resolution could be 10 bit.
```
uint8_t analog_resbit = 2;
uint8_t analog_resdir = 1;
```

LaZsolt, 2020-10-23:
@seisfeld 
The gain error correction is the reason of the maximum ADC value is 4064.
```
// gain-error correction
#if defined(__LGT8FX8E__)
	pVal -= (pVal >> 5);
#elif defined(__LGT8FX8P__)
	pVal -= (pVal >> 7);
#endif
```

seisfeld, 2020-10-23:
Good catch @LaZsolt !

LaZsolt, 2020-10-24:
Here is a demonstration program of the ADC gain error. I think the ADC gain error should be mentioned in readme.
```
void setup() {
  Serial.begin(38400);
  
  analogReference(INTERNAL2V048);
  analogReadResolution(12);
  pinMode(DAC0, ANALOG);

  
//------------------------------------------------------
  uint16_t value;

  Serial.print(F("If ADMUX set AGND then ADC output: "));
  value = analogRead(AGND);
  Serial.print(value);  
  Serial.println( (value == 0) ? F("") : F("    ( Must be 0, is'n it? ADC not linear around zero valtage.)"));

  bitSet(ADMUX,4);      // ADMUX switch to DAC output
  printvalueatDAC(0);
  printvalueatDAC(1);
  printvalueatDAC(2);
  printvalueatDAC(3);
  printvalueatDAC(4);
  printvalueatDAC(250);
  printvalueatDAC(251);
  printvalueatDAC(252);
  printvalueatDAC(253);
  printvalueatDAC(254);
  printvalueatDAC(255);
  Serial.println();
  Serial.println(F("The maximum ADC value should be at the maximum DAC output value, but not. The reason is the gain error."));
  Serial.println();
  Serial.println(F("End test."));
  Serial.println();
}// end setup  

void printvalueatDAC(uint8_t x){
  delay(100);
  analogWrite(DAC0, x);
  Serial.print(F("If  DAC  ==  "));
  if (x < 10 ) Serial.print(F(" "));
  if (x < 100) Serial.print(F(" "));
  Serial.print(DALR);
  Serial.print(F("  then ADC output: "));
  Serial.println(directADCread());
}

static uint16_t adcReadNow()
{
  // start the conversion
  bitSet(ADCSRA, ADSC);

  // ADSC is cleared when the conversion finishes
  while (bit_is_set(ADCSRA, ADSC));

  return ADC;
}

int directADCread(){
  uint16_t pVal;
  uint16_t nVal;

#if defined(ADCSRA) && defined(ADCL)
  #if defined(__LGT8FX8P__)
  bitSet(ADCSRC, SPN);
  nVal = adcReadNow();
  bitClear(ADCSRC, SPN);
  #endif
  
  pVal = adcReadNow();

  #if defined(__LGT8FX8P__)
  pVal = (pVal + nVal) >> 1;
  #endif
#else
  // we dont have an ADC, return 0
  pVal = 0;
#endif

  // gain-error correction
#if defined(__LGT8FX8E__)
  pVal -= (pVal >> 5);
#elif defined(__LGT8FX8P__)
  pVal -= (pVal >> 7);
#endif

  return pVal;
}// end directADCread

void loop() {
}
```

seisfeld, 2021-01-26:
@LaZsolt I have a question regarding this gain correction thing. I'm using 2.048V as reference and set the ADC to 12bit. When I want to calculate expected ADC values like:

[image]

What do i use as ADC resolution in that formula? 4096 or 4064?

LaZsolt, 2021-01-26:
The resolution is 4064. Slightly bigger gain decrease the resolution. But replace "System Voltage" to "Reference Voltage".
Voltage Measured = Reference / Resolution * ADC Reading

seisfeld, 2021-01-26:
Thanks. Will take 4064 in all calculations and verify with multimeter. You're right about the ref voltage of course. (The formula image is not mine, I grabbed it off the web just for illustration.)

LaZsolt, 2021-04-02:
@seisfeld  
> Thanks. Will take 4064 in all calculations and verify with multimeter. You're right about the ref voltage of course. (The formula image is not mine, I grabbed it off the web just for illustration.)

As I writed above "Slightly bigger gain decrease the resolution" is true, but in calculations must use 4096.
If the gain error wasn't so big then the ADC value reach the maximum when the input reach the reference voltage.
But becase of error the ADC value reach the maximum **before** the input reach the reference voltage. So the ADC value decreased by gain correction for the good calculation. I hope it is understandable.
```
4096         Reference Voltage
----   =   ---------------------
4063       Max Voltage Mesasured

```

LaZsolt, 2021-04-02:
@jayzakk and @dbuezas: Help wanted? Here it is.

For the backward compatibility of the default analogReadResolution 
I merged this maintenance to my last Pull request. https://github.com/dbuezas/lgt8fx/pull/130

dwillmore, 2023-01-12:
Is this issue also closed by #130 ?

dbuezas, 2023-01-12:
Yes.

closed 2023-01-12 by dbuezas

ariefadha, 2025-05-13:
Sorry for opening this issue again. Can anyone explain to me why I get max value 4064 in 12 bit ADC instead of 4095?

jayzakk, 2025-05-13:
> Sorry for opening this issue again. Can anyone explain to me why I get max value 4064 in 12 bit ADC instead of 4095?

@LaZsolt already explained here in detail.
Plus, we're talking about a cheap microcontroller, not some expensive lab equipment. The controllers linearity and precision isn't that perfect, too, so why exactly are you missing that single one out of of 4096 values? The lowest and highest couple of values are usually the most unreliable anyways.

Is just, I don't get it.

ariefadha, 2025-05-13:
@jayzakk i am not using this for expensive lab equipment. just for controlling bldc. nor do i care about precision. but lowest value and highest value is mater in some code. it will effect your low and max range in code. instead giving 100% throttle it only give you 98% instead.

what I've read above is because "Slightly bigger gain decrease the resolution" thats why, i need someone to elaborate with this. 

heres why i bothered about 4064 instead 4096 : 

since this is "official" library, (not a random user who made this lib), so I expect to cross microcontrollers behaviour. and of course I can using map(adc,04064,0,4096)
but, once again this is official library, im expecting same result across microcontrollers.

imagine one user made with atmega328 (arduino nano) getting value from 0% to 100%. and another user using lgt8fx, digging hair out wondering why i can only get 98% max throttle.

dbuezas, 2025-05-13:
Note that I am indeed a random user who uploaded something to GitHub and other random (great) guys chimed in to make further improvements.

The only official thing you'll find here is the rule of being polite and thankful that others shared their work for free :)

hmeijdam, 2025-05-13:
> The only official thing you'll find here is the rule of being polite and thankful that others shared their work for free :)

You have just earned yourself the "quote of the week" award.

[image]

ariefadha, 2025-05-13:
well can you quote my exact word that I'm indicating not being polite?

1st I post with the word "sorry for opening this issue" (i tot sorry is the most polite way to ask something)

2nd I get answered that not answering my question 
(cheap microcontrollers, not some expensive lab equipment, etc2)

3rd I answer the 2nd comment by point to point for what he mention about.

well im not here to insult someone. and thanks for your hard work and free library.

and for reminder of me, can you please point me what word that im using so you guys feel im not polite. since english is not my language. and I answering according to the 1st reply point @jayzakk

dbuezas, 2025-05-13:
I'll delete all this chitchat (also including your expectations) in a couple of minutes to keep the issue relevant.
Now, although I technically didn't say you weren't polite, I implied I did it. Same way you implied you could give orders like a parent does to a child: "I expect xyz". That's impolite in English. It would be impolite even if you were paying for this, even if you owned it.
You are of course free to say whatever you want, but this won't do much motivating random people to help you. And I may delete it. Cos I can

ariefadha, 2025-05-13:
@dbuezas  well, im sorry if using word "expecting" is form an insult. its because of my vocabulary limitation to pick and choose word. since my original language only has around 120.000words vs english 600.000words. and different regions has difference culture, and yet we can not judge someone by texting. because we can't see the face and intonation/accent.

and i use that just for example and answering. not my point, while my point is asking someone to elaborate, so might i can help and pull. instead forcing my code and ruin the whole library.

in the end i am sorry if i insult @jayzakk and you. i will say no more, thanks for your and others hard works for this library. cheers

i will also delete this comment for a couple minutes. and feel free to delete all my comments as you are the author.

dbuezas, 2025-05-13:
I'll absolutely not delete all the comments! Just removed the offtopic stuff :)

My grain of sand is that normalising analog input would be risky because different boards may have slightly different ranges, and would also affect the output when a different analog reference is used (e.g internal or aref pin).

Also the atmega328p has a 10 bit adc, so it goes to 1024 instead of 4096 as the 12 bit adc in the lgt chip. They are different chips after all.