dbuezas/lgt8fx #42: millis() unreliable?

jayzakk, 2020-08-12:
In some project, I do quite a lot:
- receive about 150 udp packets/second using Ethernet.h (SPI)
- push 280 WS2812 pixels 25 times a second (light_ws2812).
- show some pretty info every 2 seconds using OLED display(I2C)

For "performance measuring", I use millis() to check on the network receiving loop, if there are about 2 seconds gone by, and calculate statistics "by second" using the current millis() and the last one I did the stats.

This now brings totally wrong data, and I found millis() to blame.

So, tests.
I output the millis() via Serial in the network receiving loop, and get some (timestamped) results:

```
18:47:52.022 -> 2022
...
18:48:11.989 -> 17288
```
thats 20 seconds in real world, but only 15.2 seconds in LGT8F-world. _It feels like some time-diverting star trek episode..._

A test sketch (`void loop() { delay(1000); Serial.println(millis(),DEC); }`) shows correct values:
```
19:01:42.919 -> 1580
...
19:02:02.895 -> 21582
```
20 seconds are 20 seconds.

**Questions:**
Where does that millis() come from?
What could corrupt millis() that much?

Greets,
Oliver - no ideas.

jayzakk, 2020-08-12:
Ok, I can answer myself.
millis() returns a local variable, which is incremented by timer0 overflows.
If some library disables ISRs for some time, (timer) interrupts will be skipped.

And that happens with the light_ws2812 library (before I was using FastLED, but the projekt got too large for the flash).
Light_ws2812 disabled the ISRs for the FULL update cycle, which FastLED seems NOT to do.
After changing the light_ws2812 to only disable ISRs in the critical "bit-bang-a-byte" phase, and allow ISRs again while fetching a new byte, millis() came back to MY universe. Short interruptions of the ws2812 full-update-cycle seem to be totally no problem at all.

Thanks for reading ;)

dbuezas, 2020-09-20:
Interesting gotcha!

closed 2020-09-20 by dbuezas