dbuezas/lgt8fx #48: Package name with space makes arduino-cli awkward

majenkotech, 2020-08-30:
Having the global package name "LGT8fx Boards" in the packages .json file makes it awkward to use arduino-cli.  All boards, and the core itself, get an awkward name that has a space in it, making escaping that space a pain, and not simple to do in a platform-agnostic way.

```
$ arduino-cli board listall | grep -i lgt8
LGT8F328                                    LGT8fx Boards:avr:328                         
LGT8F88                                     LGT8fx Boards:avr:88                   
```

Ideally the package name should be just `lgt8fx` which results in the much easier to use:

```
$ arduino-cli board listall | grep -i lgt8
LGT8F328                                    lgt8fx:avr:328                                
LGT8F88                                     lgt8fx:avr:88  
```

You still get the "pretty" name of the core in the platform name:

```
$ arduino-cli core search lgt8fx
ID         Version Name         
lgt8fx:avr 1.0.6   LGT8fx Boards
```

No other core has a space in its name - and for good reason: it's a right pain if there is a space.

dbuezas, 2020-09-09:
Thanks for the issue,
I'll change the name in the json in the next release (to LGT8fx)

dwillmore, 2023-01-12:
Did this actually get fixed in the next release?

dbuezas, 2023-01-12:
It looks like I had "fixed" it wrong in the build script. Fixing this now.

closed 2023-01-12 by dbuezas