dbuezas/lgt8fx #52: WAVGAT Nano Faux clone

pixelwaster, 2020-09-13:
Will this info apply to WAVGAT Nano LGT8F boards? They have a 12 MHz resonator which stopped uploading. Now that we can flash a different boot loader, so can I fish this out of the parts bin and use it?

dbuezas, 2020-09-20:
Hello there!
It may work if the problem is the bootloader in the first place. The only complication is that those boards don't have the SWC and SWD pins exposed so you will have to hook directly to the chip.
[image]

dbuezas, 2020-09-20:
Let me know if you succeeded and I'll close the issue :)

pixelwaster, 2020-09-24:
After a but of messing about, removing the resonator, etc; I was able to flash Optiboot. The steetings were LGT8F328 with internal 32 MHz clock with the variant set for QFP32 nano-style.

The SWC and SWD pins are exposed on the six pins opposite the USB, PB5/SCK, Reset, Vcc, and gnd are the rest. Five more pins if I don't need crystal accuracy. I could always use an external ttl osc and go down to 4 pins.

closed 2020-09-24 by pixelwaster