dbuezas/lgt8fx #53: Optiboot Source for LGT8F328P

seisfeld, 2020-09-15:
I played around with the boot loader recently, and I found when I compile it for the LGT8F328P from the source coming with this core, the hex file differs from the supplied one and it also does not seem to work when flashed on the chip. While the supplied hex works like charm I wonder with what source it was built. Anyone has a clue?

I compiled with `make lgt8f328p` after setting the `TOOLROOT` in the Makefile to `/Applications/Arduino.app/Contents/Java/hardware/tools` to match my local installation on the Mac.

dbuezas, 2020-09-20:
Interesting finding. Maybe @jayzakk knows something about this?

jayzakk, 2020-09-20:
I don't have a Mac, only Windows and Ubuntu.
And I never tried to compile it, thinking it is working as committed. Should I take a look into?

seisfeld, 2020-09-20:
I don’t think this is a Mac specific issue hehe. As said, the supplied hex works fine, I flashed it a dozen times having no issues. Then I was curious and tried to compile it from the given source, finding the hex is different, even in size. And it does not work, can’t flash anymore via the IDE when using the self compiled Boot loader. So I was just wondering what is different and why they apparently don’t supply the right source (or I do something wrong). 

I was thinking, what if we can compile a working one, maybe we can strip it down a bit to save some flash or whatever.

mzhboy, 2021-03-06:
I tried `avr-objcopy -O ihex optiboot_lgt8f328p.elf output.hex` ,the `optiboot_lgt8f328p.elf` comes from this core, the output is different from provided.

output:
```ihex
:1074000011248FEF98E09EBF8DBF94B714BE892FD3
:107410008D7011F0892FE0D190E89093F20083E90C
:107420008093F2009093610081E08093610096BFA9
:107430008CE486BF85E08093810088E18093C10061
:1074400086E08093C20080E18093C4008DE0AED1DD
:10745000259A96E020E33CEF51E040E2309385002E
:107460002093840056BBB09BFECF85B1842785B99D
:10747000A895915099F7662477244301CC24DD2404
:1074800064E9A62E51E0252E312C412C512C7DD1C2
:10749000813461F47AD1182F8FD1123829F1113843
:1074A00011F485E001C083E068D164C1823411F435
:1074B00084E103C0853419F485E086D15BC185354C
:1074C00079F463D1E82EFF2460D1082F10E0102F4B
:1074D00000270E291F29000F111F6ED168014AC114
:1074E000863521F484E070D180E0DECF843609F067
:1074F000C5C04BD190E0F82EEE2447D1E80EF11C28
:1075000044D1B82EE70100E011E03FD1F8018193AA
:107510008F012197D1F711BC8D2D992782BD82B59E
:1075200086958695282FD401C301022E04C0B695F6
:10753000A795979587950A94D2F780FD15C0F6E434
:10754000BF1691F4D201C10104C0880F991FAA1F70
:10755000BB1F2A95D2F7682A792A8A2A9B2AAFBAB2
:1075600082E98FBB0000000027D1E5E4BE16B1F42C
:10757000960110C021BD2F5F3F4F832F992782BDF9
:10758000C050DF4F888180BDF4E0FFBB82E08FBB3D
:10759000CF5FD040CE15DF0568F3DBC0C114D10446
:1075A00009F04EC06091000170910101872F90E0B9
:1075B000982F8827262F30E0822B932B90938501DC
:1075C000809384014091020150910301852F90E046
:1075D000982F8827242F30E0822B932B90938701BC
:1075E00080938601209118018091190190E0982FD5
:1075F000882730E0822B932B909389018093880118
:1076000020911A0180911B0190E0982F882730E08B
:10761000822B932B90938B0180938A0160931801A6
:107620007093190140931A0150931B018CE08093D1
:107630000001A092010110920201EAE3E09303012C
:10764000E0E0F1E011BC808180BD81E081BD8181FD
:1076500080BD82E081BD828180BD83E081BD838168
:1076600080BD8E2F8C0D81BD84EA8FBB82EA8FBBDB
:10767000349681E0E038F80729F77CC0843709F0B8
:107680006BC083D090E0F82EEE247FD0E80EF11C82
:107690007CD0182F91D0153489F4E6018701C1BD43
:1076A00021968D2F992782BDE1E0EFBB00000000FD
:1076B00080B563D00150104091F74BC0E6018701BF
:1076C000209729F480918401909185013CC0C130BC
:1076D000D10529F480918401909185010FC0C230B9
:1076E000D10529F480918601909187012CC0C33087
:1076F000D10539F48091860190918701892F9927CE
:1077000022C0C831D10529F4809188019091890166
:107710001AC0C931D10529F480918801909189015D
:10772000EDCFCA31D10529F480918A0190918B0166
:107730000AC0CB31D10529F480918A0190918B0147
:10774000DDCFFE018491219618D00150104009F040
:10775000B7CFCE0CDF1C0EC0853739F42DD08EE1AB
:107760000CD085E90AD08FE09FCE813511F48CE0F2
:107770001DD022D080E101D08ACE982F8091C00008
:1077800085FFFCCF9093C60008958091C00087FFCD
:10779000FCCF8091C00084FD01C0A8958091C600F7
:1077A00008959AE29A95F1F708950895E0E6F0E0D9
:1077B00098E1908380830895E8DF803219F08DE0AE
:1077C000F5DFFFCF84E1D9CF1F93182FDEDF1150F3
:1077D000E9F7F2DF1F910895282E80E0E7DFECE063
:0477E000FF270994E2
:040000030000740085
:00000001FF
```
[... 60 lines cut ...]
:107770001DD022D080E101D08ACE982F8091C00008
:1077800085FFFCCF9093C60008958091C00087FFCD
:10779000FCCF8091C00084FD01C0A8958091C600F7
:1077A00008959AE29A95F1F708950895E0E6F0E0D9
:1077B00098E1908380830895E8DF803219F08DE0AE
:1077C000F5DFFFCF84E1D9CF1F93182FDEDF1150F3
:1077D000E9F7F2DF1F910895282E80E0E7DFECE063
:0477E000FF270994E2
:0277FE00000584
:040000030000740085
:00000001FF
```

--------------------

convert to binary
`objcopy -I ihex output.hex -O binary output.bin`

size
|file name|size|
|----------|----|
|output.bin |996|
|optiboot_lgt8f328p.bin|30720|

30KB is wired

dwillmore, 2021-03-06:
Visually, them seem to differ only by the one line:

:0277FE00000584

I'm guessing that this is at a different memory range than the rest of
the boot code and that's throwing off the 'size' calculation.  I don't
read Intel hex format natively, so I'm not sure what this line does.

LaZsolt, 2021-03-07:
@dwillmore:
> Visually, them seem to differ only by the one line:
:0277FE00000584

This line contain the version number. The version section start address is 0x77FE and the stored version number is 0x0005.

A short piece form optiboot...lst :
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         000003e4  00007400  00007400  00000054  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .bootv        00000004  00000000  00000000  00000438  2**0
                  CONTENTS, READONLY
  2 .version      00000002  000077fe  000077fe  0000043c  2**0
                  CONTENTS, READONLY
```

And from optiboot.c:
```
/* Version Numbers!                                       */
/*                                                        */
/* Arduino Optiboot now includes this Version number in   */
/* the source and object code.                            */
/*                                                        */
/* Version 3 was released as zip from the optiboot        */
/*  repository and was distributed with Arduino 0022.     */
/* Version 4 starts with the arduino repository commit    */
/*  that brought the arduino repository up-to-date with   */
/*  the optiboot source tree changes since v3.            */
/* Version 5 was created at the time of the new Makefile  */
/*  structure (Mar, 2013), even though no binaries changed*/
/* It would be good if versions implemented outside the   */
/*  official repository used an out-of-seqeunce version   */
/*  number (like 104.6 if based on based on 4.5) to       */
/*  prevent collisions.                                   */
```

dwillmore, 2023-01-12:
Sounds like this is not a functional difference and that the source does compile comparably to the existing binary. @seisfeld Does that sound correct to you?  If so, please close this issue.

seisfeld, 2023-01-12:
As said, when I compiled it back then and flash it, I could not install any firmware using that bootloader. Anyone tried this recently? I don't need my own bootloader at this moment but IMHO it should be possible to do if required.

dwillmore, 2023-01-12:
If the only difference is the version included in the compiled version, we
should look for a way to have the linker no produce that section--which
would render the issue moot.  If the issue is that we cannot program the
bootloader at all, then a new issue should be opened for that.

seisfeld, 2023-01-12:
You can install the bootloader to the chip. But you can't install any firmware USING that bootloader. If someone could verify this to rule out issues with my system I would appreciate it. I don't see why we need a new issue for that though.

dwillmore, 2023-01-12:
You wouldn't, it was conditional.  What boards are needed to reproduce
this?  I have the SOP20 boards and some of the XI boards which I could test
with.

seisfeld, 2023-01-12:
The board I used is not available anymore (we used the LGT8F328P on a custom board). I would assume a LGT8F328P Nano clone would fit best?! I don't really know.

dwillmore, 2023-01-12:
The SSOP20 boards are LGT8F328P core, so could I test with one of those?

seisfeld, 2023-01-12:
Would say yes. If you have the hardware to change a bootloader this is easily reversible. So not much risk I would say!?

dwillmore, 2023-01-12:
I have a lot of these boards and they were very cheap.  I only got them to
support this project, anyway. :)  I'll look into how to program the
bootloader.  I know I did it years ago, but I'll have to google around to
refresh my memory.  Give me a week or so and poke me if I don't show up,
it's likely I got busy and forgot about it.

seisfeld, 2023-01-12:
For me this is no issue anymore, so no pressure. We switched the MCU on our custom board. So for me this is just FYI. But I think for others it might be still interesting if you would want to modify the bootloader and compile it from source.

dwillmore, 2023-01-12:
Understood.  Most of these older issues only remain as a form of
documentation for others who may come upon these problem later.  It's best
to have all of the information in one place to make their journey easier.

flybabo, 2023-01-30:
I need to rebuild the bootloader to relocate it with different EEPROM requirements as current bootloader is configured to 1KB EEPROM.
I compiled the lgt8f328p bootloader both at Mac and Windows from the latest 2.0.0 source code with avr-gcc 7.3.0 which comes with the latest Arduino ide.
Both Mac and Windows produced 92 bytes bigger binaries and both are not working.
Is there anyone ever rebuild the same original working bootloader from 2.0.0 source code tree?
If so, please share how to rebuild.
TIA.

flybabo, 2023-01-30:
After trying various versions of avr-gcc compiler on various platforms (PC/Mac/Linux), I finally got the exact same hex file at the Windows platform.
You'll need to install WinAVR 20100110 pkg with avr-gcc version 4.3.3.

dbuezas, 2023-01-30:
That's a great finding!
I think you can configure the EEPROM space in the sketch, and then set the available flash in the boards.txt file. If you succeed, it would be amazing to add it to the build menu (like we did with the F_CPU)

closed 2023-03-21 by dwillmore