dbuezas/lgt8fx #55: LGT8F328P LQFP32 bricked, unable to flash bootloader

leifclaesson, 2020-10-07:
Hello there! Thanks for the excellent work with the LGT8F library.

Due to a mistake entirely my own, one of my WAVGAT Pro Mini LGT8F boards is bricked. I was working on an LGT8F project, I had hooked up TX and RX but not RST (DTR) -- I was using a button and resetting manually to upload.. and one time I pushed the reset button again just as it had started uploading. All it takes is once, it seems.

I normally work with the ESP8266 and I'm used to how resilient it is -- no matter what you do, you can always re-program it by RX/TX, but apparently this is not the case with the LGT8F. I have now learned my lesson. The question is, is it possible to revive it? Yes, I have a few more in the drawer, but I do like a challenge, and I'm sure this will happen again at some point even though I now always make sure to connect the reset pin. :)

I found this post and tried to follow [these instructions](https://forum.arduino.cc/index.php?topic=661012.0)

This is the board in question:

[image]

This is my soldering work, adding 30 AWG wires to access the SWC and SWD pins.

[image]

I'm trying to use an Arduino Uno clone with what appears to be a genuine Atmel 328P part as an ISP to burn a new bootloader, and I've wired things up according to [this](https://github.com/LGTMCU/LarduinoISP) and also running that sketch of course.

I then try to burn a new bootloader as follows:

[image]

..and I get the following message:

```
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

Error while burning bootloader.
```

I don't know how to get access to the command line from the Arduino environment (i normally use Sloeber for development) and I have of course double checked the connections. :)

To pre-emptively attempt to find out what the actual LGT8F revision is, I ran [Kevin Darrah's boot signature dump sketch](https://gist.github.com/krdarrah/6e6cb94c1df015e8e9f910ae5cf85299) on an identical WAVGAT Pro Mini board. Result:

```
C	94	0	3A	C	94	8F	0	
C	94	8F	0	C	94	8F	0	
C	94	8F	0	C	94	8F	0	
C	94	67	0	C	94	8F	0	
```

I don't know whether it tells us anything. I'm not used to the LGT8F toolchain yet, just trying to think ahead.

Using Arduino 1.8.13, lgt8fx 1.0.6.

I don't know what else to try at this point. Any ideas? I would appreciate any help, and so would the poor little LGT8F328P that fell victim to my untimely resetting.

seisfeld, 2020-10-07:
Can you try to follow these instructions?

https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md

dbuezas, 2020-10-07:
Those solder points look very clean!
The link @seisfeld sent should work, just make sure to use another lgt board as the ISP.

leifclaesson, 2020-10-07:
> Those solder points look very clean!

I learned to SMD solder on youtube. Who knew it could be useful for anything other than cat videos and crazy politics. :)

I tried the LarduinoISP sketch with a Wavgat R3 board, and got the following error message:

```
 ***failed;  
avrdude: WARNING: invalid value for unused bits in fuse "efuse", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x07 (double check with your datasheet first).
```

...then i noticed the triple-flashing of the Pro Mini LED.

**IT WORKED!** It's alive again, I was able to upload the blink sketch no problem.

Thank you both so much.

But hey, I have a couple of question/comments regarding the (instructions)[https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md]:

> iv. Short RESET pin and VCC pin of arduino board to avoid bootloader executing.

That would prevent resetting I guess (as holding RESET low is what makes it reset) but why is that necessary? Should that step be there? I skipped this step.

Also, if it's recommended to use another LGT board as the ISP, should we really refer to it as the "arduino"?

Thanks again!

leifclaesson, 2020-10-07:
Just leaving it open for now to give the comments at the end a chance to be noticed.

dbuezas, 2020-10-07:
> **IT WORKED!** It's alive again, I was able to upload the blink sketch no problem.

🎉🎉🎉 Happy to hear! There's no better feeling than reviving a bricked board. The cheaper the board, the better the feeling 😁

> > iv. Short RESET pin and VCC pin of arduino board to avoid bootloader executing.
> 
> That would prevent resetting I guess (as holding RESET low is what makes it reset) but why is that necessary? Should that step be there? I skipped this step.

When the computer opens the serial port, the serial adapter resets the ISP board through the DTR pin, that may result in the programmer not taking the code to send to the target board. I see you are using pro-mini style boards and you reset the board manually so in your case that's not necessary. I assume your usb<->serial adapter doesn't have a DTR pin, or you've chosen not to connect it (?)

> Also, if it's recommended to use another LGT board as the ISP, should we really refer to it as the "arduino"?

That's a very good point. I'll fix that so it is clearer.

dbuezas, 2020-10-07:
Btw, can I use your pics in the docs to show how to connect the isp to these wavgat boards?

leifclaesson, 2020-10-07:
> There's no better feeling than reviving a bricked board. The cheaper the board, the better the feeling 😁
You are so right!!! I'm so glad you understand :-D

>  I assume your usb<->serial adapter doesn't have a DTR pin, or you've chosen not to connect it (?)

I actually don't have a proper generic USB serial adapter. That's what got me into trouble in the first place. I've used ESP8266s and ESP32s and UNOs and Megas and ESP-01s and even ESP-12s. Most of those have USB onboard, and I have custom dedicated programming boards for the last two.

When I first bricked my board, I was using a WAVGAT R3 board (UNO) as a USB serial converter, with the R3's MCU held in RESET, leaving the RX/TX pins free to connect to the Pro Mini. This of course also voided the DTR RESET signal which is why I was handling the reset signal myself -- and too well.
Later, I modified that R3 board by cutting a trace in just the right place, and adding a 2-way switch to connect the onboard MCU's RESET pin either to DTR or to ground. With the MCUs RESET pin connected to ground, that leaves the reset pin in the header still connected to DTR, so that I can now connect that to a pro mini and have it totally seamless.

Tonight, I modified an ultra-cheap ESP-01 programming board (which completely lacked DTR->Reset connection) by soldering directly to the CH340 and adding a 100nf ceramic capacitor, and then finally made a dupont adapter cable from the 2x4 ESP-01 layout to the 1x6 short side Pro Mini layout. I think that counts as a proper Pro Mini programming cable. :)

Fun stuff. So many toys, so little time, but soldering is therapeutic.

I'd be honored if you use my photo, have at it. :)

///Leif

dbuezas, 2020-10-08:
Wow, very creative! I never thought of the hack of keeping the reset btn pressed while hijacking the serial interface of the board. I think you'll like a project I'm about to upload: an no-extra-components, lgt328p-based (very slow) stand-alone-web-oscilloscope.

[image]

leifclaesson, 2020-10-08:
Nice! But wait, no extra components? Certainly you're not bit-banging WiFi on one of the GPIO pins??? :)

dbuezas, 2020-10-08:
Hehe, well, for boards with serial adapters. And a pc.

closed 2020-10-10 by leifclaesson

dbuezas, 2020-10-11:
@leifclaesson check it out! https://dbuezas.github.io/arduino-web-oscilloscope/

dwillmore, 2020-10-11:
Just goes to a page that says that it needs a newer version of Chrome
>86.0.  But I'm using Firefox...

seisfeld, 2020-10-11:
> Just goes to a page that says that it needs a newer version of Chrome

Currently it is Chrome only.

Yodinsn, 2024-01-26:
> Can you try to follow these instructions?
> 
> https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md

Can I use arduino uno to burn lgtisp?

dbuezas, 2024-05-05:
[Here's a different guide ](https://osoyoo.com/2022/08/11/how-to-download-bootloader-for-lgt-nano/) that says you can. They use another arduino core and ISP code, but if theirs work on both Atmel and Logic Green chips, this one should probably work too.

Rtyuakatsuki, 2025-05-12:
Hallo anyone can help me to revive lgt8f328p ? 
Lqfp32 mini evb 

I was playing with gaminator soldering station 
And fixing adc problem 
And i flash hex file from there 
And my lgt8f328p not responding even i tries reflash it 
With blink code it not work 
The programmer 
Not responding