dbuezas/lgt8fx #58: Have you noticed the extra pads on the Pro Mini style LQFP32 ?

SuperUserNameMan, 2020-10-13:
Hello.

Have you noticed the 2 little SMD pads next to the red power LED of the Pro Mini style LQFP32 ?

The pad close to the LED is GND. 
The other pad is connected to the pin 07 of the chip (PB6 / XTALO ).

I thought it could be used to add an external oscillator, but PB7 / XTALI is on pin 08 which is not connected to anything. 

Then I thought is could be used to solder an extra LED, but there is no room for a resistor.

Any idea what could be the purpose of this extra pad ?

[image]

-------

LGT8FX8P_databook v1.0.5-English.pdf : page 6
> PB6 / XTALO :
> - PB6 : programmable interface B6.
> - XTALO : oscillator IO output interface.

LGT8FX8P_databook v1.0.5-English.pdf : page 84
> Port B Pins Alternate Functions
> PB6 : 
> - XTALO / TOSC1 ( External Master Oscillator pin XO )
> - PCINT6 ( Pin change interrupt 6)

LGT8FX8P_databook v1.0.5-English.pdf : page 85
> XTALO/TOSC1/PCINT6- Port B Pin 6
> - XTALO: external crystal oscillator pin XO
> - TOSC1: Timer external Oscillator pin 1. Used only if internal RC is selected as master working clock, and the asynchronous timer is enabled by the correct setting in ASSR register. When the AS2 bit in ASSR is set (one) and the EXCLK bit is cleared (zero) to enable asynchronous clocking of Timer/Counter2 using the Crystal Oscillator, PB6 is disconnected from internal I/O port, and becomes the input pin of the Oscillator amplifier. In this mode, external crystal Oscillator is connected to this pin.
> - PCINT6: Pin Change Interrupt 6. The PB6 pin can serve as an external interrupt source. If PB6 is used as a crystal oscillator pin, DDB6, PORTB6 and PINB6 will all read 0.

SuperUserNameMan, 2020-10-14:
I thought it could be a secret switch to make the flash read only, but it does not seem to have any effect on the bootloader.

LaZsolt, 2021-03-28:
> PS : my version is 3.3V and the version of the picture below is 5V, so it seems not to be a way to tell to the program which version of the regulator.

Here is a way to figure out your board's regulator voltage: https://github.com/dbuezas/lgt8fx/discussions/123#discussion-3287695

LaZsolt, 2022-05-29:
The secret of the extra pad if they solder the LGT8F328D is:

[image]

dwillmore, 2023-01-12:
@LaZsolt seems this issue can be close and still serve as a reference in the future.

closed 2023-01-12 by dwillmore

SuperUserNameMan, 2023-01-14:
oh, forgot about this.
Was not really an issue. I think the forum was not available back then when i posted.
Maybe this could be moved into the forum ?