dbuezas/lgt8fx #59: PGM direct access optimization : performance comparison

SuperUserNameMan, 2020-10-19:
Hello,

Regarding the discussion and benchmarks in #33 about accessing PROGMEM directly, I've done some more benchmarks.

The sketch used as benchmark outputs an analog wave through the pin DAC0.
The wave is stored into an array which is read from SRAM or from PROGMEM using direct access or pg_read_byte().
Then, using an oscilloscope, I measured the frequency of the wave.

Here are the results at 32 MHz :

\# | Storage | Access | Output Frequency | Flash usage | RAM usage
--|-----------|-----------|----------------------|---------|-------------
1 | SRAM | table[ adr ] | 4800 Hz | 1402 | 283
2 | PROGMEM | pgm_read_byte( adr ) | 4600 Hz | 1402 | 11
3 | PROGMEM |  (\*((uint8_t*)(0x4000 + (adr)))) | 4600 Hz | 1402 | 11
4 | PROGMEM |  (\*((uint16_t*)(0x4000 + (adr)))) | 4300 Hz | 1400 | 11

----

SRAM access is the fastest.

using ``pgm_read_byte()`` is as fast as the 8 bits direct access, and consume as much flash memory.

the 16 bits direct access is the slowest, but it saves 2 bytes of flash.

----

```` C++

//#define PGM_READ( adr ) table[ adr ] //!\\ <--- don't forget to comment the PROGMEM
//#define PGM_READ( adr ) pgm_read_byte( table + (adr) )
#define PGM_READ( adr ) (*((uint8_t*)(0x4000 + table + (adr))))
//#define PGM_READ( adr ) (*((uint16_t*)(0x4000 + table + (adr))))

PROGMEM		//!\\ <--- don't forget to comment when testing SRAM mode
const byte table[] = {
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
	0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
};

#define SAMPLES 255

void setup() 
{
	analogReference(DEFAULT);
	
	pinMode(DAC0, ANALOG);
}

void loop() 
{
	for (byte i = 0; i <= SAMPLES; i++) {
		analogWrite(DAC0, PGM_READ( i ) );
	}
}

````

SuperUserNameMan, 2020-10-19:
ok, i closed this issue temporarily because my LQFP32 board was resetting itself for unknown reasons whith ``TEST_3_PGM_READ_BYTE_10_INLINE``. But it stopped doing that ... so i updated the benchmark above, and reopen the issue.

LaZsolt, 2020-10-19:
analogWrite is very slow. Try my code to get higher freq. ( Your table contains 272 byte. 17 x 16 byte )

Anyway your test is completely useless. In fact every comparison is useless, because compiler does not know what are we wanted to address at above 0x4000. That's why the compiler generate a less optimized code.

This code below I designed 3 cases. RAM addressing, progmem addressing and remapped progmem addressing.
- In first case compiler does everything perfect when I watch the generated assembly.
- In second case the compiled code is not optimal, because overwrite the address register with readed data. So extra code generater to restore.
- In 3-rd case inside the for cycle the reading address is calculated every times. (Adding 0x4000 to the pointer at every reads) But in C language is written: *((char*)p++)
```
#define TESTTYPE 1

#if   TESTTYPE == 1
  // RAM read
  #define PTR_INIT( _var ) (uint16_t) (&_var)
  #define PGM_READ( _adr ) *((char*)_adr++)
#elif TESTTYPE == 2
  // PROGMEM read
  #define PTR_INIT( _var ) (uint16_t) &_var
  #define PGM_READ( _adr ) pgm_read_byte( _adr++ ) 
#elif TESTTYPE == 3
  // Remapped PROGMEM read
  #define PTR_INIT( _var ) 0x4000 + (uint16_t) (&_var)
  #define PGM_READ( _adr ) *((char*)_adr++)
#endif

PROGMEM    //!\\ <--- don't forget to comment when testing SRAM mode
const byte table[] = {
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
  
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
  0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
};

#define SAMPLES 255

void setup(){
  analogReference(DEFAULT);
  pinMode(DAC0, ANALOG);
}

void loop(){
  uint16_t p;

while (1) {                                  // Without this while, compiler generate call 0 at the end of for cycle.
      p = PTR_INIT( table );

    for (uint16_t i = 0; i <= SAMPLES; i++) {
      DALR = PGM_READ( p );                  // Write direct into DAC register
    }
  }
}
```

dbuezas, 2020-10-19:
I'd recommend to disable timer0 to avoid the millis interrupt getting in the way.
```cpp
  bitClear(TIMSK0, TOIE0); 
```

### TESTTYPE = 1 --> 64us = 15,625 Hz
[image]

### TESTTYPE = 2 --> 96us = 10,417 Hz
[image]

### TESTTYPE = 3 --> 104us = 9.615 Hz
[image]

---

by the way, if you allow me I'd rephrase "**Anyway your test is ~completely useless~**" to "**Anyway your test is skewed by compiler optimisations**"

I personally appreciate the brains and curiosity behind all these experiments! ❤️

SuperUserNameMan, 2020-10-20:
@LaZsolt :

> ... because compiler does not know what are we wanted to address at above 0x4000. That's why the compiler generate a less optimized code.

Yeah, I don't know yet how to look at the assembly code, so I have no idea what the compiler does behind the curtain. 

In that condition, the only thing I can compare are performances in an imperfect benchmark that I tried to make closer to a real use case of mine.
Direct write to ``DALR`` is a good idea indeed, but using the ``while(1)`` you enclosed into the ``loop()`` or disabling timer0 as proposed by @dbuezas could be incompatible with a real use-case scenario.

Maybe I'm wrong, but even I we manage to write an over-optimized benchmark, we have no clue that the compiler will behave the same way in a random real use case. So, in that perspective, we could say that any benchmark is useless. And that the only useful benchmark is the real use-case that we want to optimize.

BTW, I've noticed that ``for (uint16_t i = 0; i <= SAMPLES; i++)``  was slower than ``for (uint8_t i = 0; i <= SAMPLES; i++) {``, so if our point is to get the maximum output speed, we might want to avoid ``uint16_t`` or ``auto`` for loops and stick to ``uint8_t``, unless of course if the compiled assembly code shows otherwise in a real use-case.

Even if I was aware that it is still not a perfect benchmark, this test shown me that it was possible to access ``PROGMEM`` as ``table[0x4000 + adr]`` and that it could be faster than the other methods we previously tested in #33.

Again, I don't know if the ``table[0x4000 + adr]`` is calculated each time, but in this benchmark I noticed that using ``const uint8_t * pgm_table = 0x4000 + table;`` and ``pgm_table[ adr ]`` is slower. (``table[0x4000 + adr]`` is the fastest).

----

@dbuezas :

> I'd recommend to disable timer0 to avoid the millis interrupt getting in the way.

Oh ! So that was it ! I was wondering why the signal was slightly unstable.

And that's because of this instability that I reduced the frequency by a factor of 10 so I could read an exact frequency (range 0-999) on my DSO213.

----

Maybe I'll come back again later with a better "almost real-use-case scenario" benchmark and share my results with the community.

closed 2020-10-20 by SuperUserNameMan

LaZsolt, 2020-10-20:
@SuperUserNameMan 

> in this benchmark I noticed that using const uint8_t * pgm_table = 0x4000 + table; and pgm_table[ adr ] is slower. (table[0x4000 + adr] is the fastest).

Strange behaviour. I'll investigate it later.

SuperUserNameMan, 2020-10-20:
> @SuperUserNameMan
> 
> > in this benchmark I noticed that using const uint8_t * pgm_table = 0x4000 + table; and pgm_table[ adr ] is slower. (table[0x4000 + adr] is the fastest).
> 
> Strange behaviour. I'll investigate it later.

Yes that would be appropriate to investigate further, because this afternoon, I was trying to use ``table[0x4000+adr]`` in an other project, and it was not working at all.

SuperUserNameMan, 2020-10-21:
Ok, after further tests, I've found that my beloved brand new ``table[0x4000+adr]`` was not reliable, and that it had some strange unexpected behaviors.

Here is a test code that makes it obvious :
```` C++

const uint8_t data[] PROGMEM = { 123, 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20 };

int data_len = sizeof( data );

const uint8_t * pgm_data = (uint8_t*)((uint16_t)&data + 0x4000);

void setup()
{
	//delay(3000);
	
	Serial.begin(9600);
	
	Serial.println("---"); // ---- display memory adresses 
	
	Serial.print( " data         = 0x" ); Serial.println( (uint16_t) data, HEX );
	Serial.print( "&data         = 0x" ); Serial.println( (uint16_t) &data, HEX );
	Serial.print( "&data[0]      = 0x" ); Serial.println( (uint16_t) &data[0], HEX );
	Serial.print( "&data[0x4000] = 0x" ); Serial.println( (uint16_t) &data[0x4000], HEX );
	Serial.print( " pgm_data     = 0x" ); Serial.println( (uint16_t) pgm_data, HEX );
	Serial.print( "&pgm_data     = 0x" ); Serial.println( (uint16_t) &pgm_data, HEX );
	
	
	Serial.println("---"); // ---- display reference
	
	Serial.print("pgm_read_byte( data ) = ");
	Serial.print( pgm_read_byte( data ) );
	Serial.println();
	
	Serial.print("pgm_read_byte( data + a ) = { ");
	for( int a=0; a < data_len ; a++ )
	{
		Serial.print( pgm_read_byte( data + a ) );
		Serial.print(", ");
	}
	Serial.println(" }");
		
	
	Serial.println("---"); // --- display data[ a ]
	
	Serial.print("data[ 0 ] = ");
	Serial.print( data[ 0 ] );
	Serial.println();
	
	Serial.print("data[ a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( data[ a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
	
	
	Serial.println("---"); // --- display table[ 0x4000 +a ]
	
	Serial.print("data[ 0x4000 ] = ");
	Serial.print( data[ 0x4000 ] );
	Serial.println();
		
	Serial.print("data[ 0x4000 + a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( data[ 0x4000 + a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
			
			
	Serial.println("---"); // --- display pgm_data[ a ]
	
	Serial.print("pgm_data[ 0 ] = ");
	Serial.print( pgm_data[ 0 ] );
	Serial.println();
	
	Serial.print("pgm_data[ a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( pgm_data[ a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
	
		
	Serial.flush();
}

void loop()
{
}

````
----
Unexpected results that i get are :
```
---
 data         = 0x68
&data         = 0x68
&data[0]      = 0x68
&data[0x4000] = 0x4068
 pgm_data     = 0x4068
&pgm_data     = 0x100
---
pgm_read_byte( data ) = 123
pgm_read_byte( data + a ) = { 123, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20,  }
---
data[ 0 ] = 123
data[ a ] = { 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 135, 0, 0,  }
---
data[ 0x4000 ] = 0
data[ 0x4000 + a ] = { 123, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20,  }
---
pgm_data[ 0 ] = 123
pgm_data[ a ] = { 123, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20,  }

```
----
I have no clue why, and it's starting to make me crazy ... is it my brain ? is it GCC ? it it covid ? i don't know.

My config : 
- Linux 5.4.0-52-generic x86_x64
- Arduino 1.8.13 (downloaded from the official web site)
I have not clue which version of GCC it uses.

SuperUserNameMan, 2020-10-21:
Ok, I think i understood why ``table[0x4000+adr]`` was not reliable.

``` C++
const uint8_t table[] PROGMEM = { 123, 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20 };
```

``table[ 7 ]`` will work because the compiler will replace it with the constant value from the array.

``table[ 0x4000 + 7 ]`` will NOT work because the compiler will replace with constant 0 because out of bound.

``table[ adr ]`` will not work because the offset is missing.

``table[ 0x4000 + adr ]`` will work.

On the other hand :

``` C++
const uint8_t table[] PROGMEM = { 123, 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20 };

const uint8_t * pgm_data = (uint8_t*)((uint16_t)&data + 0x4000);
```

``pgm_table[ 7 ]`` will NOT work because the compiler will replace it with the constant 0 because it thinks it is out of bound.

``pgm_table[ adr ]`` will work.

----

Here is a sketch that show this beahavior :

```` C++

const uint8_t data[] PROGMEM = { 123, 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20 };

int data_len = sizeof( data );

const uint8_t * pgm_data = (uint8_t*)((uint16_t)&table + 0x4000);

void setup()
{
	//delay(3000);
	
	Serial.begin(9600);
	
	Serial.println("---"); // ---- display memory adresses 
	
	Serial.print( " data         = 0x" ); Serial.println( (uint16_t) data, HEX );
	Serial.print( "&data         = 0x" ); Serial.println( (uint16_t) &data, HEX );
	Serial.print( "&data[0]      = 0x" ); Serial.println( (uint16_t) &data[0], HEX );
	Serial.print( "&data[0x4000] = 0x" ); Serial.println( (uint16_t) &data[0x4000], HEX );
	Serial.print( " pgm_data     = 0x" ); Serial.println( (uint16_t) pgm_data, HEX );
	Serial.print( "&pgm_data     = 0x" ); Serial.println( (uint16_t) &pgm_data, HEX );
	
	
	Serial.println("---"); // ---- display reference
	
	Serial.print("pgm_read_byte( data ) = ");
	Serial.print( pgm_read_byte( data ) );
	Serial.println();
	
	Serial.print("pgm_read_byte( data + a ) = { ");
	for( int a=0; a < data_len ; a++ )
	{
		Serial.print( pgm_read_byte( data + a ) );
		Serial.print(", ");
	}
	Serial.println(" }");
		
	
	Serial.println("---"); // --- display data[ a ]
	
	Serial.print("data[ 0 ] = ");
	Serial.print( data[ 0 ] );
	Serial.println();
	
	Serial.print("data[ a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( data[ a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
	
	
	Serial.println("---"); // --- display table[ 0x4000 +a ]
	
	Serial.print("data[ 0x4000 ] = ");
	Serial.print( data[ 0x4000 ] );
	Serial.println();
		
	Serial.print("data[ 0x4000 + a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( data[ 0x4000 + a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
			
			
	Serial.println("---"); // --- display pgm_data[ a ]
	
	Serial.print("pgm_data[ 0 ] = ");
	Serial.print( pgm_data[ 0 ] );
	Serial.println();
	
	Serial.print("pgm_data[ a ] = { ");
	for( int a=0; a < data_len ; a ++ )
	{
		Serial.print( pgm_data[ a ] );
		Serial.print(", ");
	}
	Serial.println(" }");
	
		
	Serial.flush();
}

void loop()
{
}
````