dbuezas/lgt8fx #61: Vectors

Anobium, 2020-10-27:
Looking the code produced (show below) and the datasheet - how does the compiler and the chip know what to do? 

This code segment clearly shows the vectors address are 0,4,8,12 etc.  
```
00000000 <.sec1>:
   0: 0c 94 66 00   jmp 0xcc  ;  0xcc
   4: 0c 94 78 00   jmp 0xf0  ;  0xf0
   8: 0c 94 78 00   jmp 0xf0  ;  0xf0
   c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  10: 0c 94 78 00   jmp 0xf0  ;  0xf0
  14: 0c 94 78 00   jmp 0xf0  ;  0xf0
  18: 0c 94 78 00   jmp 0xf0  ;  0xf0
  1c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  20: 0c 94 78 00   jmp 0xf0  ;  0xf0
  24: 0c 94 78 00   jmp 0xf0  ;  0xf0
  28: 0c 94 78 00   jmp 0xf0  ;  0xf0
  2c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  30: 0c 94 78 00   jmp 0xf0  ;  0xf0
  34: 0c 94 78 00   jmp 0xf0  ;  0xf0
  38: 0c 94 78 00   jmp 0xf0  ;  0xf0
  3c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  40: 0c 94 2f 01   jmp 0x25e ;  0x25e
  44: 0c 94 78 00   jmp 0xf0  ;  0xf0
  48: 0c 94 78 00   jmp 0xf0  ;  0xf0
  4c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  50: 0c 94 78 00   jmp 0xf0  ;  0xf0
  54: 0c 94 78 00   jmp 0xf0  ;  0xf0
  58: 0c 94 78 00   jmp 0xf0  ;  0xf0
  5c: 0c 94 78 00   jmp 0xf0  ;  0xf0
  60: 0c 94 78 00   jmp 0xf0  ;  0xf0
  64: 0c 94 78 00   jmp 0xf0  ;  0xf0
```

This short section of the datasheet vector table shows the vector number and RJMP ( jmp as above) and empirical analysis shows the vectors address are 0,2,4,6,8 etc.  
```
0x000 RJMP      RESET
0x001 RJMP      EXT_INT0
0x002 RJMP      EXT_INT1
0x003etc 
```

Can some explain to me why the difference in vector addressing?

LaZsolt, 2020-11-02:
If AVR has flash memory size 16 or 32 kB then interrupt vectors are `jmp` instructions,
but if flash memory size 8 kB or less, then interrupt vectors are `rjmp` instructions.

Please close this issue if you are satisfied with my answer.

Anobium, 2020-11-03:
@LaZsolt 

Thank you.  

The question really relates the the use of rjmp or jmp in the context of the LGT bootloader.  

Context.  Setting the interrupt vectors to rjmp <location> (if in correct memory range) or just reti has proven to work for the Mega328p (also with the optibootloader).

So, I used the same approach for the LGT.  Failed.  I need to use jmp regardless.  I have proof that using reset vector with rjmp to a valid address simply fails.  Same approach for a mega328p works.

Any insights to the root cause of the issue?

LaZsolt, 2020-11-03:
@Anobium 
Oh. Sorry, I don't know how bootloader works. I can't help to you.

Anobium, 2020-11-03:
@LaZsolt  - Thank for helping already.  The memory range made total sense.

Empirical testing has shown the approach but there must be something in the use of these bootloaders in the LGT that is documented somewhere.  :-)

SuperUserNameMan, 2020-11-03:
@Anobium : I don't know if it will be of any help, but if you speak chinese or manage to translate it, there seems to be some information about LGT bootloader here : https://github.com/Edragon/LGT/tree/master/1-LGT8F328/SDK/Arduino/DOC

seisfeld, 2020-11-03:
I don’t speak Chinese, but that one pdf there looks like what I found out es well. It’s called virtual boot partition feature of optiboot. In short: When the bootloader receives the hex file it patches the reset and wdt vectors. It puts the address of the bootloader in the reset vector and the address of the Programm in the wdt vector. So on reset, always the bootloader kicks in and the bootloader the jumps to the wdt vector where it finds the real start address and starts the program.

closed 2022-04-03 by Anobium