dbuezas/lgt8fx #62: do we have an "official" community forum somewhere ?

SuperUserNameMan, 2020-10-31:
Hello,

Do we have an "official" community forum available somewhere so we could discuss and share LGT8 related information, idea, experiences and projects ? 

If not, could you, @dbuezas, open a dedicated repository on your Github where we could use the "Issues" as a forum board ?

dbuezas, 2020-11-01:
I like the idea!

https://github.com/dbuezas/lgt8fx-forum

seisfeld, 2020-12-24:
I suggest activating the discussions feature on this repo and remove the lgt8fx-forum again. This way it's all in the same place. Here's how its activated on the main settings page:

[image]

dbuezas, 2021-02-10:
I activated discussions now, thanks for the tipp Stephan!
Do you know if there is a way to import the issues from the forum repo?

seisfeld, 2021-02-10:
I don't know. Maybe when you first transfer the issues into the main repo and then transfer them into discussions? Would that be possible? I never worked with the discussion feature...

SuperUserNameMan, 2021-02-15:
Also, how do we enable notification to know when a new discussion or answer is created ?

https://github.com/dbuezas/lgt8fx/discussions

dbuezas, 2021-02-17:
To watch discussions, I hope this does the trick:
https://docs.github.com/en/github/managing-subscriptions-and-notifications-on-github/viewing-your-subscriptions#configuring-your-watch-settings-for-an-individual-repository

dbuezas, 2021-02-17:
@SuperUserNameMan:
[image]
[image]

dbuezas, 2021-02-17:
Closing this issue now.
All issues in the forum repo have been transferred to this one and converted into discussions.
Thanks for originating this idea and following up! 🥇

closed 2021-02-17 by dbuezas