dbuezas/lgt8fx #65: some third party library expect SoftwareSerial_h to be defined, but only SoftwareSerialLGT_h

SuperUserNameMan, 2020-11-06:
Hello @jg1uaa 

Apparently, some libraries expect macro ``SoftwareSerial_h`` to be defined after an ``#include <SoftwareSerial.h>``. (found that thanks to this post on the forum [here](https://github.com/dbuezas/lgt8fx-forum/issues/3) ).

Our brand new SoftwareSerial library defines ``SoftwareSerialLGT_h`` instead. 

Maybe could define ``SoftwareSerial_h`` too ?

jg1uaa, 2020-11-07:
sorry for late reply, I think this is need to fix but I don't know why LGT-specific SoftwareSerialLGT_h is defined.

If there is no problem, I want to replace SoftwareSerialLGT_h to (standard) SoftwareSerial.h. That code is modified for LGT8Fx but work same as for AVR's.

dbuezas, 2021-02-17:
@jg1uaa I think that's a great idea, go for it! :)

LaZsolt, 2022-01-06:
I don't want to be indecent. Would you make a pull request?

dwillmore, 2023-01-12:
Was this included and, if so, could this issue be closed?

dbuezas, 2023-01-12:
Yes, I see the change in master. Closing...

closed 2023-01-12 by dbuezas