dbuezas/lgt8fx #66: Wrong pinsouts piciture?

macron0, 2020-11-15:
Hi.  
I was wondering if maybe the picture with the pinout might be wrong?

If I read it correctly it seems to tell me that pin d2 has pwm. But after stumbling arround for a while this afternoon it seems it is not a pwm pin. 
My cheap dso150 scope just shows a flat line and when flipping it with digitalWrite I see  pwm blocks.
Checking an nano pinout it does not have pwm on D2.

So is the diagram flawed? Or does the board have pwm on D2 but do I need to do some trick to get it to work?
 
I am talking about this picture: https://raw.githubusercontent.com/dbuezas/lgt8fx/master/docs/boards/pinouts/svg/LGT8F328P-nano.png

dbuezas, 2020-11-15:
D2 is both bit 2 of PORTD, and bit 3 of port F. F3 can do pwm through Timer3, so this is probably not implemented in analogWrite yet. Good catch!

LaZsolt, 2020-12-05:
Can LGT8Fx Arduino core handle this type of wiring of the chip? I mean how could be prevent digitalWrite, analogWrite to make a short circuit on this type of pins?

maxint-rd, 2021-01-24:
When adding Timer3 support to my [FastPwmPin](https://github.com/maxint-rd/FastPwmPin) to get PWM on pin D2 of a QFP32 board, I found it works fine. However to get the port F output routed to the D2 pin, the output to the shared pin by PORTD needs to be disabled. In other words, PD2 needs to be set to input (DDRD:2 low) and PF2 needs to be set to output (DDRF:2 high).

wollewald, 2023-03-26:
@dbuezas - first of all, thanks for your great work!

Maybe it would be helpful to add an example sketch to show how to get a PWM signal at PF1 / PF2 like this one:

````
void setup(){ 
  // Clear OC3A on Compare Match / Set OC3A at Bottom; Wave Form Generator: Fast PWM 14, Top = ICR3
  TCCR3A = (1<<COM3A1) + (1<<WGM31); 
  TCCR3B = (1<<WGM33) + (1<<WGM32) + (1<<CS30); // prescaler = 1; 
  ICR3 = 15999;
  OCR3A = 3999;
  DDRF |= (1<<PF1);
} 
void loop() {} 
````

What I think is wrong in the pinout picture is that D8 is labelled as a PWM pin. PF3 / OC3C, which would be the ninth PWM Pin, is not available on the QFP32.

If you want you can use this pinout scheme which I created:

[image]

Another aspect that is related with the confusing PD1/PF1 and PD2/PF2 shared pins is 80mA output. I found this function needs to be activated in the HDR register. The output is at PF1, PF2, PF4, PF5, PD5, PD6 where PF4 an PF5 are not available on the QFP32. You don't get 80 mA through PD1 / PD2.

For example, in order to make 80 mA available at PF2, you could use the following code below, which is a kind of high current blink sketch.

````
/*
       HDR      Port/Pin    Pin Label (Nano)
       HDR0 --> PD5       --> D5
       HDR1 --> PD6       --> D6
       HDR2 --> PF1       --> TX
       HDR3 --> PF2       --> D2 
       HDR4 --> PE4 / PF4 --> none
       HDR5 --> PE5 / PF5 --> none 
*/

void setup(){    
/* example for activating PF2 for high current */
    HDR |= (1<<HDR3);
    DDRF |= (1<<PF2);  // ~ pinMode OUTPUT
}
  
void loop(){ 
    PORTF |= (1<<PF2); // Pin HIGH
    delay(1000);
    PORTF &= ~(1<<PF2); // Pin LOW
    delay(1000); 
}
````

dbuezas, 2023-03-26:
Oh, very nice pinout pic! Do you want to replace the one in the readme yourself (you can do it directly from GitHub btw)

Regarding the high current pins and timers, you may find this tool useful too: https://dbuezas.github.io/arduino-web-timers/#mcu=LGT8F328P&timer=3

wollewald, 2023-03-26:
Cool Tool! Timers of the AVR MCUs and on the LGT8FxP are always a bit painful. Therefore this is a great help.

I have just started a pull request for the new pinout scheme.

LaZsolt, 2023-03-26:
@wollewald 
Really good pinout picture. May I ask how to draw it? Where can I found a drawing tool?
I would like to draw some pinouts my other boards.

wollewald, 2023-03-26:
@LaZsolt, thanks. I used inkscape for all the boxes, text and lines. 
https://inkscape.org/?switchlang=en
Then I exported it as a png. Finally, I combined it with the nano board with Photoshop. Inkscape is not suitable for pixel based images, and I don't like Photoshop for vector images. That's why I took this rather complex route.
Below  you find the "svg" file of the inkscape image which you could use as a basis. 

[image]

dbuezas, 2023-03-26:
Thank you @macron0 , really useful pinout!
I added a link and pic to the timers tool to the readme too.

closed 2023-03-26 by dbuezas