dbuezas/lgt8fx #67: LGT8F328P MiniEVB not programming

leozzito, 2020-11-18:
Hello, my name is Leo. I'm having problem to program an LGT8F328P MiniEVB using a CP2102 USB UART adapter.

Details:
Arduino 1.8.13 environment and your support software 1.0.6
Board LGT8F328P MiniEVB
Clock source internal
Clock 16Mhz
Variant 328P-SSOP20 (even is not a SSOP chip). I tried different settings, still the same)
Arduino as ISP: Default (64). 
Port COM 6
Programmer AVR ISP

After compiling sketch, it stuck in uploading (no rx or tx led blinking) and after a couple of minutes it says Done.
But nothing has changed, no error messages. 

What's wrong?

Hope in your help. I got 30 of this boards and to program them.

Thanks in advance

Leo

SuperUserNameMan, 2020-11-18:
As you're referring to "COM6" I guess you're under Windows. I personally had some drivers issue when I was under Windows 8.1. Under Linux, everything is okay so far.

Regarding the variant, I recommend you keep using the correct one, because the default settings and the bootloaders are not the same between SSOP20 and LQFP32/48.

Quick check-list :
- [ ] are you sure the CP2102 drivers are correctly installed ? (does it work with other Arduino boards ?)
- [ ] does the Blue LED of the MiniEVB blinks when powered up ?
- [ ] are you sure CP2102.RX is connected to MiniEVB.TX, and CP2102.TX is connected to MiniEVB.RX ? 
- [ ] are you sure you have selected the correct COM port ?
- [ ] does it display some useful error message if you enable "File / Settings / Display detailed results during [_]  uploading" ?
- [ ] does it display some useful error message if you enable "File / Settings / Display detailed results during [_] compilation" ?
- [ ] does it display some useful error message if you enable "File / Settings / Compiler Warning / All" ?

leozzito, 2020-11-18:
Many Thanks for your prompt reply. I'm running Windows 7, I do not use Linux.

**> Regarding the variant, I recommend you keep using the correct one, because the default settings and the bootloaders are not the same between SSOP20 and LQFP32/48.**
I installed the board from your json link. Board present on IDE are:
328P-LQFP48 miniEVB
328P-LQPF32 (e.g. MiniEVB nanostyle and WAVGAT)
328P-LQPF32 wemos-TTGO-XI
328D rare
328P_SS0P20 (e.g. green pseudo pro mini)

From the seller page I need LGT8F32P-LQFP32 MiniEVB
[image]

> **are you sure the CP2102 drivers are correctly installed ? (does it work with other Arduino boards ?)**
Yes. I use the CP2102 with Arduino Pro mini (168p or 328p) and it works without problem.

**> does the Blue LED of the MiniEVB blinks when powered up ?**
Yes, it blinks. At the same time on the CP2102 rx and tx leds are blinking.

**>  are you sure CP2102.RX is connected to MiniEVB.TX, and CP2102.TX is connected to MiniEVB.RX ?**
Yes they are.

>  **are you sure you have selected the correct COM port ?**
[image]

**>  does it display some useful error message if you enable "File / Settings / Display detailed results during [_] uploading" ?**
it stuck on this
[image]

**>  does it display some useful error message if you enable "File / Settings / Display detailed results during [_] compilation" ?**
it stuck on this
[image]

**>  does it display some useful error message if you enable "File / Settings / Compiler Warning / All" ?**
it stuck on this
[image]

As variant in these pictures, I used 328P-LQFP48 MiniEVB

SuperUserNameMan, 2020-11-18:
This is strange. I do use a CP210x adapter too.

- [ ] This is this board, correct ?
https://github.com/dbuezas/lgt8fx/blob/master/docs/boards/pro_mini_LQFP32.png

- [ ] If yes, you have not forgot to connect the DTR pin bellow the TXD labelled on the back of this board ? (on some boards, the label is damaged and might be confused with a GND)
- [ ] If yes again, you have not confused it with the RAW pin next to it neither ?

- [ ] if you don't connect your adapter to the pins labelled on the back, and if you connect it directly to the TX/RX pins labelled in front instead, does it help if you press the RESET button when avrdude is doing "attempt x/10" in the console ?

LaZsolt, 2020-11-19:
Sometimes the 3.3V powered serial adapter cannot drive 5V powered MCU RESET pin. In this case use Reset button, mentioned above.
http://nerdralph.blogspot.com/2020/10/lgt8f328p-edmini-board.html

xuancong84, 2022-06-17:
I have tested extensively on LGT8F328P MiniEVB. I am running Debian-based Linux x86-64.
In general, the chip works. However,
- certain functions of the chip is buggy or does not work at all, especially the sleep (DPS1 and DPS2) and watchdog interrupt
- when your code deals with chip power configurations such as sleep, power state, etc., you must put a `delay(200)` at the beginning of `setup()`, otherwise, upload might fail. This is because serial upload requires UART clock, core clock, interrupts, etc. If your code have modified or disabled them too soon, then upload will fail.

LaZsolt, 2023-05-21:
There were no problem reported on this topic since 2020.
In my opinion it can be closed.

closed 2023-05-21 by dbuezas