dbuezas/lgt8fx #70: error: macro "nop" passed 1 arguments, but takes just 0

rbonat, 2020-12-17:
Hi,

Trying to compile Tiny4kOLED example "TinyFontDemo", i've got this error:
..Documents\Arduino\libraries\Tiny4kOLED\src/Tiny4kOLED_common.h:141:16: error: macro "nop" passed 1 arguments, but takes just 0

   void nop(void);

Same example can be compilled without errors on any boards i've tested, even on ATTinyCore ;)

Any idea how can I make it working on LGT8F32?

LaZsolt, 2020-12-17:
The reason of error message is, in around of end of file `C:\Users\(yourusername)\AppData\Local\Arduino15\packages\LGT8fx Boards\hardware\avr\1.0.5\cores\lgt8f\Arduino.h` a `nop()` definition. Try to remove this definition from this `Arduino.h`

Tiny4kOLED library designed for an ATTiny85. Please send a message if this library work with LGT8Fx. :)

rbonat, 2020-12-17:
Thanks for such a rapid and accurate response ! 👍 
I was so close, but trying to change  C:\Program Files (x86)\Arduino\hardware\arduino\lgt8f\cores\lgt8f\arduino.h  instead!

Now library works perfect, thanks so much!
And I've learned where to find board definition files used by arduino IDE :)
By curiosity, is there any reason why definition of nop on this board is not compatible by default?

LaZsolt, 2020-12-17:
This nop definition used in USB HID, in this package only. So I think for compatibility it can be removed from Arduino.h

rbonat, 2020-12-17:
I see. ok, thank you again.
Problem solved, Tiny4kOLED is working on LGT8F328, now I can start playing with embedded opamp :)

closed 2020-12-17 by rbonat