dbuezas/lgt8fx #72: warnings if including <Arduino.h>

5chufti, 2020-12-26:
including <Arduino.h> brings a warning about redefining PRR
```
In file included from d:\arduino\hardware\tools\avr\avr\include\avr\io.h:272:0,
                 from d:\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,
                 from D:\arduino\portable\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/Arduino.h:28,
                 from C:\Users\Gogo\AppData\Local\Temp\arduino_build_10071\sketch\DCC.h:21,
                 from C:\Users\Gogo\AppData\Local\Temp\arduino_build_10071\sketch\DCC.cpp:20:
d:\arduino\hardware\tools\avr\avr\include\avr\iom328p.h:370:0: note: this is the location of the previous definition
 #define PRR _SFR_MEM8(0x64)
```
this can easily be solved by this simple patch in lgtx8p.h@490
```
//  Power Reduce Register
#ifdef PRR
  #undef PRR
#endif
```
sorry, too lazy to create pullrequest for 3 lines of code ...

LaZsolt, 2020-12-26:
Other solution is to delete the PRR (re)definition line. This IO register completely same as in ATmega328P.
But still need to delete the following 7 lines. This bit definitions are also duplicated.

5chufti, 2021-02-16:
still open ....

LaZsolt, 2021-02-16:
The corrected file is here: https://github.com/LaZsolt/lgt8fx/blob/master/lgt8f/cores/lgt8f/lgtx8p.h

5chufti, 2021-04-05:
there now is a new one:
`d:\arduino\portable\packages\lgt8fx boards\hardware\avr\1.0.6\variants\standard\pins_arduino.h:102:22: error: redefinition of 'const uint8_t A10'
 static const uint8_t A10 = 25;
` 
referring to the A10

LaZsolt, 2021-04-05:
LGT8F328P has 12 analog inputs. Its name begin with A0 and the last is A11.
So A10 is the name of an analog input pin. This pin label on the MiniEvb board is AREF, but can be used as A10 or PE6.

So there are 2 solution for you. 
- rename your A10 variable/constant/function or
- delete the definiton of A10 from your ...\variants\standard\pins_arduino.h

5chufti, 2021-04-05:
it's not one of my definitions, it is interfering with another definition in lgt package.
(sorry I forgot to copy that part of the error message)
`In file included from D:\arduino\portable\packages\LGT8fx Boards\hardware\avr\1.0.6\variants\lgt8fx8p48/pins_arduino.h:40:0,
                 from D:\arduino\portable\packages\LGT8fx Boards\hardware\avr\1.0.6\cores\lgt8f/Arduino.h:626,
                 from C:\Users\Gogo\AppData\Local\Temp\arduino_build_153566\sketch\RB_DCC_Sniffer.ino.cpp:1:
d:\arduino\portable\packages\lgt8fx boards\hardware\avr\1.0.6\variants\standard\pins_arduino.h:102:22: error: redefinition of 'const uint8_t A10'
 static const uint8_t A10 = 25;
`

LaZsolt, 2021-04-06:
Okay. Thank you for reporting this warning. Corrected: https://github.com/dbuezas/lgt8fx/pull/134
You need to mark the full line 98 in comment in file: `...\variants\standard\pins_ardunio.h`

dwillmore, 2023-01-12:
@LaZsolt is this now resolved?  If so, please close this issue.

LaZsolt, 2023-01-12:
Yes, resolved.
The DCC sniffer sketch has a variable name A10 which is the same as with an Arduino pin name.
The same naming collision may occur using Arduino Mega board. So the reason of issue came from the DCC sniffer sketch.

closed 2023-01-12 by LaZsolt