dbuezas/lgt8fx #75: Looking for help maintaining the repo (merges and releases)

dbuezas, 2021-02-07:
I'm unfortunately very short on time lately, does any of you @seisfeld @jayzakk  @jg1uaa @SuperUserNameMan @LaZsolt  have interest?
The release process is basically running a script, pushing and dropping the result in github releases. If there is interest I can would explain and put it in the readme.

Thanks in advance, I'm humbled by the good ideas and effort you and others put into PRs and issues!

SuperUserNameMan, 2021-02-07:
I'm not good at all with git, but you can give me access to the process just in case (I will be the last resort).

I was worried not seeing you as often here. So I hope everything is okay for you.

dwillmore, 2021-02-07:
If no one else wants to do this, I can.

jg1uaa, 2021-02-08:
How about testing before merging?
For example, I used I2C-based OLED display and SPI-based LoRa communication module to test I2C/OLED driver code.
Due to my equipment, I cannot perform difficult tests, but maybe I can help about simple testing...

dbuezas, 2021-02-08:
> How about testing before merging?

Yes testing is an issue, I don't see a full solution for this, only "best effort" plus looking at issues.

jayzakk, 2021-02-09:
I'm damn busy since that "C" thing, but may be able to help out with some testing (at weekends).

For merges, I can help out.
For (the more compex) release, if you write down your usual walk-through, everyone with time may be able to fetch open topics/basic tests in some agile way and end up with some release.

Thanks for your trust in us 👍

dbuezas, 2023-01-11:
Some of you have rights to merge PRs, releases will be triggered automatically now.

closed 2023-01-11 by dbuezas