dbuezas/lgt8fx #78: Undocumented registers?

LaZsolt, 2021-02-13:
I found two IO registers, which seems to me, are undocumented.

[image]

In my opinion:
- OCPUE is a register somehow can control port E pull-up resistors. I found this register's definition in file `lgtx8e.h`, next to port E definition.
- LDOCR is could be a register which may control the internal LDO (Low-Dropout Regulator). VSELx bits may select the LDO output voltage, and PDEN bit disable the LDO.

jayzakk, 2021-02-13:
OCPUE may be related to this part of the datasheet, which tries to explain all that pin multiplexing shi.. uhm... features:

[image]

PUOE - Pull-up multiplexing enable
If this bit is 1, the pull-up enable is controlled by PVOV; if this bit is 0, the pull-up enable is controlled by DDxn, PORTxn and PUD

PUOV - Pull up multiplexing value
If PUOE is 1, this bit is 1 to enable the pin's pull-up resistor, otherwise it will disable the pin's pull-up resistor

The LGT8-**D** datasheet has some interesting relation-tables, which are missing in the **P** datasheet. They may help sort out things:
[image]

But it does NOT conain a registern named OCPUE (but also LDOCR, without any description).

LaZsolt, 2021-02-14:
This discussion moved here: https://github.com/dbuezas/lgt8fx/discussions/81#discussion-3222825

closed 2021-02-14 by LaZsolt