dbuezas/lgt8fx #82: is it just me ....

5chufti, 2021-02-16:
maybe I am stupid or ignorant about githubs features but on all other repositories when I go to download I get the current master branch. For this repo I only get the old 1.0.6 release.
and btw, is it necessary to bloat the download with all the old version .zip archives?
as old versions are allways available via the releases section, or no?

LaZsolt, 2021-02-16:
I think something not workig good.
I did a synchronize my fork with the dbuzeas/lgt8fx repository. (I don't know how, because now I can't find any button for this function) After synchronize, a message appeared in the same line with the synchronized item:
[image]

After that I modified 2 files and create a pull request. The github system did not consider the synchronized items.

LaZsolt, 2021-02-16:
And the dowload of master behaved the same as at you did. I was suprised.

dbuezas, 2021-02-17:
You are right with the full branch download, that surprises me also. Maybe it is linked to releases?
I also agree with cleaning up the older zips from the repo.

dbuezas, 2023-01-11:
It looks like I had a tag called master (like the branch), I deleted it now and this issue should be solved.

closed 2023-01-11 by dbuezas