dbuezas/lgt8fx #83: double definition in "variants/standard/pins_arduino.h"

5chufti, 2021-02-16:
the double definition of orig lines 443-446 in lines
```
216 #define digitalPinToPCICR(p)    (((p) >= 0 && (p) <= 21) ? (&PCICR) : ((uint8_t *)0))
217 #define digitalPinToPCICRbit(p) (((p) <= 7) ? 2 : (((p) <= 13) ? 0 : 1))
218 #define digitalPinToPCMSK(p)    (((p) <= 7) ? (&PCMSK2) : (((p) <= 13) ? (&PCMSK0) : (((p) <= 21) ? (&PCMSK1) : ((uint8_t *)0))))
219 #define digitalPinToPCMSKbit(p) (((p) <= 7) ? (p) : (((p) <= 13) ? ((p) - 8) : ((p) - 14)))
```
leads to compilation errors

LaZsolt, 2021-03-22:
I made a pull request to resolve this warnings. https://github.com/dbuezas/lgt8fx/pull/122#issue-598056535

jayzakk, 2021-03-23:
Merged, thank you all!

closed 2021-03-23 by jayzakk