dbuezas/lgt8fx #89: RemoteXY won't compile wiht lgt8f

navr32, 2021-02-17:
Hi all !
Many thanks.
And congratulation for all the job done by all of contributors. It's woohoo...
I have done some little project with the lgt8f328 since some time and start with the wavgat package first..after i found yours and this very very good and promising for futur. Now i try to do more big project and want to drive little HMI with bluetooth by RemoteXY throught my phone with softwareserial but failing..

When i build the remoteXY for  the Atmega328p all is ok..but if i try with the Lgt8f328 board set
the build crash..here the report..someone have an idea ?
I have test with Arduino 1.8.10 to 1.8.13...and RemoteXY ver 2.4.6 and 2.4.5 install from the Arduino IDE libraries manager
Here the output of build
****************************************************************************************************************************

In file included from /home/nico/Arduino/libraries/RemoteXY/src/RemoteXY.h:168:0,
                 from /home/nico/Arduino/libraries/RemoteXY/examples/Bluetooth/Bluetooth_HC05_SoftSerial/Bluetooth_HC05_SoftSerial.pde:35:
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h: In constructor 'CRemoteXY::CRemoteXY(const void*, void*, const char*, uint8_t, uint8_t, long int)':
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:18:5: error: 'initSerial' was not declared in this scope
     initSerial (_serialRx, _serialTx, _serialSpeed);
     ^~~~~~~~~~
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:18:5: note: suggested alternative: 'Serial'
     initSerial (_serialRx, _serialTx, _serialSpeed);
     ^~~~~~~~~~
     Serial
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h: In member function 'virtual void CRemoteXY::sendByte(uint8_t)':
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:26:5: error: 'serial' was not declared in this scope
     serial->write (b);
     ^~~~~~
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:26:5: note: suggested alternative: 'Serial'
     serial->write (b);
     ^~~~~~
     Serial
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h: In member function 'virtual uint8_t CRemoteXY::receiveByte()':
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:33:17: error: 'serial' was not declared in this scope
     uint8_t b = serial->read ();
                 ^~~~~~
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:33:17: note: suggested alternative: 'Serial'
     uint8_t b = serial->read ();
                 ^~~~~~
                 Serial
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h: In member function 'virtual uint8_t CRemoteXY::availableByte()':
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:42:12: error: 'serial' was not declared in this scope
     return serial->available ();
            ^~~~~~
/home/nico/Arduino/libraries/RemoteXY/src/modules/serial.h:42:12: note: suggested alternative: 'Serial'
     return serial->available ();
            ^~~~~~
            Serial
exit status 1
Erreur de compilation pour la carte LGT8F328

******************************************************************************************************************************

***************************************************************************************************************************
the code example:

```C++
RemoteXY example. 
  Smartphone connect via bluetooth HC-05 module
  (software serial connected).

  This shows an example of using the library RemoteXY.
  In the example you can control the LED pin 13 using the button on the 
  smartphone. The example uses the SoftwareSerial library. You need to 
  connect Bluetooth module HC-04(05,06,07) or similar specified contacts: 
  Pin 2(RX) ---  TX Bluetooth module           
  android app, as it will determine which interface you have 
  downloaded the arduino.
  
*/

///////////////////////////////////////////// 
//        RemoteXY include library         // 
///////////////////////////////////////////// 

/* RemoteXY select connection mode and include library */ 
#define REMOTEXY_MODE__HC05_SOFTSERIAL 
#include <SoftwareSerial.h> 
#include <RemoteXY.h> 
  Pin 3(TX) ---  RX Bluetooth module    
  
  Download the mobile app from the 
  website: http://remotexy.com/download/ for connect this sketch.
  
  Use the website http://remotexy.com/ to create your own management 
  interface your arduino with your smartphone or tablet.
  You can create different management interfaces. Use buttons, 
  switches, sliders, joysticks (g-sensor) all colors and sizes 
  in its interface. Next, you will be able to get the sample 
  code for arduino to use your interface for control from a 
  smartphone or tablet. You will not need to re-install the 

/* RemoteXY connection settings */ 
#define REMOTEXY_SERIAL_RX 2 
#define REMOTEXY_SERIAL_TX 3 
#define REMOTEXY_SERIAL_SPEED 9600 

/* RemoteXY configurate  */ 
unsigned char RemoteXY_CONF[] = 
  { 1,0,11,0,1,5,1,0,21,2
  ,59,59,2,88,0 }; 
   
/* this structure defines all the variables of your control interface */ 
struct { 

    /* input variable */
  unsigned char button_1; /* =1 if button pressed, else =0 */

    /* other variable */
  unsigned char connect_flag;  /* =1 if wire connected, else =0 */

} RemoteXY; 

///////////////////////////////////////////// 
//           END RemoteXY include          // 
///////////////////////////////////////////// 

#define PIN_BUTTON_1 13

void setup()  
{ 
  RemoteXY_Init ();  
   
  pinMode (PIN_BUTTON_1, OUTPUT);
   

  // TODO you setup code 
   
} 

void loop()  
{  
  RemoteXY_Handler (); 
   
  digitalWrite(PIN_BUTTON_1, (RemoteXY.button_1==0)?LOW:HIGH);
   

  // TODO you loop code 
  // use the RemoteXY structure for data transfer 

}

``

dwillmore, 2023-01-12:
Did this change get incorporated into the library at some point?  If so, please close this issue.

dbuezas, 2023-01-12:
Yes, it's the same as #65

closed 2023-01-12 by dbuezas

luvaibhuriwala, 2023-05-14:
I am facing the same issue but defining softwareserial didn't solve it