dbuezas/lgt8fx #107: add documentation for lgt8f88a variant

SuperUserNameMan, 2021-02-19:
Hello,

https://www.avrfreaks.net/sites/default/files/forum_attachments/LGT8F88A%20V1_1.pdf

SuperUserNameMan, 2021-02-19:
Migrating from Atmega88 to LGT8F88A v1.0 : https://www.5hertz.com/pdfs/40001_DI2.pdf

SuperUserNameMan, 2021-02-20:
A full chinese datasheet of 328d variants :

http://wiki.ocrobot.com/lib/exe/fetch.php?media=ocrobot:alpha:8f328d:lgt8fx8d-series-databook-v1.0.5.pdf

LaZsolt, 2021-02-20:
A full, chinese databook v.1.1.1 of LGT8F88A: [LGT8F88A_CH_databook_v1.1.1.pdf](https://github.com/dbuezas/lgt8fx/files/6014923/LGT8F88A_CH_databook_v1.1.1.pdf)

LaZsolt, 2023-01-24:
LGT8F328D documents added:
https://github.com/dbuezas/lgt8fx/wiki#docs--links

LGT8F88A databook not uploaded because of LGT8F88x not supported.

closed 2023-01-24 by LaZsolt