dbuezas/lgt8fx #108: LGT8F88A core : SERIAL_RX_BUFFER_SIZE not defined ?

SuperUserNameMan, 2021-02-19:
Hello,

I was checking if my EEPROM library was compiling with the LGT8F88A variant.

Apparently it is somehow missing ``SERIAL_RX_BUFFER_SIZE`` definition :

````
home/xxx/.arduino15/packages/LGT8fx Boards/hardware/avr/1.0.6/cores/lgt8f/HardwareSerial.h:61:28: error: operator '>' has no left operand
 #if  (SERIAL_RX_BUFFER_SIZE>256)
````

By the way, do you thin it is really worth it to keep support for the LGT8F88A ?

LaZsolt, 2021-02-19:
Few days ago I noticed that compiling error too. And I don't understand why. At least in line 51 or in line 53 was SERIAL_RX_BUFFER_SIZE defined, if was not defined as a garbage earlier. I will try to insert a valid value into boards.txt, because not found this in LGT8F88A section.
```
49  #if !defined(SERIAL_RX_BUFFER_SIZE)
50  #if ((RAMEND - RAMSTART) < 1023)
51  #define SERIAL_RX_BUFFER_SIZE 16
52  #else
53  #define SERIAL_RX_BUFFER_SIZE 64
54  #endif
55  #endif
 ...
61  #if  (SERIAL_RX_BUFFER_SIZE>256)
62  typedef uint16_t rx_buffer_index_t;
63  #else
64  typedef uint8_t rx_buffer_index_t;
65  #endif
```

> By the way, do you think it is really worth it to keep support for the LGT8F88A ?
I think support not really necessary.

LaZsolt, 2021-02-19:
> I was checking if my EEPROM library was compiling with the LGT8F88A variant.

You wasn't checked compiling with LGT8F88A, because you selected LGT8F88D-SSOP20.
"A" variant not selectable from the menu. Look inside of `boards.txt`

SuperUserNameMan, 2021-02-19:
> > I was checking if my EEPROM library was compiling with the LGT8F88A variant.
> 
> You wasn't checked compiling with LGT8F88A, because you selected LGT8F88D-SSOP20.
> "A" variant not selectable from the menu. Look inside of `boards.txt`

Okaaaaay .... also i did not notice it was written LGT8F88**D**-SSOP20 into the Arduino IDE menu "Tool/Variant" ....

Also it is confusing that the ``lgt8fx8ds20/pins_arduino.h`` defines it as a ``__LGT8FX8E__``.

Well, I will recheck my EEPROM PR again later.

LaZsolt, 2021-02-19:
My idea is to delete LGT8F88x support from the Menu, because of many conflict of new interrupts feature.

SuperUserNameMan, 2021-02-20:
> My idea is to delete LGT8F88x support from the Menu, because of many conflict with new interrupts feature.

Agree.
Also, there is no LGT8F88x available on Aliexpress ... 

A few WEMOS XI using LGT8F328d can still be found there, but Wemos say they are retired. Maybe we could get rid of the LGT8F328d as well given that WEMOS already provide support for them ? 

And maybe we could focus on the LGT8F328p ...

LaZsolt, 2021-02-20:
I will launch a pull request soon to remove LGT8F88x support from this package.
But first I need to test my earlier modifications of **lgtx8e.h** and **lgtx8f.h** files.

youxiaojie, 2021-03-31:
> Agree.
> Also, there is no LGT8F88x available on Aliexpress ...
>
> A few WEMOS XI using LGT8F328d can still be found there, but Wemos say they are retired. Maybe we could get rid of the LGT8F328d as [...]

328d is popular in china. a lot of it in taobao.com

dwillmore, 2023-01-12:
@LaZsolt did you commit the removal of LGT8F88x support? If so, please determine if this issue is relevant anymore.  Thank you.

LaZsolt, 2023-01-19:
This issue not relevant anymore, because support for LGT8F88x has been removed and has been unavailable for purcase for a while.

closed 2023-01-19 by LaZsolt