dbuezas/lgt8fx #109: remove support of lgt8f88x and lgt8f328d/e ?

SuperUserNameMan, 2021-02-20:
Hello, 

as discussed here #108, we were wondering if it is worth it to keep support of the 88x, 328d and 328e variants ?

My point is :
- 88x : this variant seems to be not available anywhere (could not find on Aliexpress)
- 328d  : is only available as Wemos XI board which is marked as retired by Wemos (and they already provide support for it)
- 328e : does it really exist ?

Maybe we could focus on the 328p variants only ?

Does anyone use boards with other MCU than lgt8f328p ?

SuperUserNameMan, 2021-02-20:
( @dbuezas : poke ! )

LaZsolt, 2021-02-20:
1.) I think 328d and 328e is a different name of 328d 
In file <b>lgtx8e.h</b> the definition:
```
#ifndef _IOLGT8F88D_H_
#define _IOLGT8F88D_H_
```

2.) Boards with Larduino brand names equipped with LGTx8D MCU.
| Red Larduino uno | Green Ladruino uno |  Larduino mini (BTE17-10) | Larduino nano (BTE17-11) |
| - | - | - | - |
| [image] | [image] | [image] |  [image] |

- and boards with Massduino names equipped with LGTx8D MCU too.

SuperUserNameMan, 2021-02-20:
About Massduino boards : 
- http://www.inhaos.com/products.php?cPath=136
- http://www.inhaos.com/product_info.php?products_id=155
- https://www.aliexpress.com/item/32853817704.html
- https://www.aliexpress.com/item/32653405632.html

They say they use a MD-328D mcu : is it different clone or a rebranded LGT8F328D ?

--
Regarding Larduino : I could not find any board on aliexpress, but the [LGTMCU/Larduino_HSP](https://github.com/LGTMCU/Larduino_HSP) redirects to [OCROBOTS](http://wiki.ocrobot.com/doku.php?id=en:products) where we can find these :
- http://wiki.ocrobot.com/doku.php?id=en:ocrobot:alpha:8f328d-u:index (datasheet available)
- ALPHA 8F328D-U : http://wiki.ocrobot.com/doku.php?id=ocrobot:alpha:8f328d-u:index
- ALPHA 8F328D : http://wiki.ocrobot.com/doku.php?id=ocrobot:alpha:8f328d:index
- ALPHA 8F328D-RF : http://wiki.ocrobot.com/doku.php?id=ocrobot:alpha:8f328d-rf:main#dokuwiki__top

jg1uaa, 2021-02-20:
Before reading this issue, I ordered MassDuino(https://www.aliexpress.com/item/32655296623.html) and it might be arrived in next month if there should be no derivery problem.
I could not find the evidence that MD328D is completely same as LGT8F328D. But I think MD328D is close to LGT8F328D by comparing bootloader code.

And I ordered WAVGAT's Arduino R3 with WAVGAT-printed processor (https://www.aliexpress.com/item/4000587200875.html) 10 days ago. Do I have to check WAVGAT's BSP to identify LGT8F328-D or -P?

SuperUserNameMan, 2021-02-20:
@jg1uaa : 
About the MassDuino, if they provide you datasheet to MD328D, that would be interesting.

About the WAVGTAT board, the aliexpress page says it is ATmega328. Maybe a rebranded genuine ATmega ? maybe counterfeit clone ? maybe recycled ? ...
If they don't tell you the truth about it, a benchmark might help you find if it is a real ATmega, a real LGT or an other clone version ...

And BTW : I'd be interested to see the results of the [benchmarks here]( https://github.com/dbuezas/lgt8fx/discussions/77) on your boards once you'll have them !

SuperUserNameMan, 2021-02-20:
@jg1uaa :
Actually, the Wavgat MCU is printed : AVGA328P

SuperUserNameMan, 2021-02-20:
MD328D datasheet : https://cdn.instructables.com/ORIG/F5W/QO0M/IN4W2GA5/F5WQO0MIN4W2GA5.pdf

LaZsolt, 2021-02-20:
@jg1uaa :
> Do I have to check WAVGAT's BSP to identify LGT8F328-D or -P?

LGT8F328P core could work on 32 MHz.
LGT8F238**D** maximum working frequency is 20 MHz.
 So if you set the internal clock prescaler without dividing at least 2, the CPU won't work.

pixelwaster, 2021-02-20:
WAVGAT's Arduino R3 with WAVGAT-printed processor (and the Nano) are LGT8F328P chips. There is a link on the page for driver updates. The bootloader is optiboot_lgt8f [88/328] {a/d/ds/e/p}

jg1uaa, 2021-02-20:
Thanks everyone, I will test #77 benchmarks when I have received boards.
Programming sketch that works at 32MHz will be also good test... ok.

We can get MD328D stuff from http://www.inhaos.com/product_info.php?products_id=156#fragment-2 and datasheet is at http://www.inhaos.com/uploadfile/otherpic/DS-MD-328D-V01-20160412.pdf but this is same one (DOC ID:DS-MD-328D-V01-20160412) introduced by previous comment.

I hope SWDICE mkII/SWDISP will tell correct processor name to us, maybe this test needed.

jg1uaa, 2021-03-08:
Today I got MassDuino MD328D board and I am testing now.
[image]
[image]

To program MassDuino board, avrdude requires to set 115200bps, not 57600bps like this.
`/home/pi/arduino-1.8.13/hardware/tools/avr/bin/avrdude -C /home/pi/arduino-1.8.13/hardware/tools/avr/etc/avrdude.conf -v -c arduino -p atmega328p -P /dev/ttyUSB0 -b 115200 -D -Uflash:w:bench_primenumbers.ino.hex:i
`

I used LGT8Fx (our project) to build sketch and I set variant as 328P-LQFP32 wemos-TTGO-XI.  Variant 328D(rare) did not work.

the result of prime number benchmark(bench_prime numbers) is:
118sec @ 16MHz (UART 9600bps)
236sec @ 8MHz (UART 19200bps curiously)
did not work at 32MHz

And finally I tested tiny uDSC test (http://www.uaa.org.uk/gomitext/2021/20210301/udsc_test.ino), there is no evidence of uDSC.
At this point, I think MD328D is LG8F328"D" and we have to some work for supporting this board.

I have not connected SWDICE mkII/SWDISP yet, is this test needed?

SuperUserNameMan, 2021-03-08:
@jg1uaa : 
> At this point, I think MD328D is LG8F328"D" and we have to some work for supporting this board.

I think you're going to become the MD-328D and LGT8F328-D specialist from now.

If you want to improve support for this D variant, I wrote this wiki page that might help you with the preprocessor :
https://github.com/dbuezas/lgt8fx/wiki/Identifying-MCU-and-board-variants-using-preprocessor

jg1uaa, 2021-03-11:
Have anyone tested D variant already?
I tried to run Blink.ino, something strange... the code turns LED on and off with each 1000ms delay, but it looks about 2000ms. I changed LGT8Fx's optiboot_lgt8f328d_extosc.hex bootloader from MassDuino's default, same result.

postscript:
I found Green Ladruino uno at taobao, https://s.taobao.com/search?q=larduino+lgt8f328d

SuperUserNameMan, 2021-03-11:
@jg1uaa  : @LaZsolt  mentionned issues with D variants here : https://github.com/dbuezas/lgt8fx/pull/64#issuecomment-783633366

LaZsolt, 2021-03-11:
@jg1uaa 
I ordered a Massduino nano. Perhaps it will arrive next week.
Try to run the blink sketch both D and P variant parallel.

jg1uaa, 2021-03-28:
This issue is related to #118, bootloader.
Currently there is five bootloaders; for lgt8f88a, lgt8fx8ds20, lgt8fx8e, lgt8fx8p and lgt8fx8ps20. If dropping support for 88x series, we can maintain bootcode easier. 
I think source tree for 328D, 328P and 328P-SSOP20 can be merged.

LaZsolt, 2021-03-28:
@jg1uaa 
Yes, I agree.

youxiaojie, 2021-03-31:
> Have anyone tested D variant already?
> I tried to run Blink.ino, something strange... the code turns LED on and off with each 1000ms delay, but it looks about 2000ms. I changed LGT8Fx's [...]

I have 328d how to test?
l just test eeprom seems 512bytes available. howto set it to 1k?

LaZsolt, 2021-04-01:
I found an article which tells how to modify the Arduino bootloader of LGT8F328P to support a baud rate of 115200bps and adjusting the system clock frequency to decrease baud rate error.
https://www.programmersought.com/article/44233408885/

youxiaojie, 2021-04-01:
> I found an article which tells how to modify the Arduino bootloader of LGT8F328P to support a baud rate of 115200bps and adjusting the system clock frequency to decrease baud rate error.
> https://www.programmersought.com/article/44233408885/
```

Error 1020 Ray ID: 639393675adf3b3f • 2021-04-01 17:35:45 UTC
Access denied
What happened?

This website is using a security service to protect itself from online attacks.

Cloudflare Ray ID: 639393675adf3b3f • Your IP: 240e:82:2f00:• Performance & security by Cloudflare
```
why?

LaZsolt, 2021-04-01:
@youxiaojie :

I don't know why but
try to open https://www.programmersought.com and type into the search bar `bootloader of LGT8F328P`

youxiaojie, 2021-04-03:
MD328d is the same as lgt8f328d,
md3248p = 48pin lgt8f328p
from lgt official web shop.

dwillmore, 2023-01-12:
Has a consensus been reached on this issue?

LaZsolt, 2023-01-24:
Yes, consensus has been reached because in v2.0.0 LGT8F328P and D are supported but none of the LGT8F88x variants.

closed 2023-01-24 by LaZsolt