dbuezas/lgt8fx #118: optiboot_lgt8f328d.hex corrupt?

jg1uaa, 2021-03-12:
Blink.ino could not boot with optiboot_lgt8f328d.hex, others boots.

I checked .hex's MD5 to check bootloader, all files are different.

```
MD5 (optiboot_lgt8f328d.hex) = 5829aff59380b60bdd1ec77dc5d6402a
MD5 (optiboot_lgt8f328d_extosc.hex) = 0640fe83172c27f832ea30179657ffd0
MD5 (optiboot_lgt8f328d_intosc.hex) = b0c8f030204ecf1c3146e49e50811226

```
I also checked LGTMCU/Larduino_HSP(https://github.com/LGTMCU/Larduino_HSP)'s bootloader hex.

```
MD5 (optiboot_lgt8f328d.hex) = b0c8f030204ecf1c3146e49e50811226
MD5 (optiboot_lgt8f328d_extosc.hex) = 0640fe83172c27f832ea30179657ffd0
MD5 (optiboot_lgt8f328d_intosc.hex) = b0c8f030204ecf1c3146e49e50811226
```
I think our optiboot_lgt8f328d.hex should be same as optiboot_lgt8f328d_intosc.hex like Larduino_HSP, what's happened?

FYI:
Due to -mshort-calls option of gcc is obsoleted, optiboot requires old gcc (4.3.2; arduino-1.5.6 and before). Shold we simply copy optiboot_lgt8f328d_intosc.hex to optiboot_lgt8f328d.hex, or overhaul for modern environment?

jg1uaa, 2021-03-21:
here is the test that compiling LGT8F328**D** bootloader on avr-gcc-7.3.0
internal OSC, 16MHz, 57600bps configuration only (no plan to support external OSC).
https://github.com/jg1uaa/lgt8fx/tree/bl_lgt8fx8e_test/lgt8f/bootloaders/lgt8fx8e

dwillmore, 2023-01-12:
Is this issue still valid or has it been overtaken by other events?  Please reevaluate @jg1uaa.

jg1uaa, 2023-02-10:
Normally Arduino-capable LGT8Fx is programmed with suitable bootloader and it is special case that re-program it. If there is a problem, please reopen this issue. Once closed.

closed 2023-02-10 by jg1uaa