dbuezas/lgt8fx #124: can not compile code with lgt8f328

hitronics4trade, 2021-03-29:
I am using emonlib.h with lgt8f328 using arduino ide but can not compile code

C:\Users\hitro\Documents\Arduino\hardware\lgt8fx-master\lgt8f\cores\lgt8f/lgtx8p.h:590:17: error: expected unqualified-id before 'volatile'

 #define VCAL    _SFR_MEM8(0xC8)

                 ^

could you please tell me what is the problem?

LaZsolt, 2021-03-29:
The problem is `emonlib.h` contain this line (number 74)
`    double VCAL;`

This variabe name same as the LGT8F328P's reference voltage calibration register's name.
You must rename every reference of VCAL variable in `emonlib.h` and `emonlib.cpp`.

LaZsolt, 2021-03-29:
Another thing.
You must modify `readVcc()`, because ADC accessing the different way of 3 reference voltages than ATmega, ADC value is 12 bit wide, ADC has gain error which has to be compensated and  `READVCC_CALIBRATION_CONST` must be recalculated.
(LGT8F328P reference voltages: 1.024 V, 2.048 V and 4.096 V)

My readVcc modification may help you: https://github.com/dbuezas/lgt8fx/discussions/123#discussion-3287695

closed 2023-01-12 by dwillmore