dbuezas/lgt8fx #139: Uploading to SSOP20(mini clone) with uno clone instead of FTDI

Gestreleth, 2021-04-14:
I was dumb enough not to order an FTDI so I found out that apparently an UNO can be used instead. Apparently you can do so by connecting the UNO's reset pin to ground, and connecting the UNO's ground, 5v, TX and RX pins accordingly to the destination board. This doesn't work with the green mini board I'm trying to upload to, only resulting in this:

`Arduino: 1.8.12 (Windows 10), Board: "LGT8F328, Default (64), Internal, 16 MHz, 328P-SSOP20 (e.g. green pseudo pro mini)"

Sketch uses 1100 bytes (3%) of program storage space. Maximum is 29696 bytes.
Global variables use 9 bytes of dynamic memory.
An error occurred while uploading the sketch
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\lgt8fx-master\docs. Is it pre-1.5?
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\lgt8fx-master\docs. Is it pre-1.5?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

In addition, the TX and RX lights on the UNO flash at around 1 second intervals, as does the blue LED on the mini clone. Is it possible to use an Uno board to upload to the green pin or should I just order an FTDI and shelf the mini clone for now?

youxiaojie, 2021-04-17:
the uno' serial port are taken by usb. when comminicate with pc.  and what is the baud of usb ttl(32u2 as converter)?

dbuezas, 2021-04-17:
On top of what you are doing, try reseting the mini board when the upload begins. And make sure the uno has an empty sketch that doesn't use serial.
The only other option without an ftdi is to use the uno as ISP.

dbuezas, 2021-05-07:
Did it work? I'll move this to a discussion

closed 2021-05-07 by dbuezas