dbuezas/lgt8fx #155: Power management library (pmu.h) doesn't work after the first analogRead is made

dbuezas, 2021-05-10:
This works fine:
```cpp
#include <PMU.h>
void setup(){
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, 1);
    PMU.sleep(PM_POFFS0, SLEEP_1S);
    digitalWrite(LED_BUILTIN, 0);
}
void loop(){
}
```

This doesn't work. It doesn't go to sleep:
```cpp
#include <PMU.h>
void setup(){
    pinMode(LED_BUILTIN, OUTPUT);
    analogRead(A1); // this breaks it!
    digitalWrite(LED_BUILTIN, 1);
    PMU.sleep(PM_POFFS0, SLEEP_1S);
    digitalWrite(LED_BUILTIN, 0);
}
void loop(){
}
```

After some debugging, I pinned point the line that causes this behaviour. It is `cores/lgt8f/wiring_analog.c:97`
```cpp
sbi(ADCSRA, ADSC);
```
which just starts one measurement.

Curiously, setting `ADCSRA = 0;` before sleeping doesn't do anything, but wrapping the sleep fn like this:
```cpp
bitClear(ADCSRA, ADEN);
PMU.sleep(PM_POFFS0, SLEEP_1S);
bitSet(ADCSRA, ADEN);
```
(i.e disabling the ADC temporarily) makes it work again.

I see that in the lib there is a `#if defined(__LGT8FX8E__)` before clearing and restoring the ports, dir registers and the ADCSRA, but replicating that behaviour doesn't work.

My assumption is that the PMU library in this core is not actually made for the lgt328p, but the 328d instead.
There's probably a lot more current that can be saved by making a new version for the 328p.

Thoughts?

dbuezas, 2021-05-10:
I see. But not only that, the board doesn't even go to sleep at all. I assume it gets woken back up. Maybe the analog comparator wakes it up?

LaZsolt, 2021-05-14:
I found an example, how to use sleep modes on page 7 in this doc: https://github.com/dbuezas/lgt8fx/blob/master/docs/Porting_from_LGT8FX8D_to_LGT8FX8P_v1.0.0.pdf

LaZsolt, 2023-01-03:
Uhh. It appeared to me after several days of studying the sleep modes this sentence:
> _In sleep modes, LGT8FX8P will not automatically disable function of analog module, such as ADC, DAC, AC (analog comparator), LVD (low voltage detector) **and so son**._

The third sentence of the Power management section. And what could be the "and so on" thing?

dbuezas, 2023-01-03:
> what could be the "and so on" thing?

Maybe the other peripherals? (serial, spi, timers, ...)

LaZsolt, 2023-01-04:
Serial, spi, timers are not analog modules so they are not "and so on", if the quoted sentence above can be taken seriously.

Anyway, after I finish studying the sleep modes I plan to fix this package's power management examples.

dbuezas, 2023-01-04:
Right I see. Then the differential amplifier could be the missing one

LaZsolt, 2023-01-04:
I found 2 more analog modules. These 2 modules can be shut down by writing ADCSRD register to 0x0.
The internal reference voltage source and the Voltage division circuit which a resistor divider with a multiplexed input.

So there are 8 analog devices with their own shutdown control bits. ADC, DAC0, AC0, AC1, LVD, DAP, IVREF and the internal voltage division circuit.

LaZsolt, 2023-02-02:
I have read this issue many times, but I just now understood, what was your idea about the issue.
> My assumption is that the PMU library in this core is not actually made for the lgt328p, but the 328d instead.
> There's probably a lot more current that can be saved by making a new version for the 328p.
>
>Thoughts?

[Low-Power library](https://github.com/rocketscream/Low-Power) for Arduino is a useful, simple and widely used library with better power settings in sleep. 
Examples:
`LowPower.standby();`
`LowPower.idle(SLEEP_8S, ADC_OFF, TIMER2_OFF, TIMER1_OFF, TIMER0_OFF, SPI_OFF, USART0_OFF, TWI_OFF);`
```
    pinMode(wakeUpPin, INPUT);   
    attachInterrupt(0, wakeUp, LOW);
    LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF); 
    detachInterrupt(0); 
```

This library could be copied into this repository and could be expanded with LGT8F328x's capabilities. With this library the PMU library will be obsolete.

sullivanzheng, 2023-02-06:
> Unfortunately I don't have a unit w/o LDO so I think 90% of the remaining consumption is that and the power LED. I'll have to find an excuse to butcher a mini-style just to see how low this chip can go in deep sleep.

I am driving hard to get that 1uA. I decimated LDO, power LED, and another LED onboard (I used some version of mini pro board with LGT8F328p without USB-TTL chip such as CH341 etc.) @dbuezas . 

@LaZsolt I tried PMU lib and also found that is only for 328D and maybe I should use some of 328E's code which is embraced by #if defined(__LGT8FX8E__). It looks to me that these code section
`#if defined(__LGT8FX8E__)
	adcsra_reg = ADCSRA;
	ddrd_reg = DDRD;
	portd_reg = PORTD;
	ddrb_reg = DDRB;
	portb_reg = PORTB;
	didr0_reg = DIDR0;

	DIDR0 = 0xff;
	ADCSRA = 0;

	DDRD &= 0xdf;
	DDRB &= 0xf9;
	PORTD |= 0x20;
	PORTB |= 0x06;
#endif`

carefully set up the peripherals that may have substantial power consumption.

Now I am getting ~41uA for DPS2 mode, just by setting DDRx to 0 and PORTx to 0xff to enable pull-up to avoid floating input. 
Clearly that is not the end of the road.

dbuezas, 2023-02-06:
Did you try turning the low voltage detector (LVD) off? AFAIK it is like the atmega's BOD but has more features.

```cpp
VDTCR |= 1<<WCE; // this should allow changing other bits during the next 6 cycles. Not sure if it is needed for VDTEN
VDTCR &= ~(1<<VDTEN);
```

LaZsolt, 2023-02-06:
Turning off the LVD:
```
      uint8_t tmp = VDTCR & ~( (1<<VDTEN) | (1<<VDREN) ); 
      VDTCR = (1<<WCE);                                     // 0x80
      VDTCR = tmp;
```

LaZsolt, 2023-02-06:
Comment edited https://github.com/dbuezas/lgt8fx/issues/155#issuecomment-1419818081

dbuezas, 2023-02-06:
does it make a difference on the consumption?

sullivanzheng, 2023-02-07:
> does it make a difference on the consumption?

I am in flight and will give you guys a heads-up once I land.

sullivanzheng, 2023-02-07:
> 

Should I turn them all off to get 1uA DPS2 sleep?

sullivanzheng, 2023-02-07:
> does it make a difference on the consumption?

Turned LVD off (the ATmega328p BOD equivalent) and there is no additional power saving. Still got ~6uA current. Trying to set ADSCRD = 0 to see if shutting down internal voltage reference and voltage divider could save more.

LaZsolt, 2023-02-07:
@sullivanzheng 
> Should I turn them all off to get 1uA DPS2 sleep?

No, you shouldn't. In DPS2 mode the LGT8F328P turning off all internal devices. Watch my experiment: https://github.com/dbuezas/lgt8fx/issues/202#issuecomment-1419681782

sullivanzheng, 2023-02-07:
> 

OMG you are right. I used your code and got **0.67uA (670nA)!** The best sleep current I ever got on any MCUs including ATtiny, ATmega, STM32, GD32, CH32, ESP8266, ESP32 and ESP32-C3. New record.

dbuezas, 2023-02-07:
That's astounding! do you wake up via reset or did you manage to get wake on pin at those currents?

sullivanzheng, 2023-02-07:
> That's astounding! do you wake up via reset or did you manage to get wake on pin at those currents?

Wake up via reset pin or wake on PortD pins doesn't change the power consumption. 

Only if you turn on many PortD pins for pin change wake up from DPS2, the MCU is more susceptible to interference  (in my test I use my hand to touch PortD pins randomly) and may drain 10s or 100s of uA additional current via the weak (20k-40k according to datasheet) pull-up resisters. 

I am trying to use LPRC to enable periodic wake up (by setting DPS2R= 0b00001111) but without success so far. The MCU can only sleep for 50ms and then wake up instead of the claimed 1s on datasheet.

dbuezas, 2023-02-07:
> instead of the claimed 1s on datasheet

maybe you need to increase the prescaler on some clock to reach the full second

sullivanzheng, 2023-02-07:
> > instead of the claimed 1s on datasheet
> 
> maybe you need to increase the prescaler on some clock to reach the full second

I don't think so. the LPRC is an independent low power clock residing in AWSON block, neither associated with 32Khz RC internal oscillator, nor 16Mhz RC oscillator.

I used my joulescope to measure the wake-sleep cycle by setting DPS2R differently:

DPS2R = 0b11**00**; // sleep for 2.37s
DPS2R = 0b11**01**; // sleep for 0.53s
DPS2R = 0b11**10**; // sleep for 0.29s
DPS2R = 0b11**11**; // sleep for 0.14s

Aha, I guess I found a datasheet error here. According to datasheet, the lowest 2 bits of DPS2R are TOS1 and TOS0., With TOS[1:0] = 00 being the shortest sleep time and 11 the longest. 

Actually it is just the opposite. Someone **may have written a wrong datasheet**, or the **MCU designer wrote the wrong verilog code.**

LaZsolt, 2023-02-07:
> Bad news is the LPRC accuracy is very crappy.

> Interestingly time for wake up is now 2.78s instead of 2.37s.

So lower VCC means longer timing. I suppose timing of this RC timer temperature dependent also.

LaZsolt, 2023-02-07:
@sullivanzheng 
Your work is great. We got valuable information from these time and current measurements, I have no such a good metering devices like joulescope.
> And by `PMU.sleep(PM_POFFS1, SLEEP_16S);` I got 15.6µA. Looks like the WDT costs ~9uA.

The value of 16.5 µA does not seem too much compared to DPS2 5.78 µA without SRAM and register retention.
Chinese quality.

sullivanzheng, 2023-02-07:
> @sullivanzheng Your work is great. We got valuable information from these time and current measurements, I have no such a good metering devices like joulescope.
>
>
> The value of 16.5 µA does not seem [...]

I don't think 16uA is the end game. That number was measured when I don't know how to properly get sub 1uA power consumption by DPS2 sleep. (I wrote a bunch of random code to fiddle with all kinds of registers including ADSCRA=0 ADSCRD=0 etc.) may incur extra power consumption. Will see if I can get that number down a bit.

In addition, 16uA additional sleep current may be offset by shorter active time of lgt due to faster clock and single cycle instruction. Here is my calculation:

I got several project using ATmega328p with 1% duty cycle and sleep current of 5uA and active current of 4.7mA. Overall average current is 52uA.

If I switch to LGT, active time may be reduced by 70% （32Mhz vs 8Mhz）, duty cycle 0.3%, active current is 11mA. sleep current  16uA, total average is 49uA.

So it seems that the key is if I can fully leverage the faster core speed and slash the active time effectively...

LaZsolt, 2023-02-08:
@sullivanzheng 
>When Vcc=3.3V
    5.78µA without SRAM and register retention for lgt8f328p
    vs
    4.7µA for ATmega328p with WDT and full SRAM and register retention

4.7 µA for ATmega328p ? Not good.
I just read the ATmega databook and it says:
```
Power Consumption at 1 MHz, 1.8V, 25°C
– Active mode: 0.2 mA
– Power-Down mode: 0.1 μA
– Power-Save mode: 0.75 μA (Including 32 kHz RTC)
```
and
```
If the On-chip Debug System is enabled by the fuse and the chip enters Sleep mode, the main clock
source is enabled and hence always consumes power. In the deeper Sleep modes, this will contribute
significantly to the total current consumption.
```

dbuezas, 2023-02-08:
Heads up: it seems there are atmega328p clones which are out of spec in regards to power consumption:
See here: https://youtu.be/PlGycKwnsSw
The rest of the videos in the series are in the comments, very interesting material finishing with a specialist looking at the dye

I assume we all source these carelessly (ebay, ali,...), so no wonder.

dbuezas, 2023-02-08:
Although last time I got 40uA on an aliexpress pro mini  and I also had other chips powered (sleeping too). He reports >100uA on his likely-fake atmega328p

LaZsolt, 2023-02-08:
[image]

dbuezas, 2023-02-08:
Out of curiosity, has anybody measured the consumption of the lgt while deep sleeping on different F_CPU clock speeds? I assume it should be the same since it isn't even running, but if isn't, there may be the possibility of dialling up the prescaler right before sleep. Again, I doubt this is a thing, but who knows

sullivanzheng, 2023-02-09:
> 

> [image]

Are we looking at the same part? I checked ATmega328p official product page but I got very different result. In my datasheet, 1.8V is no more officially specified. For that reason, I bought several ATmega168V which officially support 1.8V to deal with a 1.8V circuit design.
[image]

sullivanzheng, 2023-02-09:
> Out of curiosity, has anybody measured the consumption of the lgt while deep sleeping on different F_CPU clock speeds? I assume it should be the same since it isn't even running, but if isn't, there [...]

That is actually a very valuable question. In the PMU lib, there is a function:
```
// switch main clock to rc32k
void PowerControl::clock_switch(pmu_t mode)
{
	uint8_t btmp;

	if(mode > PM_IDLE) {
		// switch main clock to rc32k 
		btmp = (PMCR & 0x1F) | 0x50;
		PMCR = 0x80;
		PMCR = btmp;
		asm ("nop");
		asm ("nop");

		CLKPR = 0x80;
		CLKPR = 0x00; // 0x00 - no division
	}
}
```

And before PMU lib do_sleep(), it calls this clock_switch function.

However I tested with or without this function for DPS0 and DPS1 mode and found it makes no impact on current consumption.
But for ADC noise reduction mode, changing main clock saves power. (Use 32Mhz clock 1.1mA, switch to 32Khz clock 0.5mA). But I guess this also has impact on ADC speed, since ADC_clk derives from main clock from CMU.
[image]

LaZsolt, 2023-02-09:
Check these new power management library files: [lgt8fx_lowpower](https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/lgt_lowpower)
These files are modeled after the Arduino sources. Usage:
```
#include <avr\sleep.h>
#include "lgt8fx_sleep.h"
#include <avr\power.h>
#include "lgt8fx_power.h"

```

LaZsolt, 2023-02-12:
You can check what is powered on-chip before sleep [with this program](https://github.com/LaZsolt/lgt8fx/tree/master/lgt8f/libraries/ScanPower).

LaZsolt, 2023-02-13:
I found minor error.
Corrected. https://github.com/dbuezas/lgt8fx/issues/155#issuecomment-1427133482

sullivanzheng, 2023-02-13:
> Corrected. #155 (comment)

very handy library.

发自我的iPhone

LaZsolt, 2023-03-19:
> Power management library (pmu.h) doesn't work after the first analogRead is made.

Solved.

After the analogRead(); The ADIF bit of ADCSRA is set. ADIF = ADC Interrupt Flag
So when any sleep mode started, the interrupt flag is immediately break it.

Interesting. It raises further questions, why hasn't this caused problems for others?

LaZsolt, 2023-05-21:
Can this issue report be closed?

sullivanzheng, 2023-07-13:
> 

I think this issue is very well discussed. I have used DPS2 mode and LPRC as my "default deep-sleep and reset" routine for quite a few sensor and IoT projects. Very handy.

jayzakk, 2023-08-02:
Forgot to close

closed 2023-08-02 by jayzakk