dbuezas/lgt8fx #156: How to save and access LGT8F328D reference voltage calibration value?

LaZsolt, 2021-05-12:
Hi!

In LGT8F328D has a voltage referece calibration value accessing bug. @youxiaojie pointed to it https://github.com/dbuezas/lgt8fx/discussions/123#discussioncomment-583035

In `main.cpp` the `VCAL1` calibration value stored into `GPIOR1` register. But this register not protected, so any user software may overwrite it. I tried to replace it with an unique named global variable, but I was unable to define a variable which is accessible everywhere like `GPIOR1`. My idea was this variable must be placed into `Arduino.h`.

Does anyone have any idea how to make this?

dbuezas, 2021-05-13:
I think the `extern` keyboard would be the best way
https://stackoverflow.com/questions/1433204/how-do-i-use-extern-to-share-variables-between-source-files

LaZsolt, 2021-05-13:
I tried to define with extern keyword, but it didn't work.
Maybe the file `Arduino.h` wasn't a good place for the global variable definiton, because `Arduino.h` not incuded where I wanted to use.
I'll try put the definiton into file `lgtx8e.h`.

LaZsolt, 2021-05-13:
Working! :) Thanks for the idea.

A part of the `lgtx8e.h`:
```
#define VCAL    _SFR_MEM8(0xC8)
#define VCAL1   _SFR_MEM8(0xCD)
#define VCAL2   _SFR_MEM8(0xCE)
// Declare a variable for save and store ivref calibration_1
extern uint8_t _VCAL_1_;
```

A part of the `main.cpp`:
```
uint8_t _VCAL_1_;

void lgt8fx8x_init()
{
#if defined(__LGT8F_SSOP20__)
	GPIOR0 = PMXCR | 0x07;
	PMXCR = 0x80;
	PMXCR = GPIOR0;
#endif

#if defined(__LGT8FX8E__)
// store ivref calibration value (According to the LGT8F328D databook,
//              if VCAL is rewritten, the value of VCAL1 can be lost.)
    _VCAL_1_ = VCAL1;

// enable 1KB E2PROM 
	ECCR = 0x80;
	ECCR = 0x40;
.
.
.
```

A part of the `wiring_analog.c`:
```
	analog_reference = mode;

#if defined(__LGT8FX8E__) || defined(__LGT8FX8P__)
	#if defined(__LGT8FX8E__)
	if (analog_reference == INTERNAL2V56) {
        VCAL = VCAL2;
	} else {
        VCAL = _VCAL_1_;
	}
	#else
.
.
.
```

dwillmore, 2023-01-12:
Has this change been committed to a relase?  If no, please do.  If so, please test and close this issue.

LaZsolt, 2023-01-21:
This issue has been solved with the v2.0.0 relesase. Tested.

closed 2023-01-21 by LaZsolt