dbuezas/lgt8fx #157: How to build the serial number?

RudolfAtHome, 2021-05-14:
The file "**lgt8xp.h**" contains 4 pointers for reading out the serial number. In which order do the values have to be linked with one another?
#define GUID0	(*((volatile unsigned char *)0xF3))
#define GUID1	(*((volatile unsigned char *)0xF4))
#define GUID2	(*((volatile unsigned char *)0xF5))
#define GUID3	(*((volatile unsigned char *)0xF6))

LaZsolt, 2021-05-14:
I think no rule how to use it.

```
uint32_t guid = (uint32_t)&GUID0;
Serial.println(guid, HEX);
```

```
uint32_t guid = ((uint32_t)GUID3 << 24) | (GUID2 << 16) | (GUID1 << 8) | GUID0;
Serial.println(guid, HEX);
```

```
#define guid (*((volatile uint32_t *)0xF3))
Serial.println(guid, HEX);
```

RudolfAtHome, 2021-05-15:
To Lazolt: This gives very strange / incorrect results, so my question remains open!

uint32_t guid = (uint32_t)&GUID0;
Serial.println(guid, HEX);
    **Result: F3** -> wrong, that's an address?
uint32_t guid = ((uint32_t)GUID3 << 24) | (GUID2 << 16) | (GUID1 << 8) | GUID0;
Serial.println(guid, HEX);
    **Result: FFFFAB8A**  -> looks not very serious
#define guid (*((volatile uint32_t *)0xF3))
Serial.println(guid, HEX);
    **Result: 6F49AB8A** -> another result...

LaZsolt, 2021-05-15:
I tested it. I think we run into a compiler bug.
If GUID1 bigger than 0x7F then the (unit32_t) shift result upper two bytes will be 0xFFFF.
With another words. If an unsigned 8 bit variable which bigger than 0x7F and shifted left by 8, then the result negative 16 bit number.

LaZsolt, 2021-05-15:
Here is the tested GUID printing sketch.
```
/* GUID test */

uint32_t guid_a, guid_b, guid_c, guid_d;

void setup() {
  Serial.begin(38400);

  guid_a = (*((uint32_t *)&GUID0));
  Serial.println(guid_a, HEX);

  Serial.println((*((uint32_t *)&GUID0)), HEX);

#define guid_b (*((uint32_t *)0xF3))
  Serial.println(guid_b, HEX);

  Serial.println((*((uint32_t *)0xF3)), HEX);

  guid_c = ((uint32_t)GUID3 << 24) | ((uint32_t)GUID2 << 16)| ((uint32_t)GUID1 << 8) | GUID0;
  Serial.println(guid_c, HEX);

  Serial.println();
  Serial.println();
}

void loop() {}
```

dwillmore, 2023-01-12:
@RudolfAtHome does this code work for you?  If so, please close this issue.

RudolfAtHome, 2023-01-12:
It's solved for me since May 15, 2021

closed 2023-01-12 by RudolfAtHome