dbuezas/lgt8fx #165: LGT8F328 nano style SPI (PS2 stick) doesnt work

vincentj87, 2021-05-26:
i tried to use lgt8f328 to replace my arduino nano in my recent project which using ps2 stick as main controller. It was running ok with the arduino nano but when i tried to replace it with the lgt8f328 it doesnt work at all

vincentj87, 2021-05-26:
``
#include <PS2X_lib.h>
PS2X ps2x;

int error = 0;
byte type = 0;
byte vibrate = 0;
String condition="diam";

void setup() {
Serial.begin(9600);

  error = ps2x.config_gamepad( 8, 12, 4, 11, true, true);//           //GamePad(clock, command(MISO), attention(SS), data(MOSI), Pressures?, Rumble?)

  if (error == 0) {
    Serial.println("Stik terhubung");
  }

}

void loop() {
 ps2x.read_gamepad(false,vibrate);
    if ((ps2x.Analog(PSS_LY)) < 50)
  {
    condition="MAJU";
    //Serial.println("Maju");
    //maju();
  }
   else if ((ps2x.Analog(PSS_LY)) >200)
  {
    condition="MUNDUR";
    //Serial.println("Mundur");
     //mundur();
  }
   else if ((ps2x.Analog(PSS_LX)) > 200)
  {
    condition="KANAN";
    //Serial.println("Kanan");
    //kanan();
  }
   else if ((ps2x.Analog(PSS_LX)) <100 )
  {
    condition="KIRI";
   // Serial.println("Kiri");
    //kiri();
  }
  else{
    condition="DIAM";
    //Serial.println("diam");
    //diam();
  }
  Serial.print(ps2x.Analog(PSS_LX));Serial.print(" ");
  Serial.print(ps2x.Analog(PSS_LY));Serial.print(" ");
  Serial.print(ps2x.Analog(PSS_RX));Serial.print(" ");
  Serial.print(ps2x.Analog(PSS_RY));Serial.print(" ");
  Serial.print(condition);
  Serial.println(" ");
  delay(50);
}

vincentj87, 2021-05-26:
i also tried to use lgt8f328 for another project which didnt include any SPI protocol , which running fine even with i2c ,

LaZsolt, 2021-05-26:
In the latest release of the **dbuezas/lgt8fx** package, which is 1.0.6, the `delayMicroseconds()` not working correct.
Replace `wiring.c` and `Arduino.h` on your hard disk with this files:
https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/wiring.c
https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/cores/lgt8f/Arduino.h

LaZsolt, 2023-01-16:
@vincentj87 
Has the new release fixed your PS2 stick issue?

Schawen, 2023-07-31:
I'm trying to connect RFM95W to LGT8f328P (miniEVB) but getting error code -2 all time.
Where do I find the files that need to get replaced (platformIO / VS Code)?